tbsdtv/linux_media #96: TBS5922 Fails to intialize dvb-usb: found a '(null)' in warm state.

ncorrare, 2018-01-03:
Driver compiles successfully on Kernel 4.8.13 but upon rebooting I get the following in dmesg
[    5.059716] dvb-usb: found a '(null)' in cold state, will try to load a firmware
[    5.060414] dvb-usb: downloading firmware from file 'dvb-usb-tbsqbox-id5922.fw'
[    5.066029] usb 1-6: USB disconnect, device number 4
[    5.170013] dvb-usb: found a '(null)' in warm state.
[    5.170021] dvb-usb: no i2c algorithm specified
[    5.170027] dvb-usb: (null) error while loading driver (-22)
[    5.170166] usbcore: registered new interface driver tbsqbox22
[    6.846060] usb 1-6: new high-speed USB device number 6 using xhci_hcd
[    7.015915] usb 1-6: New USB device found, idVendor=734c, idProduct=5922
[    7.015918] usb 1-6: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[    7.015920] usb 1-6: Product: TBS 5922
[    7.015921] usb 1-6: Manufacturer: TBS-Tech
[    7.017073] dvb-usb: found a '(null)' in cold state, will try to load a firmware
[    7.017076] dvb-usb: downloading firmware from file 'dvb-usb-tbsqbox-id5922.fw'
[    7.129806] dvb-usb: found a '(null)' in warm state.
[    7.129812] dvb-usb: no i2c algorithm specified
[    7.129851] dvb-usb: (null) error while loading driver (-22)

Needless to say that /dev/dvb doesn't exist. I'm using Fedora 23, but I also found a similar issue referenced in the Fedora Forums using a recent Kernel
https://forums.fedoraforum.org/showthread.php?316712-dvb-usb-no-i2c-algorithm-specified&p=1801140&posted=1#post1801140

ncorrare, 2018-01-04:
Also, it's worth noting that the proprietary drivers work... albeit the tuner becoming unresponsive at times

realtmf, 2018-03-20:
Hi, did you manage to fix this pls? I got the same after building on  Ubuntu 17.10, kernel 4.13.0-37-generic, tried on both USB 2.0 and USB 3.0 controllers (with the same dmesg output):

[  358.961916] dvb-usb: downloading firmware from file 'dvb-usb-tbsqbox-id5980.fw'
[  358.961916] tbsqbox2ci: start downloading TBSQBOX2CI firmware
[  359.135213] dvb-usb: found a '(null)' in warm state.
[  359.135220] dvb-usb: no i2c algorithm specified
[  359.135276] dvb-usb: (null) error while loading driver (-22)

(Similar to https://github.com/tbsdtv/linux_media/issues/108)