tbsdtv/linux_media #102: TBS6909 Unicable not working

3rv4n1x, 2018-01-23:
TBS6909 Unicable not working

The unicable LNB is connected to input 3 like in manual.
LNB - unicable ii programmable 40mm lnb with 32ub

my kernel 
Linux asus 4.4.0-109-generic #132-Ubuntu SMP Tue Jan 9 19:52:39 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux

mode Unicable selected
grep mxl58x /etc/modprobe.d/*
/etc/modprobe.d/mxl58x.conf:options mxl58x mode=2

cat /var/log/syslog | grep TBSECP3
Jan 19 14:06:41 asus kernel: [   13.304200] TBSECP3 driver 0000:04:00.0: TurboSight TBS 6909 DVB-S/S2
Jan 19 14:06:41 asus kernel: [   13.305586] dvbdev: DVB: registering new adapter (TBSECP3 DVB Adapter)
Jan 19 14:06:41 asus kernel: [   13.425313] TBSECP3 driver 0000:04:00.0: MAC address 00:22:ab:91:b4:48
Jan 19 14:07:02 asus kernel: [   38.422430] TBSECP3 driver 0000:04:00.0: DVB: registering adapter 0 frontend 0 (TurboSight TBS 6909 DVB-S/S2 )...
Jan 19 14:07:02 asus kernel: [   38.422527] dvbdev: DVB: registering new adapter (TBSECP3 DVB Adapter)
Jan 19 14:07:03 asus kernel: [   38.544001] TBSECP3 driver 0000:04:00.0: MAC address 00:22:ab:91:b4:49
Jan 19 14:07:03 asus kernel: [   38.544023] TBSECP3 driver 0000:04:00.0: DVB: registering adapter 1 frontend 0 (TurboSight TBS 6909 DVB-S/S2 )...
Jan 19 14:07:03 asus kernel: [   38.544082] dvbdev: DVB: registering new adapter (TBSECP3 DVB Adapter)
Jan 19 14:07:03 asus kernel: [   38.665560] TBSECP3 driver 0000:04:00.0: MAC address 00:22:ab:91:b4:4a
Jan 19 14:07:03 asus kernel: [   38.665568] TBSECP3 driver 0000:04:00.0: DVB: registering adapter 2 frontend 0 (TurboSight TBS 6909 DVB-S/S2 )...
Jan 19 14:07:03 asus kernel: [   38.665626] dvbdev: DVB: registering new adapter (TBSECP3 DVB Adapter)
Jan 19 14:07:03 asus kernel: [   38.787100] TBSECP3 driver 0000:04:00.0: MAC address 00:22:ab:91:b4:4b
Jan 19 14:07:03 asus kernel: [   38.787108] TBSECP3 driver 0000:04:00.0: DVB: registering adapter 3 frontend 0 (TurboSight TBS 6909 DVB-S/S2 )...
Jan 19 14:07:03 asus kernel: [   38.787165] dvbdev: DVB: registering new adapter (TBSECP3 DVB Adapter)
Jan 19 14:07:03 asus kernel: [   38.905256] TBSECP3 driver 0000:04:00.0: MAC address 00:22:ab:91:b4:4c
Jan 19 14:07:03 asus kernel: [   38.905266] TBSECP3 driver 0000:04:00.0: DVB: registering adapter 4 frontend 0 (TurboSight TBS 6909 DVB-S/S2 )...
Jan 19 14:07:03 asus kernel: [   38.905334] dvbdev: DVB: registering new adapter (TBSECP3 DVB Adapter)
Jan 19 14:07:03 asus kernel: [   39.026819] TBSECP3 driver 0000:04:00.0: MAC address 00:22:ab:91:b4:4d
Jan 19 14:07:03 asus kernel: [   39.026828] TBSECP3 driver 0000:04:00.0: DVB: registering adapter 5 frontend 0 (TurboSight TBS 6909 DVB-S/S2 )...
Jan 19 14:07:03 asus kernel: [   39.026898] dvbdev: DVB: registering new adapter (TBSECP3 DVB Adapter)
Jan 19 14:07:03 asus kernel: [   39.148395] TBSECP3 driver 0000:04:00.0: MAC address 00:22:ab:91:b4:4e
Jan 19 14:07:03 asus kernel: [   39.148404] TBSECP3 driver 0000:04:00.0: DVB: registering adapter 6 frontend 0 (TurboSight TBS 6909 DVB-S/S2 )...
Jan 19 14:07:03 asus kernel: [   39.148467] dvbdev: DVB: registering new adapter (TBSECP3 DVB Adapter)
Jan 19 14:07:03 asus kernel: [   39.269958] TBSECP3 driver 0000:04:00.0: MAC address 00:22:ab:91:b4:4f
Jan 19 14:07:03 asus kernel: [   39.269967] TBSECP3 driver 0000:04:00.0: DVB: registering adapter 7 frontend 0 (TurboSight TBS 6909 DVB-S/S2 )...
Jan 19 14:07:03 asus kernel: [   39.270024] TBSECP3 driver 0000:04:00.0: TurboSight TBS 6909 DVB-S/S2 : PCI 0000:04:00.0, IRQ 31, MMIO 0xfebc0000
Jan 19 14:07:05 asus kernel: [   41.149641] TBSECP3 driver 0000:04:00.0: DVB: adapter 7 frontend 0 frequency 0 out of range (950000..2150000)
Jan 19 14:07:05 asus kernel: [   41.150575] TBSECP3 driver 0000:04:00.0: DVB: adapter 6 frontend 0 frequency 0 out of range (950000..2150000)
Jan 19 14:07:05 asus kernel: [   41.151351] TBSECP3 driver 0000:04:00.0: DVB: adapter 5 frontend 0 frequency 0 out of range (950000..2150000)
Jan 19 14:07:05 asus kernel: [   41.152687] TBSECP3 driver 0000:04:00.0: DVB: adapter 4 frontend 0 frequency 0 out of range (950000..2150000)
Jan 19 14:07:05 asus kernel: [   41.153523] TBSECP3 driver 0000:04:00.0: DVB: adapter 3 frontend 0 frequency 0 out of range (950000..2150000)
Jan 19 14:07:05 asus kernel: [   41.154342] TBSECP3 driver 0000:04:00.0: DVB: adapter 2 frontend 0 frequency 0 out of range (950000..2150000)
Jan 19 14:07:05 asus kernel: [   41.155136] TBSECP3 driver 0000:04:00.0: DVB: adapter 1 frontend 0 frequency 0 out of range (950000..2150000)
Jan 19 14:07:05 asus kernel: [   41.155940] TBSECP3 driver 0000:04:00.0: DVB: adapter 0 frontend 0 frequency 0 out of range (950000..2150000)
Jan 19 14:09:41 asus kernel: [   13.430107] TBSECP3 driver 0000:04:00.0: TurboSight TBS 6909 DVB-S/S2
Jan 19 14:09:41 asus kernel: [   13.431560] dvbdev: DVB: registering new adapter (TBSECP3 DVB Adapter)
Jan 19 14:09:41 asus kernel: [   13.549316] TBSECP3 driver 0000:04:00.0: MAC address 00:22:ab:91:b4:48
Jan 19 14:10:01 asus kernel: [   38.578484] TBSECP3 driver 0000:04:00.0: DVB: registering adapter 0 frontend 0 (TurboSight TBS 6909 DVB-S/S2 )...
Jan 19 14:10:01 asus kernel: [   38.578580] dvbdev: DVB: registering new adapter (TBSECP3 DVB Adapter)
Jan 19 14:10:01 asus kernel: [   38.693576] TBSECP3 driver 0000:04:00.0: MAC address 00:22:ab:91:b4:49
Jan 19 14:10:01 asus kernel: [   38.693588] TBSECP3 driver 0000:04:00.0: DVB: registering adapter 1 frontend 0 (TurboSight TBS 6909 DVB-S/S2 )...
Jan 19 14:10:01 asus kernel: [   38.693654] dvbdev: DVB: registering new adapter (TBSECP3 DVB Adapter)
Jan 19 14:10:01 asus kernel: [   38.815215] TBSECP3 driver 0000:04:00.0: MAC address 00:22:ab:91:b4:4a
Jan 19 14:10:01 asus kernel: [   38.815224] TBSECP3 driver 0000:04:00.0: DVB: registering adapter 2 frontend 0 (TurboSight TBS 6909 DVB-S/S2 )...
Jan 19 14:10:01 asus kernel: [   38.815287] dvbdev: DVB: registering new adapter (TBSECP3 DVB Adapter)
Jan 19 14:10:01 asus kernel: [   38.897692] TBSECP3 driver 0000:04:00.0: MAC address 00:22:ab:91:b4:4b
Jan 19 14:10:01 asus kernel: [   38.897701] TBSECP3 driver 0000:04:00.0: DVB: registering adapter 3 frontend 0 (TurboSight TBS 6909 DVB-S/S2 )...
Jan 19 14:10:01 asus kernel: [   38.897760] dvbdev: DVB: registering new adapter (TBSECP3 DVB Adapter)
Jan 19 14:10:01 asus kernel: [   38.997867] TBSECP3 driver 0000:04:00.0: MAC address 00:22:ab:91:b4:4c
Jan 19 14:10:01 asus kernel: [   38.997874] TBSECP3 driver 0000:04:00.0: DVB: registering adapter 4 frontend 0 (TurboSight TBS 6909 DVB-S/S2 )...
Jan 19 14:10:01 asus kernel: [   38.997935] dvbdev: DVB: registering new adapter (TBSECP3 DVB Adapter)
Jan 19 14:10:01 asus kernel: [   39.098019] TBSECP3 driver 0000:04:00.0: MAC address 00:22:ab:91:b4:4d
Jan 19 14:10:01 asus kernel: [   39.098027] TBSECP3 driver 0000:04:00.0: DVB: registering adapter 5 frontend 0 (TurboSight TBS 6909 DVB-S/S2 )...
Jan 19 14:10:01 asus kernel: [   39.098089] dvbdev: DVB: registering new adapter (TBSECP3 DVB Adapter)
Jan 19 14:10:01 asus kernel: [   39.198173] TBSECP3 driver 0000:04:00.0: MAC address 00:22:ab:91:b4:4e
Jan 19 14:10:01 asus kernel: [   39.198183] TBSECP3 driver 0000:04:00.0: DVB: registering adapter 6 frontend 0 (TurboSight TBS 6909 DVB-S/S2 )...
Jan 19 14:10:01 asus kernel: [   39.198252] dvbdev: DVB: registering new adapter (TBSECP3 DVB Adapter)
Jan 19 14:10:01 asus kernel: [   39.298349] TBSECP3 driver 0000:04:00.0: MAC address 00:22:ab:91:b4:4f
Jan 19 14:10:01 asus kernel: [   39.298358] TBSECP3 driver 0000:04:00.0: DVB: registering adapter 7 frontend 0 (TurboSight TBS 6909 DVB-S/S2 )...
Jan 19 14:10:01 asus kernel: [   39.298427] TBSECP3 driver 0000:04:00.0: TurboSight TBS 6909 DVB-S/S2 : PCI 0000:04:00.0, IRQ 31, MMIO 0xfebc0000
Jan 19 14:10:03 asus kernel: [   41.312182] TBSECP3 driver 0000:04:00.0: DVB: adapter 7 frontend 0 frequency 0 out of range (950000..2150000)
Jan 19 14:10:03 asus kernel: [   41.313068] TBSECP3 driver 0000:04:00.0: DVB: adapter 6 frontend 0 frequency 0 out of range (950000..2150000)
Jan 19 14:10:03 asus kernel: [   41.313820] TBSECP3 driver 0000:04:00.0: DVB: adapter 5 frontend 0 frequency 0 out of range (950000..2150000)
Jan 19 14:10:03 asus kernel: [   41.314560] TBSECP3 driver 0000:04:00.0: DVB: adapter 4 frontend 0 frequency 0 out of range (950000..2150000)
Jan 19 14:10:03 asus kernel: [   41.315245] TBSECP3 driver 0000:04:00.0: DVB: adapter 3 frontend 0 frequency 0 out of range (950000..2150000)
[... cut ...]
Tvheadend and Astra not lock freq
Only lock in mode=1

Jan 23 02:43:13 asus tvheadend[811]: mpegts: 12246H in Unicable - tuning on TurboSight TBS 6909 DVB-S/S2  #0 : master for #0
Jan 23 02:43:13 asus tvheadend[811]: subscription: 000E: "scan" subscribing to mux "12246H", weight: 6, adapter: "TurboSight TBS 6909 DVB-S/S2  #0 : master for #0", network: "Unicable", service: "Raw PID Subscription"
Jan 23 02:43:23 asus tvheadend[811]: mpegts: 12246H in Unicable - scan no data, failed
------------------------

anthonytex, 2018-02-20:
Hello @3rv4n1x have you managed to solve this issue? i'm here with your same problem mode 2 doesn't work at all, mode 1 works fine...

anthonytex, 2018-02-20:
@crazycat69 may you take a look please :) can i provide some logs? Mode 1 as written works perfectly changing from mode 1 to mode2 without touching anything: not works

anthonytex, 2018-02-20:
@crazycat69 thank anyway for your answer, i hope that someone at tbs will take a look :) just registered to their forum:)

3rv4n1x, 2018-02-20:
Hi, yes i got it working!
Step 1: downlaod the driver:
$ git clone https://github.com/tbsdtv/media_build.git
$ git clone --depth=1 https://github.com/tbsdtv/linux_media.git -b latest ./medias

Step2: go to "media" folder， and modify a file name "mxl58x.c":
 # vi /media/dvb-frontends/mxl58x.c 
 then comment out this section of the code from line "1548" ot line "1558", like this
[image]<br /><a target='_blank' href='https://pt.imgbb.com/'>upload image</a><br />

 Step3: compile the driver and install driver-----reboot server---- check.
Mode 1 works like unicable mode!

anthonytex, 2018-02-21:
thanks thanks thanks @3rv4n1x works like a charm! ;) 
So there must be a bug somewhere in this module hope someone in TBS will have a look :)

malakudi, 2018-10-03:
Please reopen this ticket since issue still exists.
Correct fix of the issue is the following:
```
--- mxl58x.c	2018-10-03 09:19:29.000000000 +0300
+++ mxl58x.c	2018-10-03 09:31:47.128176219 +0300
@@ -801,7 +801,6 @@
 
 	switch (mode) {
 	case 2:
-			break;
 	case 1:
 		cfg->set_voltage(i2c, voltage, state->rf_in);
 		break;
@@ -824,7 +823,6 @@
 
 	switch (mode) {
 	case 2:
-			break;
 	case 1:
 		set_input_tone(fe, tone, state->rf_in);
 		break;
```
Voltage and Tone should be allowed in mode 2.

anthonytex, 2018-10-03:
I can confirm that the issue is still present.
I cannot test the patch until weekend but @malakudi thanks in advance for your fix.
@crazycat69 Please reopen this issue otherwise it will never be fixed.
Thanks

3rv4n1x, 2018-10-03:
@malakudi  Thank you , i reopen the issue.

malakudi, 2018-10-04:
@3rv4n1x , can you verify that my fix works for you too in mode 2?

anthonytex, 2018-10-06:
@malakudi Hello, so this patch seems to work: at least now mode 2 is able to tune. So many many thanks!
But sometimes i've problem that randomly appears: when i tune a channel it seems that the tbs6909 is not able to lock, so i have to "kill" the subscription and re-ask a tune than it works.
To be completely honest it might be depend to the new version of tvheadend (git). I'll test these days and report it back. if @3rv4n1x could also test we will have one more feedback.
For now thanks!

joaodrr, 2018-10-16:
I have a 4 satellite system with Unicable II (Inverto) and it still does not work.

I only get something in mode = 1 and only the satellites A and B. In mode = 2, I get nothing.

malakudi, 2018-10-16:
Most probably you failed to correctly patch the driver source file. You have to patch file /media/drivers/media/dvb-frontends/mxl58x.c, then rebuild and make install

joaodrr, 2018-10-16:
Okay, it's working randomly... sometimes works sometimes not.

As @anthonytex has reported, there are problems to lock.

malakudi, 2018-10-17:
I use w_scan for testing (latest versions support JESS) and it works all the times with my patch. Random problems in locking could be related to timeouts etc, or to simultaneous tuning attempts, what software are you using? Since unicable/jess is considered a "shared" bus, you can't have more than one concurrent tuning process in the same bus. All 8 tuners of TBS 6909 are using the same cable, so they are considered to be on the same "shared" bus. When one tuner starts the tuning process, it raises voltage to 18V and sends diseqc, then drops to 13V. If another tuner interferes at this moment, both will be unable to lock. So each tuner needs to tune with a little delay from the rest tuners.

mode 1 has a filter in the code that blocks all diseqc that don't start with EX, so JESS will never work in mode 1. Mode 2 doesn't have this. The code is in send_master_cmd
```
	if(mode == 1){
		state->diseqcsign= true;
		memcpy(&state->diseqcMsg,&diseqcMsgPtr,sizeof(MXL_HYDRA_DISEQC_TX_MSG_T));
	  return 0;

	}
```
and in senddiseqcAbortTune
```
if(state->diseqcsign){
	state->diseqcMsg.diseqcId = state->rf_in;
	state->diseqcMsg.bufMsg[0] = 0xe0;
```

mode 2 was just lucking the option to set voltage and that's what my patch changed. You should be able to tune all your tuners if you have delays between them.

malakudi, 2019-11-25:
With the latest changes in mxl58x.c code , JESS works fine for me combined with TBS6909 and Cesbo Astra latest version that support JESS. Mumudvb also works but I prefer Astra. I believe the ticket can be closed, current code supports JESS and Unicable in mode=2 correctly.

closed 2019-11-25 by crazycat69