tbsdtv/linux_media #105: Compiling under kernel 4.15.x doesn't work - dvb_ringbuffer.c:  error: implicit declaration of function 'ACCESS_ONCE'

mnowiasz, 2018-02-05:
Hi, trying to compile tbsdtv will result in an error:

```
CC [M]  /root/tbsdtv/work/media_build/v4l/dvb_ringbuffer.o
/root/tbsdtv/work/media_build/v4l/dvb_ringbuffer.c: In function 'dvb_ringbuffer_free':
/root/tbsdtv/work/media_build/v4l/dvb_ringbuffer.c:74:9: error: implicit declaration of function 'ACCESS_ONCE'; did you mean '__READ_ONCE'? [-Werror=implicit-function-declaration]
  free = ACCESS_ONCE(rbuf->pread) - rbuf->pwrite;
         ^~~~~~~~~~~
         __READ_ONCE
cc1: some warnings being treated as errors
make[3]: *** [scripts/Makefile.build:317: /root/tbsdtv/work/media_build/v4l/dvb_ringbuffer.o] Error 1
```

CvH, 2018-02-05:
Its likely because the current used media_tree is older then your used kernel, current solution - don't use a recent kernel - use an older one :( 
Same problem appears at every new kernel release as this out of tree drivers need to be adopted to the current kernel due the lack of proper linux kernel support.

estimadarocha, 2018-02-05:
same as #104

mnowiasz, 2018-03-12:
Well, it has been more than a month - and 4.15.9 is the current kernel's version, soon to be replaced by 4.16. 

is this still under active development or do I have to change my hardware to have an spectre/meltdown hardened kernel? :-1:

crazycat69, 2018-03-14:
Please check updated [media_build](https://github.com/tbsdtv/media_build) + [linux_media/latest](https://github.com/tbsdtv/linux_media)

mnowiasz, 2018-03-14:
Works, thank you very much :+1: 

uname -a
Linux zuul 4.15.9-gentoo #1 SMP PREEMPT Wed Mar 14 12:28:55 CET 2018 x86_64 Intel(R) Celeron(R) CPU N3150 @ 1.60GHz GenuineIntel GNU/Linux

dmesg | grep Turbo
[   15.140557] dvb-usb: found a 'TurboSight TBS 5220 DVB-T/T2/C' in cold state, will try to load a firmware
[   15.255128] dvb-usb: found a 'TurboSight TBS 5220 DVB-T/T2/C' in warm state.
[   15.255613] dvbdev: DVB: registering new adapter (TurboSight TBS 5220 DVB-T/T2/C)
[   15.392487] usb 1-5.1: DVB: registering adapter 0 frontend 0 (TurboSight TBS 5220 DVB-T/T2/C)...
[   15.393478] dvb-usb: TurboSight TBS 5220 DVB-T/T2/C successfully initialized and connected.
[   15.393547] dvb-usb: found a 'TurboSight TBS 5220 DVB-T/T2/C' in cold state, will try to load a firmware
[   15.503163] dvb-usb: found a 'TurboSight TBS 5220 DVB-T/T2/C' in warm state.
[   15.503641] dvbdev: DVB: registering new adapter (TurboSight TBS 5220 DVB-T/T2/C)
[   15.643526] usb 1-5.2: DVB: registering adapter 1 frontend 0 (TurboSight TBS 5220 DVB-T/T2/C)...
[   15.644897] dvb-usb: TurboSight TBS 5220 DVB-T/T2/C successfully initialized and connected.

closed 2018-03-14 by mnowiasz