tbsdtv/linux_media #107: Build errors on Arch Linux with kernel 4.15.2-2-ARCH

highfellow, 2018-02-12:
Hi, I have been trying to build drivers for my TBS-6281SE digital TV card. I followed a link to here from the relevant page of the downloads section of your website. I tried to build the code following the instructions on the wiki, but I get the following errors:

    $ make
    ...
    Patches for 4.15.2-2-ARCH already applied.
    make[2]: Leaving directory '/local/home/andy/git/github-tbsdtv/media_build/linux'
    make -C /lib/modules/4.15.2-2-ARCH/build SUBDIRS=/local/home/andy/git/github-tbsdtv/media_build/v4l  modules
    make[2]: Entering directory '/usr/lib/modules/4.15.2-2-ARCH/build'
    CC [M]  /local/home/andy/git/github-tbsdtv/media_build/v4l/saa716x_ff_ir.o
    /local/home/andy/git/github-tbsdtv/media_build/v4l/saa716x_ff_ir.c: In function 'saa716x_ir_init':
    /local/home/andy/git/github-tbsdtv/media_build/v4l/saa716x_ff_ir.c:206:2: error: implicit declaration of function 'init_timer'; did you mean 'init_timers'? [-Werror=implicit-function-declaration]
    init_timer(&ir->keyup_timer);
    ^~~~~~~~~~
    init_timers
    /local/home/andy/git/github-tbsdtv/media_build/v4l/saa716x_ff_ir.c:207:27: error: assignment from incompatible pointer type [-Werror=incompatible-pointer-types]
    ir->keyup_timer.function = ir_emit_keyup;
    ^
    /local/home/andy/git/github-tbsdtv/media_build/v4l/saa716x_ff_ir.c:208:17: error: 'struct timer_list' has no member named 'data'
    ir->keyup_timer.data = (unsigned long) ir;
    ^
    /local/home/andy/git/github-tbsdtv/media_build/v4l/saa716x_ff_ir.c:240:28: error: assignment from incompatible pointer type [-Werror=incompatible-pointer-types]
    input_dev->timer.function = ir_repeat_key;
    ^
    /local/home/andy/git/github-tbsdtv/media_build/v4l/saa716x_ff_ir.c:241:18: error: 'struct timer_list' has no member named 'data'
    input_dev->timer.data = (unsigned long) ir;
    ^
    cc1: some warnings being treated as errors

Can you help with this?

eelstrebor, 2018-02-12:
It won't build on any 4.15.* kernel on Ubuntu 16.04.

highfellow, 2018-02-12:
I did say in the issue title that I am running Arch Linux. Is the problem with the kernel version? If so is there anything I can do to fix this?

CvH, 2018-02-12:
https://github.com/tbsdtv/linux_media/issues/104
https://github.com/tbsdtv/linux_media/issues/105
:) wait for an update

crazycat69, 2018-02-12:
4757a2712ce913d1b37800ca61ae473cf2b7e580

highfellow, 2018-02-13:
I have now got it to build by downgrading my kernel to 4.14.4-1-ARCH. The modules build okay, but I have trouble loading them into the kernel:

    $ sudo modprobe saa716x-tbs-dvb
    [sudo] password for andy:
    modprobe: ERROR: could not insert 'saa716x_tbs_dvb': Invalid argument
    $ dmesg
    [ 1726.923967] dvb_core: disagrees about version of symbol media_devnode_remove
    [ 1726.923969] dvb_core: Unknown symbol media_devnode_remove (err -22)
    [ 1726.924003] dvb_core: disagrees about version of symbol media_create_intf_link
    [ 1726.924004] dvb_core: Unknown symbol media_create_intf_link (err -22)
    [ 1726.924009] dvb_core: disagrees about version of symbol media_create_pad_links
    [ 1726.924010] dvb_core: Unknown symbol media_create_pad_links (err -22)
    [ 1726.924015] dvb_core: disagrees about version of symbol media_create_pad_link
    [ 1726.924015] dvb_core: Unknown symbol media_create_pad_link (err -22)
    [ 1726.924027] dvb_core: disagrees about version of symbol media_device_register_entity
    [ 1726.924028] dvb_core: Unknown symbol media_device_register_entity (err -22)
    [ 1726.924042] dvb_core: disagrees about version of symbol media_entity_pads_init
    [ 1726.924042] dvb_core: Unknown symbol media_entity_pads_init (err -22)
    [ 1726.924060] dvb_core: disagrees about version of symbol media_devnode_create
    [ 1726.924060] dvb_core: Unknown symbol media_devnode_create (err -22)
    [ 1726.924065] dvb_core: disagrees about version of symbol media_device_unregister_entity
    [ 1726.924066] dvb_core: Unknown symbol media_device_unregister_entity (err -22)

I assume this is because the version of the kernel the modules were built against still doesn't match the running version? If I were to look for a commit to linux_media where the kernel version matched the one I am now running, would that solve the problem?

crazycat69, 2018-02-13:
REmove old media modules before install, see [media_build/install.sh](https://github.com/tbsdtv/media_build/blob/master/install.sh) script

highfellow, 2018-02-14:
Thanks @crazycat69 , I now have the saa716x_tbs-dvb module both compiling and loading against a 4.15.2-2-ARCH kernel. Your commit https://github.com/tbsdtv/media_build/commit/4757a2712ce913d1b37800ca61ae473cf2b7e580 has stopped it failing to build, and the symbol versions match okay once I remove the old media modules before installing. I can't yet confirm that the card is working as there seems to be a problem with kodi and  the mythtv backend, but the TV guide is loading so there must be some communication with the card.

closed 2018-03-14 by crazycat69