tbsdtv/linux_media #108: TBS5520 no i2c algorithm specified error

Tnix76, 2018-02-13:
Hi,
I have an issue with TBS5520 on Fedora 27. Driver compiles fine but when I reboot or connect the tuner I get the following (dmesg):

```
[    4.254988] tbs5520: end downloading tbs5520 firmware
[    4.254997] dvb-usb: found a '(null)' in warm state.
[    4.255001] dvb-usb: no i2c algorithm specified
[    4.255023] dvb-usb: (null) error while loading driver (-22)
[    4.255063] usbcore: registered new interface driver tbs5520
....
[    5.884019] usb 3-5: new high-speed USB device number 12 using xhci_hcd
[    6.010599] usb 3-5: New USB device found, idVendor=734c, idProduct=5520
[    6.010601] usb 3-5: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[    6.010602] usb 3-5: Product: TBS 5520
[    6.010603] usb 3-5: Manufacturer: TBS-Tech
[    6.011900] dvb-usb: found a '(null)' in cold state, will try to load a firmware
[    6.012043] dvb-usb: downloading firmware from file 'dvb-usb-tbs5520-01.fw'
[    6.012044] tbs5520: start downloading tbs5520 firmware
[    6.119002] tbs5520: end downloading tbs5520 firmware
[    6.119008] dvb-usb: found a '(null)' in warm state.
[    6.119011] dvb-usb: no i2c algorithm specified
[    6.119034] dvb-usb: (null) error while loading driver (-22)
```

Any help?

Thanks.

crazycat69, 2018-02-14:
This related to your USB 3.0 host (xhci).
Better connect to USB 2.0 port (ehci)

Tnix76, 2018-02-14:
Hi again and thanks for your answer. My machine is a MSI AG240-2PE All-in-one. It has several USB ports both 2.0 and 3.0. Tuner is connected in a pair of USB 2.0 ports. It was running fine until a few later kernel releases of Fedora 26 I think. I'm afraid that USB 2.0 and 3.0 ports are managed by xhci driver and ehci driver is for several USB internal devices like webcam. If this is the case I don't know how I can I force to use ehci driver instead of xhci because I think USB drivers are compiled into kernel and there's no kernel modules for them. About of this I've posted in several Fedora forums and I haven't get any successful answer.
Next the output of lsusb commands:

```
$ lsusb
Bus 002 Device 002: ID 8087:8000 Intel Corp. 
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 8087:8008 Intel Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 002: ID 0bda:0411 Realtek Semiconductor Corp. 
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 007: ID 5986:055c Acer, Inc 
Bus 003 Device 012: ID 734c:5520 TBS Technologies China 
Bus 003 Device 003: ID 046d:c505 Logitech, Inc. Cordless Mouse+Keyboard Receiver
Bus 003 Device 011: ID 8087:07dc Intel Corp. 
Bus 003 Device 009: ID 0bda:0139 Realtek Semiconductor Corp. RTS5139 Card Reader Controller
Bus 003 Device 010: ID 08bb:29b3 Texas Instruments PCM2903B Audio CODEC
Bus 003 Device 008: ID 10c4:ea70 Cygnal Integrated Products, Inc. CP210x UART Bridge
Bus 003 Device 006: ID 0424:2512 Standard Microsystems Corp. USB 2.0 Hub
Bus 003 Device 004: ID 046d:c52f Logitech, Inc. Unifying Receiver
Bus 003 Device 002: ID 0bda:5411 Realtek Semiconductor Corp. 
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

```
$ lsusb -t
/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/6p, 5000M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 5000M
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/14p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 2: Dev 6, If 0, Class=Hub, Driver=hub/2p, 480M
            |__ Port 1: Dev 8, If 1, Class=Vendor Specific Class, Driver=cp210x, 12M
            |__ Port 1: Dev 8, If 0, Class=Vendor Specific Class, Driver=cp210x, 12M
            |__ Port 2: Dev 10, If 3, Class=Human Interface Device, Driver=usbhid, 12M
            |__ Port 2: Dev 10, If 1, Class=Audio, Driver=snd-usb-audio, 12M
            |__ Port 2: Dev 10, If 2, Class=Audio, Driver=snd-usb-audio, 12M
            |__ Port 2: Dev 10, If 0, Class=Audio, Driver=snd-usb-audio, 12M
        |__ Port 1: Dev 4, If 0, Class=Human Interface Device, Driver=usbhid, 12M
        |__ Port 1: Dev 4, If 1, Class=Human Interface Device, Driver=usbhid, 12M
    |__ Port 3: Dev 3, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M
    |__ Port 3: Dev 3, If 1, Class=Human Interface Device, Driver=usbhid, 1.5M
    |__ **Port 5: Dev 12, If 0, Class=Vendor Specific Class, Driver=, 480M**
    |__ Port 7: Dev 7, If 0, Class=Video, Driver=uvcvideo, 480M
    |__ Port 7: Dev 7, If 1, Class=Video, Driver=uvcvideo, 480M
    |__ Port 10: Dev 9, If 0, Class=Vendor Specific Class, Driver=rtsx_usb, 480M
    |__ Port 12: Dev 11, If 0, Class=Wireless, Driver=btusb, 12M
    |__ Port 12: Dev 11, If 1, Class=Wireless, Driver=btusb, 12M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=ehci-pci/2p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/8p, 480M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=ehci-pci/2p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/6p, 480M

```

If I can't solve this issue I'm afraid I'm having to sell my tuner :(

Thanks again.

crazycat69, 2018-02-14:
I still use Kubuntu 14.04LTS with 4.4 kernel. Now connect 5520 to USB 3.0 port, some interest in dmesg on init:
**15.02.2018 00:00:38	usb 3-3	new high-speed USB device number 5 using xhci_hcd
15.02.2018 00:00:38	usb 3-3	New USB device found, idVendor=734c, idProduct=5520
15.02.2018 00:00:38	usb 3-3	New USB device strings: Mfr=0, Product=0, SerialNumber=0
15.02.2018 00:00:38	dvb-usb	found a 'TBS 5520 USB2.0' in cold state, will try to load a firmware
15.02.2018 00:00:38	dvb-usb	downloading firmware from file 'dvb-usb-tbsqbox-id5520.fw'
15.02.2018 00:00:38	tbs5520	start downloading tbs5520 firmware
15.02.2018 00:00:38	tbs5520	end downloading tbs5520 firmware
15.02.2018 00:00:38	dvb-usb	found a 'TBS 5520 USB2.0' in warm state.
15.02.2018 00:00:38	dvb-usb	will pass the complete MPEG2 transport stream to the software demuxer.
15.02.2018 00:00:38	dvbdev	DVB: registering new adapter (TBS 5520 USB2.0)
15.02.2018 00:00:38	usb 3-3	media controller created
15.02.2018 00:00:38	dvb-usb	MAC address: 00:00:00:00:00:00
15.02.2018 00:00:38	dvbdev	dvb_create_media_entity: media entity 'dvb-demux' registered.
15.02.2018 00:00:38	i2c i2c-14	avl6882: attach failed family id mismatch
15.02.2018 00:00:38	TBS5520	frontend attach failed
15.02.2018 00:00:38	dvb-usb	no frontend was attached by 'TBS 5520 USB2.0'
15.02.2018 00:00:38	input	IR-receiver inside an USB DVB receiver as /devices/pci0000:00/0000:00:14.0/usb3/3-3/input/input10
15.02.2018 00:00:38	dvb-usb	schedule remote query interval to 150 msecs.
15.02.2018 00:00:38	dvb-usb	TBS 5520 USB2.0 successfully initialized and connected.
15.02.2018 00:00:38	usb 3-3	USB disconnect, device number 5
15.02.2018 00:00:38	dvb-usb	TBS 5520 USB2.0 successfully deinitialized and disconnected.**
15.02.2018 00:00:40	usb 3-3	new high-speed USB device number 6 using xhci_hcd
15.02.2018 00:00:40	usb 3-3	New USB device found, idVendor=734c, idProduct=5520
15.02.2018 00:00:40	usb 3-3	New USB device strings: Mfr=1, Product=2, SerialNumber=0
15.02.2018 00:00:40	usb 3-3	Product: TBS 5520
15.02.2018 00:00:40	usb 3-3	Manufacturer: TBS-Tech
15.02.2018 00:00:40	dvb-usb	found a 'TBS 5520 USB2.0' in cold state, will try to load a firmware
15.02.2018 00:00:40	dvb-usb	downloading firmware from file 'dvb-usb-tbsqbox-id5520.fw'
15.02.2018 00:00:40	tbs5520	start downloading tbs5520 firmware
15.02.2018 00:00:40	tbs5520	end downloading tbs5520 firmware
15.02.2018 00:00:40	dvb-usb	found a 'TBS 5520 USB2.0' in warm state.
15.02.2018 00:00:40	dvb-usb	will pass the complete MPEG2 transport stream to the software demuxer.
15.02.2018 00:00:40	dvbdev	DVB: registering new adapter (TBS 5520 USB2.0)
15.02.2018 00:00:40	usb 3-3	media controller created
15.02.2018 00:00:40	dvb-usb	MAC address: 00:00:00:00:00:00
15.02.2018 00:00:40	dvbdev	dvb_create_media_entity: media entity 'dvb-demux' registered.
15.02.2018 00:00:40	i2c i2c-14	avl6882: found id=0xb family_id=0x68624955
15.02.2018 00:00:40	i2c i2c-14	r848: Rafael R848 successfully attached
15.02.2018 00:00:40	tbs5520	TBS5520: frontend attached
15.02.2018 00:00:40		
15.02.2018 00:00:40	usb 3-3	DVB: registering adapter 3 frontend 0 (Availink AVL6882)...
15.02.2018 00:00:40	dvbdev	dvb_create_media_entity: media entity 'Availink AVL6882' registered.
15.02.2018 00:00:40	input	IR-receiver inside an USB DVB receiver as /devices/pci0000:00/0000:00:14.0/usb3/3-3/input/input11
15.02.2018 00:00:40	dvb-usb	schedule remote query interval to 150 msecs.
15.02.2018 00:00:40	dvb-usb	TBS 5520 USB2.0 successfully initialized and connected.
15.02.2018 00:00:40	dvb-usb	TBS 5520 USB2.0 successfully initialized and connected.

crazycat69, 2018-02-14:
lsusb -t -v
/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 5000M
**/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 480M
    |__ Port 3: Dev 6, If 0, Class=Vendor Specific Class, Driver=tbs5520, 480M**
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=ehci-pci/2p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/6p, 480M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=ehci-pci/2p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/6p, 480M
        |__ Port 2: Dev 3, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M
        |__ Port 5: Dev 4, If 0, Class=Mass Storage, Driver=usb-storage, 480M
        |__ Port 6: Dev 5, If 0, Class=Mass Storage, Driver=usb-storage, 480M

Tnix76, 2018-02-14:
In my case I note that I have '(null)' instead of 'TBS 5520 USB2.0'

`kernel: dvb-usb: found a '(null)' in cold state, will try to load a firmware`

`dvb-usb found a 'TBS 5520 USB2.0' in cold state, will try to load a firmware`

Tnix76, 2018-02-19:
With kernel 4.15.3-300.fc27.x86_64 the error has changed:

```
[    5.887164] usb 3-5: new high-speed USB device number 12 using xhci_hcd
[    6.013571] usb 3-5: New USB device found, idVendor=734c, idProduct=5520
[    6.013573] usb 3-5: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[    6.013574] usb 3-5: Product: TBS 5520
[    6.013576] usb 3-5: Manufacturer: TBS-Tech
[    6.015005] dvb-usb: found a 'TBS 5520 USB2.0' in cold state, will try to load a firmware
[    6.015101] dvb-usb: downloading firmware from file 'dvb-usb-tbsqbox-id5520.fw'
[    6.015102] tbs5520: start downloading tbs5520 firmware
[    6.127113] tbs5520: end downloading tbs5520 firmware
[    6.127119] dvb-usb: found a 'TBS 5520 USB2.0' in warm state.
[    6.127358] dvb-usb: will pass the complete MPEG2 transport stream to the software demuxer.
[    6.127436] dvbdev: DVB: registering new adapter (TBS 5520 USB2.0)
[    6.127437] dvb-usb: MAC address: 00:00:00:00:00:00
[    6.130720] i2c i2c-8: avl6882: found id=0xb family_id=0x68624955
[    6.136718] i2c i2c-8: r848: Rafael R848 successfully attached
[    6.136720] tbs5520: TBS5520: frontend attached

[    6.136896] dvb-usb: no frontend was attached by 'TBS 5520 USB2.0'
[    6.137100] input: IR-receiver inside an USB DVB receiver as /devices/pci0000:00/0000:00:14.0/usb3/3-5/input/input20
[    6.137248] dvb-usb: schedule remote query interval to 150 msecs.
[    6.137250] dvb-usb: TBS 5520 USB2.0 successfully initialized and connected.
[    6.295222] general protection fault: 0000 [#1] SMP PTI
[    6.295226] Modules linked in: r848(OE) vfat fat arc4 avl6882(OE) snd_usb_audio intel_rapl snd_usbmidi_lib x86_pkg_temp_thermal snd_rawmidi intel_powerclamp iTCO_wdt coretemp iTCO_vendor_support kvm_intel dvb_usb_tbs5520(OE) dvb_usb snd_hda_codec_realtek btusb snd_hda_codec_hdmi snd_hda_codec_generic btrtl iwlmvm uvcvideo kvm btbcm btintel bluetooth videobuf2_vmalloc dvb_core videobuf2_memops mac80211 videobuf2_v4l2 cp210x videobuf2_core joydev videodev media irqbypass crct10dif_pclmul snd_hda_intel crc32_pclmul snd_hda_codec nvidia_drm(POE) ecdh_generic nvidia_modeset(POE) ghash_clmulni_intel rtsx_usb_ms snd_hda_core memstick intel_cstate snd_hwdep iwlwifi intel_uncore nvidia(POE) snd_seq intel_rapl_perf cfg80211 snd_seq_device snd_pcm i2c_i801 snd_timer mei_me ipmi_devintf msi_wmi ipmi_msghandler
[    6.295262]  snd rfkill sparse_keymap nuvoton_cir lpc_ich soundcore mei shpchp ie31200_edac rc_core vboxpci(OE) vboxnetadp(OE) vboxnetflt(OE) binfmt_misc vboxdrv(OE) rtsx_usb_sdmmc mmc_core rtsx_usb i915 i2c_algo_bit drm_kms_helper drm crc32c_intel alx mdio wmi video i2c_dev
[    6.295280] CPU: 2 PID: 290 Comm: kworker/2:2 Tainted: P           OE    4.15.3-300.fc27.x86_64 #1
[    6.295281] Hardware name: MICRO-STAR INTERNATIONAL CO., LTD MS-AE6711/MS-AE671, BIOS EAE67IMS V8.2 07/31/2014
[    6.295286] Workqueue: events legacy_dvb_usb_read_remote_control [dvb_usb]
[    6.295290] RIP: 0010:tbs5520_rc_query+0xcc/0x107 [dvb_usb_tbs5520]
[    6.295291] RSP: 0018:ffffbe25c114be10 EFLAGS: 00010287
[    6.295293] RAX: ffffffffc17c3148 RBX: 000000000000002a RCX: ffffffffc17c3290
[    6.295294] RDX: 0000000000000029 RSI: ffff96e02faa5040 RDI: ffff96e00f95ced0
[    6.295296] RBP: ffffbe25c114be60 R08: 0000000000025040 R09: ffffffffc17c10e7
[    6.295297] R10: fffffc3d888a5e40 R11: 000000000000034a R12: ffffbe25c114be80
[    6.295298] R13: ffffffffc17c3140 R14: af847173cc9cb742 R15: ffffbe25c114be84
[    6.295300] FS:  0000000000000000(0000) GS:ffff96e02fa80000(0000) knlGS:0000000000000000
[    6.295302] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[    6.295304] CR2: 000055ef29c3fdf0 CR3: 00000001a120a004 CR4: 00000000001606e0
[    6.295305] Call Trace:
[    6.295312]  ? legacy_dvb_usb_read_remote_control+0x42/0x110 [dvb_usb]
[    6.295315]  ? tbs5520_i2c_transfer+0x230/0x230 [dvb_usb_tbs5520]
[    6.295317]  legacy_dvb_usb_read_remote_control+0x42/0x110 [dvb_usb]
[    6.295322]  process_one_work+0x187/0x3b0
[    6.295325]  worker_thread+0x2e/0x380
[    6.295328]  ? process_one_work+0x3b0/0x3b0
[    6.295331]  kthread+0x113/0x130
[    6.295333]  ? kthread_create_worker_on_cpu+0x70/0x70
[    6.295336]  ? do_syscall_64+0x75/0x180
[    6.295340]  ? SyS_exit_group+0x10/0x10
[    6.295343]  ret_from_fork+0x35/0x40
[    6.295345] Code: 5d 41 5e 41 5f 5d 49 8d 62 f8 c3 85 db 7e d6 48 8b 45 b8 41 0f b6 4d 00 38 48 01 74 2c 49 8d 45 08 8d 53 ff 48 8d 0c d0 48 39 c8 <41> c7 06 00 00 00 00 74 b1 48 8b 55 b8 49 89 c5 48 83 c0 08 41 
[    6.295372] RIP: tbs5520_rc_query+0xcc/0x107 [dvb_usb_tbs5520] RSP: ffffbe25c114be10
[    6.295384] ---[ end trace 4c6aba4ad139ffe1 ]---
```

crazycat69, 2018-02-19:
options dvb-usb disable_rc_polling=0

Tnix76, 2018-02-20:
Now the errors are:

```
feb 20 16:31:30 marine06 kernel: usb 3-5: new high-speed USB device number 14 using xhci_hcd
feb 20 16:31:31 marine06 kernel: usb 3-5: New USB device found, idVendor=734c, idProduct=5520
feb 20 16:31:31 marine06 kernel: usb 3-5: New USB device strings: Mfr=1, Product=2, SerialNumber=0
feb 20 16:31:31 marine06 kernel: usb 3-5: Product: TBS 5520
feb 20 16:31:31 marine06 kernel: usb 3-5: Manufacturer: TBS-Tech
feb 20 16:31:31 marine06 kernel: dvb-usb: found a 'TBS 5520 USB2.0' in cold state, will try to load a firmware
feb 20 16:31:31 marine06 kernel: dvb-usb: downloading firmware from file 'dvb-usb-tbsqbox-id5520.fw'
feb 20 16:31:31 marine06 kernel: tbs5520: start downloading tbs5520 firmware
feb 20 16:31:31 marine06 kernel: tbs5520: end downloading tbs5520 firmware
feb 20 16:31:31 marine06 kernel: dvb-usb: found a 'TBS 5520 USB2.0' in warm state.
feb 20 16:31:31 marine06 kernel: dvb-usb: will pass the complete MPEG2 transport stream to the software demuxer.
feb 20 16:31:31 marine06 kernel: dvbdev: DVB: registering new adapter (TBS 5520 USB2.0)
feb 20 16:31:31 marine06 kernel: dvb-usb: MAC address: 00:00:00:00:00:00
feb 20 16:31:31 marine06 kernel: i2c i2c-8: avl6882: found id=0xb family_id=0x68624955
feb 20 16:31:31 marine06 kernel: i2c i2c-8: r848: Rafael R848 successfully attached
feb 20 16:31:31 marine06 kernel: tbs5520: TBS5520: frontend attached
feb 20 16:31:31 marine06 kernel: dvb-usb: no frontend was attached by 'TBS 5520 USB2.0'
feb 20 16:31:31 marine06 kernel: dvb-usb: TBS 5520 USB2.0 successfully initialized and connected.
feb 20 16:31:31 marine06 mtp-probe[5810]: checking bus 3, device 14: "/sys/devices/pci0000:00/0000:00:14.0/usb3/3-5"
feb 20 16:31:31 marine06 mtp-probe[5810]: bus: 3, device: 14 was not an MTP device

```
Under /dev/dvb/adapter0/:

```
/dev/dvb/adapter0 $ ls -lt
total 0
crw-rw----+ 1 root video 212, 1 feb 20 16:31 dvr0
crw-rw----+ 1 root video 212, 2 feb 20 16:31 net0
crw-rw----+ 1 root video 212, 0 feb 20 16:31 demux0

```

Tnix76, 2018-03-09:
Hi again,
any clue or something that I could do to solve this?

Thanks.