tbsdtv/linux_media #111: 6984 and 6908 not compatible with current ArchLinux (4.14) with AMD Ryzen

wdirksen, 2018-02-24:
Hi All,

Both 6984 and 6908 were working perfectly on ArchLinux (up to 4.15) on previous Intel i5-2660 + P68 chipset. I am replacing the system with AMD Ryzen 1600x. I have tried both X370 and B350 chipset motherboards but not working with driver and PCI errors. I have played extensively with BIOS settings, eg Gen settings and turning off all unused hardware. Thoughts?

dmesg [code]
[    6.012634] stv0910: found STV0910 id=0x51
[    6.048625] TBSECP3 driver 0000:05:00.0: DVB: registering adapter 0 frontend 0 (TurboSight TBS 6908 DVB-S/S2 )...
[    6.048699] dvbdev: DVB: registering new adapter (TBSECP3 DVB Adapter)
[    6.073873] pps pps0: new PPS source ptp0
[    6.073876] igb 0000:03:00.0: added PHC on eth0
[    6.073877] igb 0000:03:00.0: Intel(R) Gigabit Ethernet Network Connection
[    6.073879] igb 0000:03:00.0: eth0: (PCIe:2.5Gb/s:Width x1) b0:6e:bf:c2:7b:7d
[    6.073881] igb 0000:03:00.0: eth0: PBA No: FFFFFF-0FF
[    6.073883] igb 0000:03:00.0: Using MSI-X interrupts. 2 rx queue(s), 2 tx queue(s)
[    6.075089] igb 0000:03:00.0 enp3s0: renamed from eth0
[    6.151336] TBSECP3 driver 0000:05:00.0: MAC address 00:22:ab:90:cb:79
[    6.151350] TBSECP3 driver 0000:05:00.0: DVB: registering adapter 2 frontend 0 (TurboSight TBS 6908 DVB-S/S2 )...
[    6.151408] dvbdev: DVB: registering new adapter (TBSECP3 DVB Adapter)
[    6.271315] TBSECP3 driver 0000:05:00.0: MAC address 00:22:ab:90:cb:7a
[    6.272108] stv0910: found STV0910 id=0x51
[    6.284080] SGI XFS with ACLs, security attributes, realtime, no debug enabled
[    6.285567] XFS (sdb1): Mounting V5 Filesystem
[    6.302278] TBSECP3 driver 0000:05:00.0: DVB: registering adapter 3 frontend 0 (TurboSight TBS 6908 DVB-S/S2 )...
[    6.302337] dvbdev: DVB: registering new adapter (TBSECP3 DVB Adapter)
[    6.421321] TBSECP3 driver 0000:05:00.0: MAC address 00:22:ab:90:cb:7b
[    6.421336] TBSECP3 driver 0000:05:00.0: DVB: registering adapter 4 frontend 0 (TurboSight TBS 6908 DVB-S/S2 )...
[    6.421399] TBSECP3 driver 0000:05:00.0: TurboSight TBS 6908 DVB-S/S2 : PCI 0000:05:00.0, IRQ 67, MMIO 0xf7300000
[    6.483279] XFS (sdb1): Ending clean mount
[    6.484780] input: HDA NVidia HDMI/DP,pcm=3 as /devices/pci0000:00/0000:00:03.1/0000:08:00.1/sound/card1/input7
[    6.485227] input: HDA NVidia HDMI/DP,pcm=7 as /devices/pci0000:00/0000:00:03.1/0000:08:00.1/sound/card1/input8
[    7.827281] ZFS: Loaded module v0.7.6-1, ZFS pool version 5000, ZFS filesystem version 5
[    9.867013] cx24117 9-0055: creating new instance
[    9.867017] i2c i2c-9: cx24117: Attaching frontend 0
[    9.878213] SAA716x Budget 0000:07:00.0: TurboSight TBS 6984  frontend 0 doesn't seem to have a isl6422b on the i2c bus.
[    9.878519] SAA716x Budget 0000:07:00.0: TurboSight TBS 6984  MAC=00:22:ab:90:16:20
[    9.878521] SAA716x Budget 0000:07:00.0: DVB: registering adapter 1 frontend 0 (TurboSight TBS 6984 DVB-S/S2)...
[    9.878652] dvbdev: DVB: registering new adapter (SAA716x dvb adapter)
[    9.878823] cx24117 9-0055: attaching existing instance
[    9.878825] i2c i2c-9: cx24117: Attaching frontend 1
[    9.886139] SAA716x Budget 0000:07:00.0: TurboSight TBS 6984  frontend 1 doesn't seem to have a isl6422b on the i2c bus.
[    9.886442] SAA716x Budget 0000:07:00.0: TurboSight TBS 6984  MAC=00:22:ab:90:16:21
[    9.886444] SAA716x Budget 0000:07:00.0: DVB: registering adapter 5 frontend 0 (TurboSight TBS 6984 DVB-S/S2)...
[    9.886537] dvbdev: DVB: registering new adapter (SAA716x dvb adapter)
[    9.886682] cx24117 8-0005: creating new instance
[    9.886684] i2c i2c-8: cx24117: Attaching frontend 0
[    9.893999] SAA716x Budget 0000:07:00.0: TurboSight TBS 6984  frontend 2 doesn't seem to have a isl6422b on the i2c bus.
[    9.894305] SAA716x Budget 0000:07:00.0: TurboSight TBS 6984  MAC=00:22:ab:90:16:22
[    9.894307] SAA716x Budget 0000:07:00.0: DVB: registering adapter 6 frontend 0 (TurboSight TBS 6984 DVB-S/S2)...
[    9.894394] dvbdev: DVB: registering new adapter (SAA716x dvb adapter)
[    9.894532] cx24117 8-0005: attaching existing instance
[    9.894534] i2c i2c-8: cx24117: Attaching frontend 1
[    9.901851] SAA716x Budget 0000:07:00.0: TurboSight TBS 6984  frontend 3 doesn't seem to have a isl6422b on the i2c bus.
[    9.902156] SAA716x Budget 0000:07:00.0: TurboSight TBS 6984  MAC=00:22:ab:90:16:23
[    9.902158] SAA716x Budget 0000:07:00.0: DVB: registering adapter 7 frontend 0 (TurboSight TBS 6984 DVB-S/S2)...
[   12.305838] IPv6: ADDRCONF(NETDEV_UP): enp3s0: link is not ready
[   12.331848] IPv6: ADDRCONF(NETDEV_UP): enp3s0: link is not ready
[   12.799346] resource sanity check: requesting [mem 0x000c0000-0x000fffff], which spans more than PCI Bus 0000:00 [mem 0x000c0000-0x000dffff window]
[   12.799465] caller _nv001170rm+0xe3/0x1d0 [nvidia] mapping multiple BARs
[   12.826941] pcieport 0000:00:03.1: AER: Corrected error received: id=0000
[   12.826947] pcieport 0000:00:03.1: PCIe Bus Error: severity=Corrected, type=Data Link Layer, id=0019(Transmitter ID)
[   12.826949] pcieport 0000:00:03.1:   device [1022:1453] error status/mask=00001000/00006000
[   12.826951] pcieport 0000:00:03.1:    [12] Replay Timer Timeout  
[   13.244930] pcieport 0000:00:03.1: AER: Multiple Corrected error received: id=0000
[   13.244939] pcieport 0000:00:03.1: PCIe Bus Error: severity=Corrected, type=Data Link Layer, id=0019(Transmitter ID)
[   13.244941] pcieport 0000:00:03.1:   device [1022:1453] error status/mask=00001000/00006000
[   13.244944] pcieport 0000:00:03.1:    [12] Replay Timer Timeout  
[   13.347487] nvidia-modeset: Allocated GPU:0 (GPU-815560b7-5f23-cc41-6b9c-e044bcec283e) @ PCI:0000:08:00.0
[   14.237161] fuse init (API version 7.26)
[   15.333930] igb 0000:03:00.0 enp3s0: igb: enp3s0 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX/TX [/code]

There must others out there using AMD Ryzen with these drivers...

crazycat69, 2018-02-24:
Just another new buggy AMD CPU and chipset :) Or buggy kernel support :) Anyway AMD is evil :)

CvH, 2018-03-09:
https://github.com/b-rad-NDi/Ubuntu-media-tree-kernel-builder/commit/1553c63e61f4a4d44f484bb1a3f2ad9baa08805c
^^^does this maybe fix/helps here ?

wdirksen, 2018-03-09:
Thanks @CvH. I will look into this. Unfortunately PC completely switched back to old Intel mobo due to this issue. Will try again

closed 2018-03-14 by crazycat69