tbsdtv/linux_media #112: TBS 6981 not working even its listed as supported

lucky79, 2018-02-26:
Hello,
I need to operate TBS6981 and TBS6909 in a single machine. Currently the 6981 card is not recognized, but 6909 works fine. 6981 is listed as supported by this driver, but in reality its not.

Thanks

dirk362, 2018-02-26:
I have just compiled this on a rebuild of my machine, which has 2 x 6981 and 2 x 6982. And it worked, but on my 2nd attempt.
In my first attempt I forgot 2 things.
1. Clear out old modules. i.e. 
`sudo rm -rf /lib/modules/$(uname -r)/kernel/drivers/media/*`

2. Make sure firmware is installed (as per wiki entry right at the bottom) - see https://github.com/tbsdtv/linux_media/wiki/CX24117-firmware

The 6981 is definitely supported and works.

lucky79, 2018-02-26:
Hmm, that's very strange, I have just compiled the drivers, 6909 works, but 6981 not... I have cleared old modules, firmware is in place. 

I am on debian 8.10 with kernel 3.16.0-5-amd64

if I try to load the cx23885 module manually, I am getting following errors:
modprobe: ERROR: could not insert 'cx23885': Unknown symbol in module, or unknown parameter (see dmesg)
dmesg:
rc_core: Unknown symbol nsecs_to_jiffies (err 0)

Will test on another machine

crazycat69, 2018-02-26:
Look like some backport issues in rc_core. Maybe need kernels like 4.4+

dirk362, 2018-02-26:
I too am using Debian, but Stable (Stretch). This uses a 4.9.x kernel - currently it's on linux-image-4.9.0-6-amd64 (4.9.82-1+deb9u2).
Suggests @crazycat69 comment is material difference and likely cause.
I've proven it on Debain Testing as well, and that works fine - but of course that is an even higher kernel release.

lucky79, 2018-02-26:
Ok guys, installed backports kernel 4.9 and it works fine now. Thank you very much!

closed 2018-02-26 by lucky79