tbsdtv/linux_media #115: videobuf2_memops Unknown symbol on Linux Kernel 4.15

j4m35, 2018-03-16:
Hi,

I have a TBS 6205 and recently tried upgrading to Linux Kernel 4.15.

When I build linux_media against Linux Kernel 4.15.10, I says the following 4 symbols are missing:
```
[    5.334401] Linux video capture interface: v2.00
[    5.334402] WARNING: You are using an experimental version of the media stack                                                                             .
                As the driver is backported to an older kernel, it doesn't offer
                enough quality for its usage in production.
                Use it with care.
               Latest git patches (needed if you report a bug to linux-media@vge                                                                             r.kernel.org):
                c038387e3827f3ebf97d07086da8dfecde19f0d5 add support for tbs5590
[    5.335843] videobuf2_memops: Unknown symbol put_vaddr_frames (err 0)
[    5.335846] videobuf2_memops: Unknown symbol get_vaddr_frames (err 0)
[    5.335847] videobuf2_memops: Unknown symbol frame_vector_destroy (err 0)
[    5.335849] videobuf2_memops: Unknown symbol frame_vector_create (err 0)
```

I have traced them to belong in the frame_vector.c file.

My Kernel Config does not have FRAME_VECTOR enabled as I assumed it would have been built by the V4L package.

I would appreciate any help in this matter.

j4m35, 2018-03-16:
I've just compiled it against my older 4.14 kernel and it is getting the same Unknown symbol errors.

I think in the past it wasn't using videobuf2? And now it does?

j4m35, 2018-03-16:
Just a consideration, but can media_build be patched to build the frame_vector.ko module?

For example: https://git.linuxtv.org/media_build.git/commit/?id=7d84743d0e17274bbe28249e1ec7aad95c1efa39

crazycat69, 2018-03-16:
FRAME_VECTOR must be enabled for 4.3+ kernels.
Current media_build compile own frame_vector from backport patches for kernels < 4.3

j4m35, 2018-03-17:
As CONFIG_FRAME_VECTOR is not manually selectable, what's the other CONFIG settings I need to turn on to get FRAME_VECTOR?

crazycat69, 2018-03-17:
You have some custom kernel build with disabled FRAME_VECTOR. I don't know what Linux distrib or some custom build like Open/LibreELEC.

j4m35, 2018-03-17:
I'm running a custom kernel on Gentoo. So I've configured only the minimum options for my hardware. I assumed anything for v4l would be compiled by this package.

belliash, 2018-03-19:
+1... What is needed in kernel to enable frame vector?

j4m35, 2018-03-24:
I felt that it is cleaner to keep my custom kernel as is and have the linux_media package build the frame_vector.ko.

To that extent, this command executed in the linux_media directory will adjust the appropriate Makefile and force the package to build the frame_vector.ko module:

`sed -i '/^ifeq.*frame_vector.*/,/^endif/ s/^/#/' v4l/Makefile`

It's probably better if the Makefile is pathced to check the current kernel's config and build the frame_vector.ko module if it is not activated in the kernel.

closed 2020-08-19 by j4m35