tbsdtv/linux_media #118: TBS5520 Tuner attach broken

ShinjiLE, 2018-03-21:
with commit 7f0a95ae6adca8758e5682f111e0c98e78dce6bd the correct identification of the frontend is not possible . It look like TBS5520_WRITE_MSG at drivers/media/usb/dvb-usb/tbs5520.c:94 is accidentally replaced by TBS5520_RC_QUERY .

```
Mar 21 13:33:07 apu kernel: i2c i2c-5: avl6882: attach failed family id mismatch
Mar 21 13:33:07 apu kernel: TBS5520: frontend attach failed
Mar 21 13:33:07 apu kernel: dvb-usb: no frontend was attached by 'TurboSight TBS 5520 DVB-S/S2/T/T2/C/ISDB-T'
```

crazycat69, 2018-03-21:
Sorry :)
37df5ec9c121d1aeb4d4a841c4a50663f1a927eb

ShinjiLE, 2018-03-22:
Solved . Thank you.

closed 2018-03-22 by ShinjiLE

alexiedb, 2023-03-19:
Hello,

For:
Linux raspberrypi 4.19.97-v7l+ #1294 SMP Thu Jan 30 13:21:14 GMT 2020 armv7l GNU/Linux
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian

and using TBS5520 I get the following error for the first time, after a power on:
"
.....
[    4.759106] tbs5520: end downloading tbs5520 firmware
[    4.759140] dvb-usb: found a 'TurboSight TBS 5520 DVB-S/S2/T/T2/C/ISDB-T' in warm state.
[    4.759532] dvb-usb: will pass the complete MPEG2 transport stream to the software demuxer.
[    4.765874] dvbdev: DVB: registering new adapter (TurboSight TBS 5520 DVB-S/S2/T/T2/C/ISDB-T)
[    4.765887] dvb-usb: MAC address: 00:00:00:00:00:00
[    4.766092] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    4.797217] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Feb 27 2018 03:15:32 version 7.45.154 (r684107 CY) FWID 01-4fbe0b04
[    4.803142] **i2c i2c-9: avl6882: attach failed family id mismatch**
[    4.803170] TBS5520: frontend attach failed
[    4.803182] dvb-usb: no frontend was attached by 'TurboSight TBS 5520 DVB-S/S2/T/T2/C/ISDB-T'
[    4.803200] dvb-usb: could not initialize remote control.
[    4.803210] dvb-usb: TurboSight TBS 5520 DVB-S/S2/T/T2/C/ISDB-T successfully initialized and connected.
[    4.803877] usbcore: registered new interface driver tbs5520
[    4.806623] dvb-usb: TurboSight TBS 5520 DVB-S/S2/T/T2/C/ISDB-T successfully deinitialized and disconnected.
"
and 2s after that I get:
"
[    6.672622] dvb-usb: found a 'TurboSight TBS 5520 DVB-S/S2/T/T2/C/ISDB-T' in cold state, will try to load a firmware
[    6.672738] dvb-usb: downloading firmware from file 'dvb-usb-tbs5520-01.fw'
[    6.672748] tbs5520: start downloading tbs5520 firmware
[    6.754347] 8021q: 802.1Q VLAN Support v1.8
[    6.818400] tbs5520: end downloading tbs5520 firmware
[    6.818435] dvb-usb: found a 'TurboSight TBS 5520 DVB-S/S2/T/T2/C/ISDB-T' in warm state.
[    6.818950] dvb-usb: will pass the complete MPEG2 transport stream to the software demuxer.
[    6.820614] dvbdev: DVB: registering new adapter (TurboSight TBS 5520 DVB-S/S2/T/T2/C/ISDB-T)
[    6.820627] dvb-usb: MAC address: 00:00:00:00:00:00
[    6.834011] i2c i2c-9: avl6882: found id=0xb family_id=0x68624955
[    6.913433] i2c i2c-9: r848: Rafael R848 successfully attached
[    6.913445] tbs5520: TBS5520: frontend attached
[    6.914406] usb 1-1.1.3: DVB: registering adapter 0 frontend 0 (TurboSight TBS 5520 DVB-S/S2/T/T2/C/ISDB-T)...
[    6.918346] dvb-usb: could not initialize remote control.
[    6.918359] dvb-usb: TurboSight TBS 5520 DVB-S/S2/T/T2/C/ISDB-T successfully initialized and connected.
"
Any idea why what is going wrong for the first time?