tbsdtv/linux_media #120: No capture devices working after install

ljones0, 2018-03-25:
Hello! I'm using a debian buster system here running kernel 4.14.0-3-amd64. But my problem is that after I compile the driver (I used the instructions at https://github.com/tbsdtv/linux_media/wiki ) although compiling seems to go ok with no major errors after make install and rebooting *none* of my capture devices work. 

I have two capture devices here - one is a technisat USB HD (DVBS2) device (works fine before, after install not working). The other device is a TBS technologies DVB tuner  - from memory it is a  TBS 6281 SE (which is why I'm trying to install this driver although lspci -vv lists it as device 6281:0002 and as a DVB-S2 tuner - it isn't it is a DVB-T2 tuner - !). Though it dosen't seem to get detected properly after compiling and make install. Here's an example:

[code]
[    3.325769] tbsecp3: disagrees about version of symbol dvb_dmxdev_init
[    3.325771] tbsecp3: Unknown symbol dvb_dmxdev_init (err -22)
[    3.325784] tbsecp3: disagrees about version of symbol dvb_ca_en50221_init
[    3.325785] tbsecp3: Unknown symbol dvb_ca_en50221_init (err -22)
[    3.325807] tbsecp3: disagrees about version of symbol dvb_register_adapter
[    3.325808] tbsecp3: Unknown symbol dvb_register_adapter (err -22)
[/code]

Also while compiling once I got to the "make allyesconfig" section I got this message:

***WARNING:*** You do not have the full kernel sources installed.
This does not prevent you from building the v4l-dvb tree if you have the
kernel headers, but the full kernel source may be required in order to use
make menuconfig / xconfig / qconfig.

Yet I do have the kernel source and headers installed;

[code]
$ ls /usr/src
linux-config-4.14             linux-headers-4.14.0-3-common  linux-source-4.14.tar.xz
linux-headers-4.14.0-3-amd64  linux-patch-4.14-rt.patch.xz
[/code]

Also when running "make allyesconfig" there was another message saying

Preparing to compile for kernel version 4.14.0

...but that dosen't match my kernel version; 

[code]
$ uname -r
4.14.0-3-amd64
[/code]

Also what is the difference between this driver (linux_media) and the one here - https://github.com/tbsdtv/media_build ("media_build"). Are both needed?

thanks
ljones

crazycat69, 2018-03-26:
You need remove media driver modules before install. See install.sh script in media_build.

ljones0, 2018-03-27:
Many thanks - that worked by running install.sh there. Maybe its worth putting that piece of info into the wiki btw?

crazycat69, 2018-03-27:
See Wiki FAQ section 5.
Anyway for end-users better use stable TBS Open Source Linux Driver Offline Package from https://www.tbsiptv.com/downloads
This contain media_build + media_tree + firmwares.