tbsdtv/linux_media #122: Build error ov772x.c

nightshadow36, 2018-04-13:
@crazycat69 

Hi you have helped before so hoping you can help again.
I have tried installing my tbs 6909 cards on the latest ubuntu 14.04 kernel.
I am getting the errors below and dont know what to do?

I have tried to install two ways.
https://github.com/tbsdtv/linux_media/wiki

And also 
git clone https://github.com/tbsdtv/media_build.git
cd media_build
./build --main-git
make install

/home/cyberstreamer-1/media_build/v4l/ov772x.c: In function 'ov772x_power_on':
/home/cyberstreamer-1/media_build/v4l/ov772x.c:827:2: error: implicit declaration of function 'gpiod_get_optional' [-Werror=implicit-function-declaration]
  priv->rstb_gpio = gpiod_get_optional(&client->dev, "rstb",
  ^
/home/cyberstreamer-1/media_build/v4l/ov772x.c:828:11: error: 'GPIOD_OUT_LOW' undeclared (first use in this function)
           GPIOD_OUT_LOW);
           ^
/home/cyberstreamer-1/media_build/v4l/ov772x.c:828:11: note: each undeclared identifier is reported only once for each function it appears in
/home/cyberstreamer-1/media_build/v4l/ov772x.c: In function 'ov772x_probe':
/home/cyberstreamer-1/media_build/v4l/ov772x.c:1292:11: error: 'GPIOD_OUT_LOW' undeclared (first use in this function)
           GPIOD_OUT_LOW);
           ^
cc1: some warnings being treated as errors
make[3]: *** [/home/cyberstreamer-1/media_build/v4l/ov772x.o] Error 1
make[3]: *** Waiting for unfinished jobs....
make[2]: *** [_module_/home/cyberstreamer-1/media_build/v4l] Error 2
make[2]: Leaving directory `/usr/src/linux-headers-3.13.0-145-generic'
make[1]: *** [default] Error 2
make[1]: Leaving directory `/home/cyberstreamer-1/media_build/v4l'
make: *** [all] Error 2
build failed at ./build line 547
root@cyberstreamer-1:~/media_build#

CvH, 2018-04-13:
your kernel is really old, try to upgrade to 4.4.x (Ubuntu 14.04.5 includes that) and try again

nightshadow36, 2018-04-13:
@CvH 
Thank you. 
I have already run a update on ubuntu and it gave me 3.13.0-135
When i run update again it shows as no more updates. 
I have had to use the kernel before this one as i get the errors as shown above.
Something about ov772x.o

crazycat69, 2018-04-13:
[media_build](https://github.com/tbsdtv/media_build)  and this repo updated, compiled without errors on 3.13 kernel (Kubuntu 14.04). Must be compatible with kernels up to 4.16

nightshadow36, 2018-04-13:
@crazycat69 

Can you tell me what I am doing wrong?
This is the commands I am using, this gives me the errors from above.
I install the updates and drivers maybe every 6 / 8 months, and forget.

Ubuntu 14.04.6 LTS 3.13.0-135

rm -rf /lib/modules/$(uname -r)/extra
rm -rf /lib/modules/$(uname -r)/kernel/drivers/media
rm -rf /lib/modules/$(uname -r)/kernel/drivers/staging/media

rm -r media
rm -r media_build

git clone https://github.com/tbsdtv/media_build.git
cd media_build
make distclean
./build --main-git
make install

Thank you.

nightshadow36, 2018-04-15:
Are there other commands I can use to make this install, been trying a few days now.
I know I am doing something wrong I just do not know what.
Thank you.

crazycat69, 2018-04-15:
This not your mistake :) media_build was updated and fixed some backport patches for old kernels.

nightshadow36, 2018-04-15:
@crazycat69 
Updated does that mean i just run the commands below to install now.
Or should I be using diffrent commands? Thank you.

git clone https://github.com/tbsdtv/media_build.git
cd media_build
make distclean
./build --main-git
make install

crazycat69, 2018-04-15:
All ok :)

nightshadow36, 2018-04-16:
@crazycat69 
HI again. Sorry to bother you but i am getting some more errors.
/home/cyberstreamer-1/media_build/v4l/mt9t112.c: In function 'mt9t112_probe':
/home/cyberstreamer-1/media_build/v4l/mt9t112.c:1097:2: error: implicit declaration of function 'devm_gpiod_get_optional' [-Werror=implicit-function-declaration]
  priv->standby_gpio = devm_gpiod_get_optional(&client->dev, "standby",
  ^
/home/cyberstreamer-1/media_build/v4l/mt9t112.c:1098:12: error: 'GPIOD_OUT_HIGH' undeclared (first use in this function)
            GPIOD_OUT_HIGH);
            ^
/home/cyberstreamer-1/media_build/v4l/mt9t112.c:1098:12: note: each undeclared identifier is reported only once for each function it appears in
/home/cyberstreamer-1/media_build/v4l/mt9t112.c: At top level:
cc1: warning: unrecognized command line option "-Wno-format-truncation" [enabled by default]
cc1: some warnings being treated as errors
make[3]: *** [/home/cyberstreamer-1/media_build/v4l/mt9t112.o] Error 1
make[3]: *** Waiting for unfinished jobs....
make[2]: *** [_module_/home/cyberstreamer-1/media_build/v4l] Error 2
make[2]: Leaving directory `/usr/src/linux-headers-3.13.0-145-generic'
make[1]: *** [default] Error 2
make[1]: Leaving directory `/home/cyberstreamer-1/media_build/v4l'
make: *** [all] Error 2
build failed at ./build line 547
root@cyberstreamer-1:~/media_build#

Davin622, 2018-04-17:
Hi 
Maybe you can try disable the module,  using the command "make menuconfig" ,find the module, disable it.

tbsiptv, 2018-04-18:
Hi You can send email to our support we can remote install the driver for you

nightshadow36, 2018-05-02:
This is the fix
#wget http://www.tbsdtv.com/download/document/linux/media_build-2018-0206.tar.bz2
# sudo rm -rf /lib/modules/"kernel name"/kernel/drivers/media/
#tar jxvf media_build-2018-0206.tar.bz2
#cd media_build 
#./install.sh 
#reboot