tbsdtv/linux_media #124: TBS6301 and FireTV box

treki1, 2018-04-16:
I receive only stereo audio.
When i set Dolby Digital on the FireTV, the sound makes like "pfrrrrrrrrrrrrrrr....".   :D
Is that a problem with HDCP crypt?

tbslucy, 2018-04-17:
tbs6301 does not support Dolby Digital audio.

tbslucy, 2018-04-17:
tbs6301 does not support Dolby Digital audio.

Lucy Lu | Software Engineerlucy@tbsdtv.com | www.tbsdtv.com   TBS TechnologyTHE BEST SOLUTIONS FOR DIGITAL TV AND MORELive TV Streaming | IPTV Head-end | OEM & ODM Solution---------------------------------------------------------------------------------

 From: treki1Date: 2018-04-17 01:53To: tbsdtv/linux_mediaCC: SubscribedSubject: [tbsdtv/linux_media] TBS6301 and FireTV box (#124)I receive only stereo audio.

When i set Dolby Digital on the FireTV, the sound makes like "pfrrrrrrrrrrrrrrr....".   :D

Is that a problem with HDCP crypt?