tbsdtv/linux_media #126: TBS6908

dastious, 2018-04-23:
I compile with success like the wiki said for TBS6908 on debian stretch
firmware is installed, but the card don't show. 
Tried with kernel 4.9, 4.9-rt, 4.15 backport

no error on dmesg, any idea ?

crazycat69, 2018-04-23:
this card not require any firmware files. check detected pci/pci-e devices with lspci

dastious, 2018-04-24:
yes there is multimedia controller: tbs technologies dvb-s2 4 tuner pci card in lspci

in dmesg i also have 
[    6.674623] media: loading out-of-tree module taints kernel.
[    6.675705] media: Linux media interface: v0.10
[    6.689888] Linux video capture interface: v2.00
[    6.689890] WARNING: You are using an experimental version of the media stack.
                As the driver is backported to an older kernel, it doesn't offer     
                enough quality for its usage in production.                          
                Use it with care.                                                    
               Latest git patches (needed if you report a bug to linux-media@vger.kernel.org):                                                                            
                efe31531b77efd3a4c94516504a5823d31cdc776 change 720p60 to 720i60 signal on tbs6312    

but nothing else, no dmesg | grep frontend or anything I don't know what to do

crazycat69, 2018-04-24:
modinfo tbsecp3modprobe tbsecp3dmesg | tail -n20

dastious, 2018-04-24:
modinfo tbsecp3:

filename:       /lib/modules/4.15.0-0.bpo.2-amd64/kernel/drivers/media/pci/tbsecp3/tbsecp3.ko
license:        GPL
description:    TBS ECP3 driver
author:         Luis Alves <ljalvs@gmail.com>
alias:          pci:v0000544Dd00006178sv00006304sd*bc*sc*i*
alias:          pci:v0000544Dd00006178sv00006301sd*bc*sc*i*                                                                                                                 
alias:          pci:v0000544Dd00006178sv0000690Asd*bc*sc*i*                                                                                                                 
alias:          pci:v0000544Dd00006178sv00006514sd*bc*sc*i*                                                                                                                 
alias:          pci:v0000544Dd00006178sv00006814sd*bc*sc*i*                                                                                                                 
alias:          pci:v0000544Dd00006178sv00006209sd*bc*sc*i*                                                                                                                 
alias:          pci:v0000544Dd00006178sv00006704sd*bc*sc*i*                                                                                                                 
alias:          pci:v0000544Dd00006178sv00006281sd*bc*sc*i*                                                                                                                 
alias:          pci:v0000544Dd00006178sv00006290sd*bc*sc*i*                                                                                                                 
alias:          pci:v0000544Dd00006178sv00006590sd*bc*sc*i*                                                                                                                 
alias:          pci:v0000544Dd00006178sv00006528sd*bc*sc*i*                                                                                                                 
alias:          pci:v0000544Dd00006178sv00006910sd*bc*sc*i*                                                                                                                 
alias:          pci:v0000544Dd00006178sv00006909sd*bc*sc*i*                                                                                                                 
alias:          pci:v0000544Dd00006178sv00006908sd*bc*sc*i*                                                                                                                 
alias:          pci:v0000544Dd00006178sv00006905sd*bc*sc*i*                                                                                                                 
alias:          pci:v0000544Dd00006178sv00006904sd*bc*sc*i*                                                                                                                 
alias:          pci:v0000544Dd00006178sv00006903sd*bc*sc*i*                                                                                                                 
alias:          pci:v0000544Dd00006178sv00006902sd*bc*sc*i*                                                                                                                 
alias:          pci:v0000544Dd00006178sv00006522sd*bc*sc*i*                                                                                                                 
alias:          pci:v0000544Dd00006178sv00006205sd*bc*sc*i*                                                                                                                 
depends:        dvb-core,tas2101                                                                                                                                            
retpoline:      Y                                                                                                                                                           
name:           tbsecp3                                                                                                                                                     
vermagic:       4.15.0-0.bpo.2-amd64 SMP mod_unload modversions                                                                                                             
parm:           adapter_nr:DVB adapter numbers (array of short)                                                                                                             
parm:           dma_pkts:DMA buffer size in TS packets (16-256), default 128 (array of int)
parm:           enable_msi:use an msi interrupt if available (bool)

modprobe tbsecp3:
modprobe: ERROR: could not insert 'tbsecp3': Invalid argument

dmesg | tail -n20:
[    7.629563] coretemp coretemp.0: Using relative temperature scale!
[    7.636919] radeon 0000:01:03.0: fb0: radeondrmfb frame buffer device
[    7.637112] coretemp coretemp.0: Using relative temperature scale!
[    7.649965] iTCO_vendor_support: vendor-support=0
[    7.652333] [drm] Initialized radeon 2.50.0 20080528 for 0000:01:03.0 on minor 0
[    7.656263] intel_powerclamp: No package C-state available
[    7.662130] iTCO_wdt: Intel TCO WatchDog Timer Driver v1.11
[    7.662193] iTCO_wdt: unable to reset NO_REBOOT flag, device disabled by hardware/BIOS
[    8.108098] bnx2 0000:03:00.0: firmware: direct-loading firmware bnx2/bnx2-mips-06-6.2.3.fw
[    8.115571] bnx2 0000:03:00.0: firmware: direct-loading firmware bnx2/bnx2-rv2p-06-6.0.15.fw
[... cut ...]
[   10.780339] bnx2 0000:03:00.0 enp3s0: NIC Copper Link is Up, 1000 Mbps full duplex
[   10.780450] IPv6: ADDRCONF(NETDEV_CHANGE): enp3s0: link becomes ready
[10416.483793] perf: interrupt took too long (2512 > 2500), lowering kernel.perf_event_max_sample_rate to 79500
[14082.368261] dvb_core: disagrees about version of symbol vb2_vmalloc_memops
[14082.368264] dvb_core: Unknown symbol vb2_vmalloc_memops (err -22)
[14196.204204] dvb_core: disagrees about version of symbol vb2_vmalloc_memops
[14196.204208] dvb_core: Unknown symbol vb2_vmalloc_memops (err -22)

crazycat69, 2018-04-24:
need delete all installed media modules before install.
see [install.sh](https://github.com/tbsdtv/media_build/blob/master/install.sh) script

dastious, 2018-04-24:
The doc is not clear, because i didn't install anything before, so i taught I didnt need to delete all media modules.
Sorry for that, and Thanks a lot ! 
Btw : TBS6908 need a firmware.

closed 2018-04-24 by dastious