tbsdtv/linux_media #130: 6928 support

mppod, 2018-05-12:
Many thanks for you work.
I’d like to ask for support for 6928.
Boiler plate code from 6922 compiles but the card does not lock any channels.
CI not working.

I can code and test.

mlt260262, 2018-12-29:
I'm also interested in support for kernel 4.15.0, I could not activate my card after system update again.

Vispen74, 2020-04-09:
Nothing new on this? I can´t use the card at all in Ubuntu 18.04 with this open source drivers.

crazycat69, 2020-04-09:
This card not supported in open-source drivers. Many hw revisions for this card with different TS wiring and CI controler.  Same story with 6991.

schrieveslaach, 2021-01-09:
@crazycat69 thanks for that hint. Do you think that we have to stick with the closed source driver? Currently, I'm trying to compile the closed source driver with Arch Linux with different kernel versions (kind of hard).

chemloul, 2021-02-05:
Hi everyone,
I have still have TBS6928 which was running under Windows. I am now trying to make it work under linux. 
Lately, I have tried all the drivers of TBS (closed and open source) that I could find on internet.
Old drivers do not compile with new kernel version and open source one does not support the card.
Any help to make it work would be very appreciated.
P.S. I do not want to buy a new one to safeguard the environment as the card is still working under windows.
Many thanks.

zhang111chao, 2021-02-05:
closed driver need the version of kernel is under v4.10 .  
please add skype jack_2431 and setup teamviewer or ssh connection .i will check