tbsdtv/linux_media #133: TBS6814 Isdbt tunner not detected

AlbertEinsteinGlitchPoint, 2018-05-22:
Hi guys

i have compiled this source 100% ok with no erros, after reboot and loading firmware 1.0b

i can see all my 3 cards tbs6909 fine, but my tbs6814 not showing.

also another machine from another person, with 2 tbs6814 and 1 tbs6909

running and compiling same source,  TBS6909 recognised ok working fine after reboot, and 2 tunners tbs6814 not detected, is there any issue with the source code ? do i need to enable compile for tbs6814 drivers???????

Any help will be appreciated, thanks.

by the way the device tbs6814 is found on lspci -vvv  together with 6909 so i dont think its a hardware issue.

tbsiptv, 2019-03-01:
please use the latest source code and try again should be ok now

cooliobr, 2019-07-24:
last source is not working many problems with signal.
only this version is working 
[    7.591704] TBS version 160405
on dimes adaptor need appears equal this
[    8.373819] DVB: registering adapter 0 frontend 0 (TurboSight TBS6814 ISDB-T frontend)
not equal this
[    5.713728] TBSECP3 driver 0000:1f:00.0: DVB: registering adapter 0 frontend 0 (Turbosight TBS 6814 (Quad ISDB-T))