tbsdtv/linux_media #135: Error loading driver.

demontanaro, 2018-06-22:
uname -a: (debian with latest source kernel)

Linux Servidor 4.9.88 #2 SMP Fri Jun 22 05:19:31 CDT 2018 x86_64 GNU/Linux

DMESG:

[    4.103702] media: loading out-of-tree module taints kernel.
[    4.103705] media: loading out-of-tree module taints kernel.
[    4.105398] media: Linux media interface: v0.10
[    4.117372] Linux video capture interface: v2.00
[    4.117377] WARNING: You are using an experimental version of the media stack.
                As the driver is backported to an older kernel, it doesn't offer
                enough quality for its usage in production.
                Use it with care.
               Latest git patches (needed if you report a bug to linux-media@vger.kernel.org):
                b858ec4ff8eb704151477928d1489101264ddfda add mac reading to tbs6308
[    4.125877] dvb_core: disagrees about version of symbol vb2_vmalloc_memops
[    4.125882] dvb_core: Unknown symbol vb2_vmalloc_memops (err -22)
[    4.149095] dvb_core: disagrees about version of symbol vb2_vmalloc_memops
[    4.149100] dvb_core: Unknown symbol vb2_vmalloc_memops (err -22)
[    4.177432] dvb_core: disagrees about version of symbol vb2_vmalloc_memops
[    4.177437] dvb_core: Unknown symbol vb2_vmalloc_memops (err -22)
[    4.201609] dvb_core: disagrees about version of symbol vb2_vmalloc_memops
[    4.201615] dvb_core: Unknown symbol vb2_vmalloc_memops (err -22)

No errors in building, CONFIG_DVB_MAX_ADAPTERS set to 32 in Kernel.

The cards nos appear in system.

What happen?

crazycat69, 2018-06-22:
Delete kernel media modules before install new from media_build. see FAQ 5 in [wiki](https://github.com/tbsdtv/linux_media/wiki)

demontanaro, 2018-06-22:
Not work:

[18163.419007] videobuf2_core: disagrees about version of symbol v4l_vb2q_enable_media_source
[18163.419012] videobuf2_core: Unknown symbol v4l_vb2q_enable_media_source (err -22)

I delete media folder, reboot, install, reboot again and appear other error.

demontanaro, 2018-06-22:
I tested:

-this drivers (opensource) *not load*
-your drivers (bitbucket) *not load*
-Closed 1604xx TBS *load*

crazycat69, 2018-06-23:
So linux-tbs-drivers 1604xx installed before ?0:45, 23 июня 2018 г., demontanaro <notifications@github.com>:I tested:
-this drivers (opensource) not load
-your drivers (bitbucket) not load
-Closed 1604xx TBS load

demontanaro, 2018-06-23:
Yes. 

But in other test, i format the drive, install Debian 9.4 and the same, the driver not load.

Error:

videobuf2_core: disagrees about version of symbol v4l_vb2q_enable_media_source

crazycat69, 2018-06-24:
try remove meda driver in wrong place
**sudo rm -r -f /lib/modules/$(uname -r)/kernel/drivers/linux**

4integration, 2018-07-13:
I also have issues.
Fresh install of Ubuntu Server 16.04 LTS 

```
joacim@thor:~$ uname -r
4.4.0-116-generic
```

```
[    1.381245] usb 2-1: new high-speed USB device number 2 using ehci-pci
[    1.444749] usb 1-1: New USB device found, idVendor=8087, idProduct=0024
[    1.697626] usb 1-1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[    1.718414] usb 2-1: New USB device found, idVendor=8087, idProduct=0024
[    1.718415] usb 2-1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[    1.969406] usb 1-1.5: new high-speed USB device number 3 using ehci-pci
[    1.993285] usb 2-1.3: new high-speed USB device number 3 using ehci-pci
[    2.066032] usb 1-1.5: New USB device found, idVendor=734c, idProduct=5880
[    2.066034] usb 1-1.5: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[    2.066036] usb 1-1.5: Product: TBS 5880
[    2.066037] usb 1-1.5: Manufacturer: TBS-Tech
[    2.086205] usb 2-1.3: New USB device found, idVendor=0424, idProduct=2660
[    2.086207] usb 2-1.3: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[    2.137369] usb 1-1.6: new high-speed USB device number 4 using ehci-pci
[    2.234845] usb 1-1.6: New USB device found, idVendor=734c, idProduct=5880
[    2.234846] usb 1-1.6: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[    2.234848] usb 1-1.6: Product: TBS 5880
[    2.234849] usb 1-1.6: Manufacturer: TBS-Tech
```
....
```
[   15.228334] media: module verification failed: signature and/or required key missing - tainting kernel
[   15.228999] media: Linux media interface: v0.10
[   15.410438] videodev: Linux video capture interface: v2.00
[   15.410441] WARNING: You are using an experimental version of the media stack.
                As the driver is backported to an older kernel, it doesn't offer
                enough quality for its usage in production.
                Use it with care.
               Latest git patches (needed if you report a bug to linux-media@vger.kernel.org):
                ea28c4ecbb81c14312b9012b87cdb3913c6568ec change tbs6314sdi name to tbs6324
[   15.474224] gpio_ich: GPIO from 436 to 511 on gpio_ich
[   15.755004] WARNING: You are using an experimental version of the media stack.
                As the driver is backported to an older kernel, it doesn't offer
                enough quality for its usage in production.
                Use it with care.
               Latest git patches (needed if you report a bug to linux-media@vger.kernel.org):
                ea28c4ecbb81c14312b9012b87cdb3913c6568ec change tbs6314sdi name to tbs6324
[   15.840390] dvb_core: disagrees about version of symbol vb2_vmalloc_memops
[   15.840393] dvb_core: Unknown symbol vb2_vmalloc_memops (err -22)
[   15.860401] dvb_core: disagrees about version of symbol vb2_vmalloc_memops
[   15.860403] dvb_core: Unknown symbol vb2_vmalloc_memops (err -22)
```
I tried the command suggested above:
`sudo rm -r -f /lib/modules/$(uname -r)/kernel/drivers/linux`

No difference.
Any ideas what I can do?

4integration, 2018-07-13:
Running:
`sudo rm -rf /lib/modules/$(uname -r)/kernel/drivers/media/*`
Solved it