tbsdtv/linux_media #136: Code model kernel does not support pic mode

TheCazz, 2018-06-22:
Hi I trying to install the drivers of TBS on a Xubuntu 18.04 but get error
I maybe have forgot to install something before build the driver

I have install

git

gcc
libelf-dev

but when I try to run `make -j4` i see error
`cc1: error: code model kernel does not support PIC mode`

crazycat69, 2018-06-23:
Run **./build --check-only** and see what packages need install for driver building.

TheCazz, 2018-06-23:
Ahh thanks, have done that now and install what I need but it still say
Error and cc1: error: code model kernel does not support PIC mode

crazycat69, 2018-06-24:
wtf is PIC mode ?:) i try install Kubuntu 18,04 and installer hang after X server init (cursor alive, i can switch to tty2+ console). WTF ?:)

TheCazz, 2018-06-24:
Only good info I have found in this and not so easy for me to fix
https://askubuntu.com/questions/851433/kernel-doesnt-support-pic-mode-for-compiling

Nalyden, 2018-09-02:
I got around this error by adding a line to the Makefile in media_build/v4l.

In the "CFLAGS Configuration" section add the following line:
EXTRA_CFLAGS += -fno-pie

I'm also using Ubuntu and ran into another issue where the compiler could not find stdarg.h, I fixed that with another line in the same section:
EXTRA_CFLAGS += -I/usr/lib/gcc/x86_64-linux-gnu/7/include

Hope that helps someone, unfortunately I've hit another compiler error that I haven't figured out yet:
/home/nalyden/media_build/v4l/mt9t112.c: In function 'mt9t112_init_pll':
/home/nalyden/media_build/v4l/mt9t112.c:409:16: error: dereferencing pointer to incomplete type 'struct mt9t112_platform_data'
      priv->info->divider.m, priv->info->divider.n,
                ^~

Nalyden, 2018-09-02:
Found the problem, my kernel version is 4.15.0-30-generic and as "make allyesconfig" helpfully informs me I need at least 4.16.0.

Nalyden, 2018-09-03:
After upgrading my kernel version all of the original issues are fixed, no need to manually edit the Makefile after all.