tbsdtv/linux_media #138: error for upgrade and reinstall driver on kernel 4.17.4 ?

sajjads24, 2018-07-12:
hi
how fix error upgrade and reinstall after update kernel to 4.17.4 on fedora 28 ?
`[sajjad@localhost ~]$ cd media
[sajjad@localhost media]$ git remote update
Fetching origin
[sajjad@localhost media]$ git pull
Already up to date.
[sajjad@localhost media]$ cd ../media_build
[sajjad@localhost media_build]$ git remote update
Fetching origin
[sajjad@localhost media_build]$ git pull
Already up to date.
[sajjad@localhost media_build]$ make
make -C /home/sajjad/media_build/v4l 
make[1]: Entering directory '/home/sajjad/media_build/v4l'
creating symbolic links...
make -C firmware prep
make[2]: Entering directory '/home/sajjad/media_build/v4l/firmware'
make[2]: Nothing to be done for 'prep'.
make[2]: Leaving directory '/home/sajjad/media_build/v4l/firmware'
make -C firmware
make[2]: Entering directory '/home/sajjad/media_build/v4l/firmware'
make[2]: Nothing to be done for 'default'.
make[2]: Leaving directory '/home/sajjad/media_build/v4l/firmware'
Kernel build directory is /lib/modules/4.17.2-200.fc28.x86_64/build
make -C ../linux apply_patches
make[2]: Entering directory '/home/sajjad/media_build/linux'
Syncing with dir ../../media
Patches for 4.17.2-200.fc28.x86_64 already applied.
make[2]: Leaving directory '/home/sajjad/media_build/linux'
make -C /lib/modules/4.17.2-200.fc28.x86_64/build SUBDIRS=/home/sajjad/media_build/v4l  modules
make[2]: Entering directory '/home/sajjad/media_build/v4l'
make[2]: *** /lib/modules/4.17.2-200.fc28.x86_64/build: No such file or directory.  Stop.
make[2]: Leaving directory '/home/sajjad/media_build/v4l'
make[1]: *** [Makefile:53: default] Error 2
make[1]: Leaving directory '/home/sajjad/media_build/v4l'
make: *** [Makefile:26: all] Error 2
[sajjad@localhost media_build]$ sudo make install
`

crazycat69, 2018-07-12:
**make distclean**

sajjads24, 2018-07-12:
thanks good
`[sajjad@localhost ~]$ cd media
[sajjad@localhost media]$ git remote update
Fetching origin
[sajjad@localhost media]$ git pull
Already up to date.
[sajjad@localhost media]$ cd ../media_build
[sajjad@localhost media_build]$ git remote update
Fetching origin
[sajjad@localhost media_build]$ git pull
Already up to date.
[sajjad@localhost media_build]$ make distclean
make -C /home/sajjad/media_build/v4l distclean
make[1]: Entering directory '/home/sajjad/media_build/v4l'
rm -f *~ *.o *.ko .*.o.cmd .*.ko.cmd *.mod.c av7110_firm.h fdump \
        config-compat.h Module.symvers Module.markers modules.order \
        *.unsigned .*.ko.unsigned.cmd
make -C firmware clean
make[2]: Entering directory '/home/sajjad/media_build/v4l/firmware'
make[2]: Nothing to be done for 'clean'.
make[2]: Leaving directory '/home/sajjad/media_build/v4l/firmware'
rm -f .version .*.o.flags .*.o.d *.mod.gcno Makefile.media \
        Kconfig Kconfig.kern .config .config.cmd .myconfig \
        .kconfig.dep config-mycompat.h
rm -rf .tmp_versions .tmp*.ver .tmp*.o .*.gcno .cache.mk
rm -f scripts/lxdialog scripts/kconfig
make -C firmware distclean
make[2]: Entering directory '/home/sajjad/media_build/v4l/firmware'
make[2]: Nothing to be done for 'distclean'.
make[2]: Leaving directory '/home/sajjad/media_build/v4l/firmware'
make[1]: Leaving directory '/home/sajjad/media_build/v4l'
[sajjad@localhost media_build]$ sudo make install
[sudo] password for sajjad: 
make -C /home/sajjad/media_build/v4l install
make[1]: Entering directory '/home/sajjad/media_build/v4l'
No version yet, using 4.17.4-200.fc28.x86_64
scripts/make_makefile.pl
\nRemoving obsolete files from /lib/modules/4.17.4-200.fc28.x86_64/kernel/drivers/media/video/sn9c102:

\nRemoving obsolete files from /lib/modules/4.17.4-200.fc28.x86_64/kernel/drivers/media/dvb/ttusb-budget:

\nRemoving obsolete files from /lib/modules/4.17.4-200.fc28.x86_64/kernel/drivers/media/dvb/ddbridge:

\nRemoving obsolete files from /lib/modules/4.17.4-200.fc28.x86_64/kernel/drivers/media/dvb/dvb-core:

\nRemoving obsolete files from /lib/modules/4.17.4-200.fc28.x86_64/kernel/drivers/media/video/au0828:

\nRemoving obsolete files from /lib/modules/4.17.4-200.fc28.x86_64/kernel/drivers/media/video/pvrusb2:

\nRemoving obsolete files from /lib/modules/4.17.4-200.fc28.x86_64/kernel/drivers/media/video/cx25821:

\nRemoving obsolete files from /lib/modules/4.17.4-200.fc28.x86_64/kernel/drivers/media/dvb/b2c2:

\nRemoving obsolete files from /lib/modules/4.17.4-200.fc28.x86_64/kernel/drivers/media/dvb/ttusb-dec:

\nRemoving obsolete files from /lib/modules/4.17.4-200.fc28.x86_64/kernel/drivers/media/dvb/ttpci:

\nRemoving obsolete files from /lib/modules/4.17.4-200.fc28.x86_64/kernel/drivers/media/dvb/mantis:

\nRemoving obsolete files from /lib/modules/4.17.4-200.fc28.x86_64/kernel/drivers/media/common/tuners:

\nRemoving obsolete files from /lib/modules/4.17.4-200.fc28.x86_64/kernel/drivers/media/dvb/pluto2:

\nRemoving obsolete files from /lib/modules/4.17.4-200.fc28.x86_64/kernel/drivers/media/video/gspca/gl860:

\nRemoving obsolete files from /lib/modules/4.17.4-200.fc28.x86_64/kernel/drivers/media/video/marvell-ccic:

\nRemoving obsolete files from /lib/modules/4.17.4-200.fc28.x86_64/kernel/drivers/media/video/hdpvr:

\nRemoving obsolete files from /lib/modules/4.17.4-200.fc28.x86_64/kernel/drivers/media/video/uvc:
[... 42 lines cut ...]

\nRemoving obsolete files from /lib/modules/4.17.4-200.fc28.x86_64/kernel/drivers/media/dvb/frontends:

\nRemoving obsolete files from /lib/modules/4.17.4-200.fc28.x86_64/kernel/drivers/media/video/saa7164:

\nRemoving obsolete files from /lib/modules/4.17.4-200.fc28.x86_64/kernel/drivers/media/dvb/firewire:

\nRemoving obsolete files from /lib/modules/4.17.4-200.fc28.x86_64/kernel/drivers/media/dvb/dm1105:

\nRemoving obsolete files from /lib/modules/4.17.4-200.fc28.x86_64/kernel/drivers/media/dvb/ngene:

\nRemoving obsolete files from /lib/modules/4.17.4-200.fc28.x86_64/kernel/drivers/media/video/bt8xx:

\nRemoving obsolete files from /lib/modules/4.17.4-200.fc28.x86_64/kernel/drivers/media/dvb/dvb-usb:

\nRemoving obsolete files from /lib/modules/4.17.4-200.fc28.x86_64/kernel/drivers/media/video/em28xx:

Installing kernel modules under /lib/modules/4.17.4-200.fc28.x86_64/kernel/drivers/media/:
/sbin/depmod -a 4.17.4-200.fc28.x86_64 
make -C firmware install
make[2]: Entering directory '/home/sajjad/media_build/v4l/firmware'
Installing firmwares at /lib/firmware: 
make[2]: Leaving directory '/home/sajjad/media_build/v4l/firmware'
make[1]: Leaving directory '/home/sajjad/media_build/v4l'
[sajjad@localhost media_build]$ `

sajjads24, 2018-07-12:
please edit wiki and add this command : make distclean