tbsdtv/linux_media #139: TBS 6618 Support

fatexs, 2018-07-14:
It would be quite nice if you could also add support for the TBS6618, it is one of the older pcie DVB-C single receivers.
lspci: 
Multimedia controller: 
Philips Semiconductors SAA7160 (rev 03)
Subsystem: Device 6618:0001

Thanks!

crazycat69, 2018-07-14:
What demod IC ?

fatexs, 2018-07-14:
TDA10024HN

PhobosK, 2018-11-27:
Any progress on this issue?
Because my card is useless now in Ubuntu 18.04 with kernel 4.15

Ablinne, 2019-09-22:
Pleeease add support for the TBS6618. I also have it here, currently running on Ubuntu 16.04 and even with the old lts kernel 4.4.x it is getting increasingly difficult to compile the years old closed source driver. Already needed to use random patches and disable random modules in  `.config`...

johnzered, 2021-11-08:
Is there any chance for adding support for this card?