tbsdtv/linux_media #141: Build error on Arch linux-4.18.0

wdirksen, 2018-08-16:
Build error against 4.18-0 in Arch currently in testing
Kernel is standard
TBS build options standard with the exception of altering the following:
`sed -i 's/CONFIG_MEDIA_CONTROLLER_DVB=y/# CONFIG_MEDIA_CONTROLLER_DVB is not set/' v4l/.config`

build log:
```
==> Starting build()...
make -C /home/mythtv/AUR/tbs-os-dvb-drivers-git/src/media_build/v4l 
make[1]: Entering directory '/home/mythtv/AUR/tbs-os-dvb-drivers-git/src/media_build/v4l'
scripts/make_makefile.pl
./scripts/make_myconfig.pl
[ ! -f "./config-mycompat.h" ] && echo "/* empty config-mycompat.h */" > "./config-mycompat.h" || true
make -C firmware prep
perl scripts/make_config_compat.pl /lib/modules/4.18.0-arch1-1-ARCH/build ./.myconfig ./config-compat.h
creating symbolic links...
make[2]: Entering directory '/home/mythtv/AUR/tbs-os-dvb-drivers-git/src/media_build/v4l/firmware'
make[2]: Nothing to be done for 'prep'.
make[2]: Leaving directory '/home/mythtv/AUR/tbs-os-dvb-drivers-git/src/media_build/v4l/firmware'
make -C firmware
make[2]: Entering directory '/home/mythtv/AUR/tbs-os-dvb-drivers-git/src/media_build/v4l/firmware'
make[2]: Nothing to be done for 'default'.
make[2]: Leaving directory '/home/mythtv/AUR/tbs-os-dvb-drivers-git/src/media_build/v4l/firmware'
Kernel build directory is /lib/modules/4.18.0-arch1-1-ARCH/build
make -C ../linux apply_patches
make[2]: Entering directory '/home/mythtv/AUR/tbs-os-dvb-drivers-git/src/media_build/linux'
Syncing with dir ../../linux_media
Patches for 4.18.0-arch1-1-ARCH already applied.
make[2]: Leaving directory '/home/mythtv/AUR/tbs-os-dvb-drivers-git/src/media_build/linux'
make -C /lib/modules/4.18.0-arch1-1-ARCH/build SUBDIRS=/home/mythtv/AUR/tbs-os-dvb-drivers-git/src/media_build/v4l  modules
make[2]: Entering directory '/usr/lib/modules/4.18.0-arch1-1-ARCH/build'
  CC [M]  /home/mythtv/AUR/tbs-os-dvb-drivers-git/src/media_build/v4l/msp3400-driver.o
  CC [M]  /home/mythtv/AUR/tbs-os-dvb-drivers-git/src/media_build/v4l/msp3400-kthreads.o
  CC [M]  /home/mythtv/AUR/tbs-os-dvb-drivers-git/src/media_build/v4l/smiapp-core.o
  CC [M]  /home/mythtv/AUR/tbs-os-dvb-drivers-git/src/media_build/v4l/smiapp-regs.o
  CC [M]  /home/mythtv/AUR/tbs-os-dvb-drivers-git/src/media_build/v4l/smiapp-quirk.o
  CC [M]  /home/mythtv/AUR/tbs-os-dvb-drivers-git/src/media_build/v4l/smiapp-limits.o
  CC [M]  /home/mythtv/AUR/tbs-os-dvb-drivers-git/src/media_build/v4l/et8ek8_mode.o
  CC [M]  /home/mythtv/AUR/tbs-os-dvb-drivers-git/src/media_build/v4l/et8ek8_driver.o
  CC [M]  /home/mythtv/AUR/tbs-os-dvb-drivers-git/src/media_build/v4l/cx25840-core.o
  CC [M]  /home/mythtv/AUR/tbs-os-dvb-drivers-git/src/media_build/v4l/cx25840-audio.o
  CC [M]  /home/mythtv/AUR/tbs-os-dvb-drivers-git/src/media_build/v4l/cx25840-firmware.o
  CC [M]  /home/mythtv/AUR/tbs-os-dvb-drivers-git/src/media_build/v4l/cx25840-vbi.o
  CC [M]  /home/mythtv/AUR/tbs-os-dvb-drivers-git/src/media_build/v4l/cx25840-ir.o
  CC [M]  /home/mythtv/AUR/tbs-os-dvb-drivers-git/src/media_build/v4l/m5mols_core.o
  CC [M]  /home/mythtv/AUR/tbs-os-dvb-drivers-git/src/media_build/v4l/m5mols_controls.o
  CC [M]  /home/mythtv/AUR/tbs-os-dvb-drivers-git/src/media_build/v4l/m5mols_capture.o
  CC [M]  /home/mythtv/AUR/tbs-os-dvb-drivers-git/src/media_build/v4l/mt9m001.o
  CC [M]  /home/mythtv/AUR/tbs-os-dvb-drivers-git/src/media_build/v4l/mt9t112.o
  CC [M]  /home/mythtv/AUR/tbs-os-dvb-drivers-git/src/media_build/v4l/mt9v022.o
  CC [M]  /home/mythtv/AUR/tbs-os-dvb-drivers-git/src/media_build/v4l/ov5642.o
  CC [M]  /home/mythtv/AUR/tbs-os-dvb-drivers-git/src/media_build/v4l/ov772x.o
  CC [M]  /home/mythtv/AUR/tbs-os-dvb-drivers-git/src/media_build/v4l/ov9640.o
  CC [M]  /home/mythtv/AUR/tbs-os-dvb-drivers-git/src/media_build/v4l/ov9740.o
  CC [M]  /home/mythtv/AUR/tbs-os-dvb-drivers-git/src/media_build/v4l/rj54n1cb0c.o
  CC [M]  /home/mythtv/AUR/tbs-os-dvb-drivers-git/src/media_build/v4l/tw9910.o
  CC [M]  /home/mythtv/AUR/tbs-os-dvb-drivers-git/src/media_build/v4l/aptina-pll.o
  CC [M]  /home/mythtv/AUR/tbs-os-dvb-drivers-git/src/media_build/v4l/tvaudio.o
  CC [M]  /home/mythtv/AUR/tbs-os-dvb-drivers-git/src/media_build/v4l/tda7432.o
  CC [M]  /home/mythtv/AUR/tbs-os-dvb-drivers-git/src/media_build/v4l/saa6588.o
  CC [M]  /home/mythtv/AUR/tbs-os-dvb-drivers-git/src/media_build/v4l/tda9840.o
  CC [M]  /home/mythtv/AUR/tbs-os-dvb-drivers-git/src/media_build/v4l/tda1997x.o
  CC [M]  /home/mythtv/AUR/tbs-os-dvb-drivers-git/src/media_build/v4l/tea6415c.o
  CC [M]  /home/mythtv/AUR/tbs-os-dvb-drivers-git/src/media_build/v4l/tea6420.o
  CC [M]  /home/mythtv/AUR/tbs-os-dvb-drivers-git/src/media_build/v4l/saa7110.o
  CC [M]  /home/mythtv/AUR/tbs-os-dvb-drivers-git/src/media_build/v4l/saa7115.o
/home/mythtv/AUR/tbs-os-dvb-drivers-git/src/media_build/v4l/tda1997x.c: In function 'tda1997x_pcm_startup':
/home/mythtv/AUR/tbs-os-dvb-drivers-git/src/media_build/v4l/tda1997x.c:2447:35: error: 'struct snd_soc_dai' has no member named 'codec'
  struct snd_soc_codec *codec = dai->codec;
                                   ^~
[... 35 lines cut ...]
/home/mythtv/AUR/tbs-os-dvb-drivers-git/src/media_build/v4l/tda1997x.c: In function 'tda1997x_probe':
/home/mythtv/AUR/tbs-os-dvb-drivers-git/src/media_build/v4l/tda1997x.c:2740:9: error: implicit declaration of function 'snd_soc_register_codec'; did you mean 'snd_soc_register_dai'? [-Werror=implicit-function-declaration]
   ret = snd_soc_register_codec(&state->client->dev,
         ^~~~~~~~~~~~~~~~~~~~~~
         snd_soc_register_dai
/home/mythtv/AUR/tbs-os-dvb-drivers-git/src/media_build/v4l/tda1997x.c: In function 'tda1997x_remove':
/home/mythtv/AUR/tbs-os-dvb-drivers-git/src/media_build/v4l/tda1997x.c:2785:3: error: implicit declaration of function 'snd_soc_unregister_codec'; did you mean 'snd_soc_unregister_card'? [-Werror=implicit-function-declaration]
   snd_soc_unregister_codec(&client->dev);
   ^~~~~~~~~~~~~~~~~~~~~~~~
   snd_soc_unregister_card
/home/mythtv/AUR/tbs-os-dvb-drivers-git/src/media_build/v4l/tda1997x.c: At top level:
/home/mythtv/AUR/tbs-os-dvb-drivers-git/src/media_build/v4l/tda1997x.c:2492:36: error: storage size of 'tda1997x_codec_driver' isn't known
 static struct snd_soc_codec_driver tda1997x_codec_driver = {
                                    ^~~~~~~~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
make[3]: *** [scripts/Makefile.build:324: /home/mythtv/AUR/tbs-os-dvb-drivers-git/src/media_build/v4l/tda1997x.o] Error 1
make[3]: *** Waiting for unfinished jobs....
make[2]: *** [Makefile:1500: _module_/home/mythtv/AUR/tbs-os-dvb-drivers-git/src/media_build/v4l] Error 2
make[2]: Leaving directory '/usr/lib/modules/4.18.0-arch1-1-ARCH/build'
make[1]: *** [Makefile:53: default] Error 2
make[1]: Leaving directory '/home/mythtv/AUR/tbs-os-dvb-drivers-git/src/media_build/v4l'
make: *** [Makefile:26: all] Error 2
==> ERROR: A failure occurred in build().
    Aborting...
```

CvH, 2018-08-16:
this media_tree is still based at 4.17 -> 4.18 fails
that is the pain of out of tree drivers instead of proper ones :( https://github.com/tbsdtv/linux_media/issues/87

wdirksen, 2018-08-16:
OK, so 4.18 will have to wait until the V4L-DVB media stack catches up.

crazycat69, 2018-08-19:
[linux_media](https://github.com/tbsdtv/linux_media) and [media_build](https://github.com/tbsdtv/media_build) updated, please check again wih 4.18 kernel

wdirksen, 2018-08-20:
Thanks builds perfect again on 4.18-1 ARCH

closed 2018-08-23 by crazycat69