tbsdtv/linux_media #142: tbsecp3 not working on Ubuntu 18.04 LTS running 4.15.0-30-generic kernel

anttito1, 2018-08-17:
Just updated my system from Ubuntu 16.04 LTS to 18.04 LTS, which is now running 4.15.0-30-generic kernel.

Tried to install this open source driver for my TBS6281SE DVB card. Build finishes without errors, but while loading the module tbsecp3 I get

$ sudo modprobe tbsecp3
modprobe: ERROR: could not insert 'tbsecp3': Unknown symbol in module, or unknown parameter (see dmesg)

$ dmesg | tail -3
[121793.067595] tas2101: Unknown symbol i2c_mux_alloc (err 0)
[121793.067623] tas2101: Unknown symbol i2c_mux_del_adapters (err 0)
[121793.067647] tas2101: Unknown symbol i2c_mux_add_adapter (err 0)

Has this Ubuntu kernel been compiled without i2c mux support? What are the options?

Tried with e109755c5c93e43b7482e123acaea799d540e3de, which is the current master.

crazycat69, 2018-08-19:
Just now compiled and installed latest drivers on Kubuntu 18.04 with 4.15.0-32-generic kernel.
i2c-mux present, drivers loaded ok (TBS 6982 with tas2101 demod).
Maybe some old kernel media modules prevent loading new media modules ? Need remove media modules before install new (see **install.sh** script in [media_build](https://github.com/tbsdtv/media_build))

anttito1, 2018-08-20:
This is odd...

$ uname -a
Linux xxx 4.15.0-30-generic #32-Ubuntu SMP Thu Jul 26 17:42:43 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux

$ sudo modinfo i2c-mux
modinfo: ERROR: Module i2c-mux not found.

Apparently upgrade from 16.04 to 18.04 did not go well. Seems that modules package is missing at least. Not driver related. Closing the ticket.

Thanks crazycat!

closed 2018-08-20 by anttito1