tbsdtv/linux_media #144: Driver Issue during/after resume from Wakeup

jpg73, 2018-09-11:
Dears,
Using kernel 4.16.3 with latest TBSdriver on TBS6590 + legal CI Card
testcase is : my server is running well, but if I
1/sudo systemctl suspend
2/wakeup by WOL or keyboard
3/seems drivers isn't loaded properly ( see dmesg line 820 of http://paste.ubuntu.com/p/JQxj4MjWFr/)

Ive tried
sudo modprobe -v -r tbsecp3
sudo modprobe -v -r si2157
sudo modprobe -v -r si2183
sudo modprobe -v -r av201x
in a shell in /usr/lib/pm-utils/sleep.d/
But I got segmentation fault...

Could you indicate me which module I ve to unload / load during the suspend/wakeup process ?
or maybe it's a driver bug ?
Thanks in advance
JP

jpg73, 2018-10-12:
After deep check, seems driver issue, could you please confirm ?
THX

crazycat69, 2018-10-13:
try unload/load only **tbsecp3** module

jpg73, 2018-10-14:
Hi , I already tried the unload /load but got a kernel panic :-(
seems driver issue
THX

crazycat69, 2018-10-16:
Check this change in tbsecp3. At least for me thsi prebent crash on tbsecp3 unload/load for 6522.
23087976705d594febef88e9160c3eb76abb2d48

Btw. you can create config **/etc/pm/config.d/modules**
> SUSPEND_MODULES="tbsecp3"

jpg73, 2018-10-17:
Looks nice ... after first tests, I ve no more kernel panic, I'll continue the tests & keep you posted in case of issue
That's BRILLIANT ! THANKS A LOT !!! 
JP