tbsdtv/linux_media #145: TBS5580 cannot lock QAM256

JohnBeton22, 2018-10-01:
I have problems with locking signal in Ubuntu 16.04.5. Kernel is Linux PC-ACER 4.4.0-116-generic #140-Ubuntu SMP Mon Feb 12 21:22:43 UTC 2018 i686 athlon i686 GNU/Linux

LSUSB
Bus 001 Device 004: ID 734c:5580 TBS Technologies China

DMESG
[22358.558645] usb 1-5: new high-speed USB device number 4 using ehci-pci
[22358.692183] usb 1-5: New USB device found, idVendor=734c, idProduct=5580
[22358.692197] usb 1-5: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[22358.692205] usb 1-5: Product: TBS 5580
[22358.692211] usb 1-5: Manufacturer: TBS-Tech
[22358.693508] dvb-usb: found a 'TBS 5580 CI USB2.0' in cold state, will try to load a firmware
[22358.693517] dvb-usb: downloading firmware from file 'dvb-usb-id5580.fw'
[22358.693522] tbs5580: start downloading TBS5580 firmware
[22358.810950] dvb-usb: found a 'TBS 5580 CI USB2.0' in warm state.
[22358.811360] dvb-usb: will pass the complete MPEG2 transport stream to the software demuxer.
[22358.811637] dvbdev: DVB: registering new adapter (TBS 5580 CI USB2.0)
[22359.036259] dvb-usb: MAC address: 00:22:ab:c0:46:3e
[22359.040487] i2c i2c-11: Added multiplexed i2c bus 12
[22359.040500] si2183 11-0067: Silicon Labs Si2183 successfully attached
[22359.049511] si2157 12-0061: Silicon Labs Si2147/2148/2157/2158 successfully attached
[22359.049554] i2c i2c-12: av201x: Airoha Technology AV201x successfully attached
[22359.050393] tbs5580: CI initialized.
[22359.051245] tbs5580: CI disabled.
[22359.051254] usb 1-5: DVB: registering adapter 0 frontend 0 (TBS 5580 CI USB2.0 DVB-T/T2/C/C2/ISDB-T)...
[22359.051367] usb 1-5: DVB: registering adapter 0 frontend 1 (TBS 5580 CI USB2.0 DVB-S/S2/S2X)...
[22359.052627] dvb-usb: TBS 5580 CI USB2.0 successfully initialized and connected.

LSBRELEASE -A
Distributor ID: Ubuntu
Description:    Ubuntu 16.04.5 LTS
Release:        16.04
Codename:       xenial

DVBLAST cannot lock

root@TIGER:/home/tiger# dvblast -a 0 -s 6875000 -5 DVBC_ANNEX_A -f 298000000 -U -d 192.168.1.197:1234  -m qam_256
DVBlast 3.0 (release)
warning: restarting
warning: raw UDP output is deprecated.  Please consider using RTP.
warning: for DVB-IP compliance you should use RTP.
debug: compiled with DVB API version 5.10
debug: using DVB API version 5.11
debug: Frontend "TBS 5580 CI USB2.0 DVB-T/T2/C/C2/ISDB-T" supports:
debug:  frequency min: 42000000, max: 870000000, stepsize: 0, tolerance: 0
debug:  symbolrate min: 1000000, max: 45000000, tolerance: 0
debug:  capabilities:
debug:   INVERSION_AUTO
debug:   FEC_1_2
debug:   FEC_2_3
debug:   FEC_3_4
debug:   FEC_5_6
debug:   FEC_7_8
debug:   FEC_AUTO
debug:   QPSK
debug:   QAM_16
debug:   QAM_32
debug:   QAM_64
debug:   QAM_128
debug:   QAM_256
debug:   QAM_AUTO
debug:   TRANSMISSION_MODE_AUTO
debug:   GUARD_INTERVAL_AUTO
debug:   HIERARCHY_AUTO
debug:   2G_MODULATION
debug:   MULTISTREAM
debug:   FE_CAN_MUTE_TS
debug:  delivery systems:
debug:   DVBT
debug:   DVBT2
debug:   DVBC_ANNEX_A
debug:   ISDBT
debug:   DVBC_ANNEX_B
debug: tuning DVB-C frontend to f=298000000 srate=6875000 inversion=-1 modulation=qam_256
debug: CA interface with 1 slot
debug:   CI link layer level interface type
debug:   0 available descramblers (keys)
CAM none
debug: setting filter on PID 0
debug: setting filter on PID 16
debug: setting filter on PID 17
debug: setting filter on PID 18
debug: setting filter on PID 19
debug: setting filter on PID 20
error: no config file
debug: frontend has acquired signal
debug: frontend has acquired carrier
debug: frontend has lost signal
debug: frontend has lost carrier
debug: frontend has acquired signal
debug: frontend has acquired carrier
debug: frontend has lost signal
debug: frontend has lost carrier
debug: frontend has acquired signal
debug: frontend has acquired carrier
debug: frontend has lost signal
debug: frontend has lost carrier
debug: frontend has acquired signal
debug: frontend has acquired carrier
debug: frontend has lost signal
debug: frontend has lost carrier
debug: frontend has acquired signal
debug: frontend has acquired carrier

What could be the issue here? Thanks! TVHEADEND or Minisatip also cannot lock signal. Cabling is fine, I have another device TBS5881 which is working just fine.

JohnBeton22, 2018-10-09:
@steven1758 @Davin622 Could you please confirm?

JohnBeton22, 2018-10-30:
Please help..

zhang111chao, 2018-12-27:
please try this command , add  "-u"   

dvblast -a 0 -s 6875000 -5 DVBC_ANNEX_A -f 298000000 -U  -u  -d 192.168.1.197:1234 -m qam_256

bccrew, 2021-03-21:
Hello, this is still a problem in all newer Ubuntu/Debian releases.