tbsdtv/linux_media #146: problem install driver on fedora 28 with vanilla kernel?!

sajjads24, 2018-10-07:
hi
i am install vanilla kernel on fedora 28 from : 
`https://fedoraproject.org/wiki/Kernel_Vanilla_Repositories`
so after install kernel i am install driver from : 
`https://github.com/tbsdtv/linux_media/wiki`
for tbs6983 but after install drivel not work on kaffeine !?
how install this driver on vanilla kernel ?
`[sajjad@localhost ~]$ lspci | grep Multimedia
09:00.0 Multimedia controller: Philips Semiconductors SAA7160 (rev 03)
[sajjad@localhost ~]$ dmesg | grep -i dvb
[   16.138174] saa716x_tbs_dvb: Unknown symbol cx24117_get_i2c_adapter (err 0)
[   18.537838] dvbdev: DVB: registering new adapter (Siano Nice Digital Receiver)
[   18.538030] usb 2-1.3: DVB: registering adapter 0 frontend 0 (Siano Mobile Digital MDTV Receiver)...
[   18.538068] smsdvb:smsdvb_hotplug: DVB interface registered.
[sajjad@localhost ~]$ /dev/dvb
bash: /dev/dvb: Is a directory
[sajjad@localhost ~]$ ls /dev/dvb
adapter0
[sajjad@localhost ~]$ `

sajjads24, 2018-10-07:
Does this driver support vanilla kernel on fedora 28 ?

zhang111chao, 2018-10-08:
Downloading and building
#git clone https://github.com/tbsdtv/media_build.git
#git clone --depth=1 https://github.com/tbsdtv/linux_media.git -b latest ./media
#cd media_build
#make dir DIR=../media
#make allyesconfig
#make -j4
#sudo make install
#sudo reboot

sajjads24, 2018-10-08:
i am install driver again and all things good : 
`https://pastebin.com/b4Mxsj2Q`
so after install complet and reboot system output some command : 
`https://pastebin.com/TY24mZHG`
and not work on vanilla kernel for fedora 28 !?
so i am try back to default kernel on fedora 28 .

crazycat69, 2018-10-08:
4.18 kernel supported.
Remove kernel media modules before install - see **media_build/install.sh script**

sajjads24, 2018-10-10:
thanks crazycat69 i am use this command before install : 
`#sudo rm -rf /lib/modules/`uname -r`/kernel/drivers/media/`
so after install driver all things gooood work for me .thanks alot
[https://pastebin.com/iu176LQr](url)

closed 2018-10-13 by crazycat69