tbsdtv/linux_media #147: driver compilation not OK under kernel 4.19

jpg73, 2018-10-30:
Dear, 
compiling was OK with 4.15, 4.16, 4.18 & 4.19rc , but failed under 4.19
please see http://paste.ubuntu.com/p/QqwFbV98Vy/
Any advice ?
THX
JP

crazycat69, 2018-11-01:
no merget yes with latest media_tree, wait some days.

crazycat69, 2018-11-05:
updated, please check again with 4.19

alanswanson, 2018-11-05:
Build still failing with 4.19 at si2183.c as i2c_[un]lock_adapter were removed upstream in this  [commit](https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/?id=3f3a89e1d7c31558c070692241e3d6146d2cf1bf). Changing to i2c_[un]lock_bus with I2C_LOCK_ROOT_ADAPTER per commit message, compile succeeds but I've not runtime tested yet.

crazycat69, 2018-11-06:
303ceb42a0d3a5a0da0d629e47b12b9ca4634853

jpg73, 2018-11-08:
works ! 👍 THX a lot CC
JP

closed 2018-11-08 by crazycat69