tbsdtv/linux_media #148: TBS6522 signal issue DVB-S2 with high SR

piotrasd, 2018-10-31:
Hi
Last week I tested from 42E first 8K channel (demo)

https://en.kingofsat.net/find.php?question=8k
Turksat 3A | 11177.00 | V | 12 | West | DVB-S2 | 16APSK | 34288 2/3 | 88.3 Mbps

I special setup dish 80cm on 42E and on this specific TP aligne signal - which my Inverto SatPal meter showed me 100%/98% 

but I couldn't capture RAW TS, TBS all time showing me a lot of errors and loosing lock :( 
i tested other channel 4K from this same satellite - and was perfect - 0 errors etc..

Im just suspect driver don't well handle high SR TP 

[log.txt](https://github.com/tbsdtv/linux_media/files/2532226/log.txt)

crazycat69, 2018-11-01:
High modulation + high SR. I think Silabs demods have some limitation for LDPC performance ~33000Ks for 16APSK. + need C/N minimum ~10 dB for 16APSK,2/3.

crazycat69, 2018-11-01:
So i have check 6522 with this transponder, 0,95m motorized dish - SNR ~9.5-10dB, BER 0, stream, ~90mbit, no any CC errors. Linux and Windows.

[image]
So this your signal issue, check SNR - must be > 9dB (>45%). You need > 0.9d dish for this transponder.

piotrasd, 2018-11-03:
@crazycat69 big thanks for your time to confirm that, in that case I will play with my satellite dish.