tbsdtv/linux_media #153: TBS 6281 MythTV signal strength (open source vs. closed source)

ni777, 2018-12-31:
After a couple of years I tried the open source driver again for my TBS 6281.

With the open source driver (at commit 669e3e92cc0fc4ce9039bd11cfb7f2e554c34e70),
MythTV 29 .1 displays a signal strength of 40% and cannot lock the channel.

With the closed source driver (v170330), MythTV displays a signal strength of 93% and can lock the channel.

That is, the open source driver still doesn't work for me. Is there anything I can do?

bas-t, 2019-01-10:
Yes you can.
MythTV still is not capable to handle multi-standard cards like your TBS 6281.
It only handles the first statement in the line SYS_DVBT, SYS_DVBT2, SYS_DVBC_ANNEX_A in
drivers/media/dvb-frontends/si2168.c

So you have to change the order. Just put SYS_DVBT2 at the beginning of that line prior to compiling.
That should do the trick.
Cheers,
Tycho.

ni777, 2019-01-11:
Well, MythTV (built from a suitable git commit back then) has been working with DVB-T2 and the closed-source driver for the TBS 6281 since March 2017, so does MythTV 29.1, therefore I don't think we can blame MythTV. I'll try you suggestion anyway.

ni777, 2019-03-04:
Putting SYS_DVBT2 first doesn't make a difference.

a20dev, 2020-05-17:
I think I found the problem: apparently autodetection of the bandwidth does not work with the open-source driver (but it does work with the closed-source driver). After setting the bandwidth to 8 MHz (value taken from w_scan output) in the transport editor of MythTV, MythTV works with the open source driver. Scanning still doesn't work.

And as someone wrote above, the signal strength reported by the open-source driver still is quite different from the signal strength reported by the closed-source driver.

a20dev, 2020-05-17:
If the bandwidth is set to Auto, MythTV seems to set the bandwidth to 5 MHz.: si2168_set_frontend() logs bandwidth_hz=5000000. I don't know why this works with the closed-source driver and doesn't work with the open-source driver.