tbsdtv/linux_media #156: TBS 5520SE not working in new kernels

misushot, 2019-03-06:
Hi i have a TBS 5520SE and it seem not working under Archlinux uname -r -> 5.0.0-arch1-1-ARCH.
git clone https://github.com/tbsdtv/media_build.git
git clone --depth=1 https://github.com/tbsdtv/linux_media.git -b latest ./media
cd media_build
make dir DIR=../media
make allyesconfig
(and with)
make -j4
make[3]: *** [scripts/Makefile.build:276: /var/dvb/src/gittbs/media_build/v4l/v4l2-compat-ioctl32.o] Fehler 1
make[3]: *** Es wird auf noch nicht beendete Prozesse gewartet....
make[2]: *** [Makefile:1553: _module_/var/dvb/src/gittbs/media_build/v4l] Error 2
make[2]: Leaving directory '/usr/lib/modules/5.0.0-arch1-1-ARCH/build'
make[1]: *** [Makefile:53: default] Fehler 2
make[1]: Verzeichnis „/var/dvb/src/gittbs/media_build/v4l“ wird verlassen
make: *** [Makefile:26: all] Fehler 2

CvH, 2019-03-06:
current drivers are not updated to 5.0 yet so its expected that it currently didn't work

misushot, 2019-03-06:
thx for your answer CvH :). which is the last working kernel version for tbs? or which distro where fine for my TBS 5520SE USB device?

crazycat69, 2019-03-06:
tested up to 4.19
will be updated soon

misushot, 2019-03-06:
> tested up to 4.19
> will be updated soon

thx crazycat :D . raspberry pi 3(ARM) support?

crazycat69, 2019-03-06:
open-source, i you can compile - must work
possible issues with usb 3.0 hosts

misushot, 2019-03-06:
fine i will test with old raspbian(debian) it in few days and come back with infos 👍

misushot, 2019-03-14:
https://aur.archlinux.org/packages/tbs-linux_media-git/ is compile is ready so full archlinux compatible thx ;)
  
  LD [M]  /var/dvb/bin/src/tbs-linux_media-git/src/media_build/v4l/zr364xx.ko
make[2]: Leaving directory '/usr/lib/modules/5.0.0-arch1-1-ARCH/build'
./scripts/rmmod.pl check
found 729 modules
make[1]: Verzeichnis „/var/dvb/bin/src/tbs-linux_media-git/src/media_build/v4l“ wird verlassen
==> Betrete fakeroot Umgebung...
==> Beginne package()...
==> Compressing modules, this will take awhile...
==> Säubere Installation...
  -> Entferne libtool Dateien...
  -> Bereinige ungewollte Dateien...
  -> Entferne statische Bibliotheken...
  -> Entferne unnötige Symbole aus Binär-Dateien und Bibliotheken...
  -> Komprimiere Man-Pages und Info-Seiten...
==> Checking for packaging issues...
==> Erstelle Paket "tbs-linux_media-git"...
  -> Erstelle .PKGINFO Datei...
  -> Erstelle .BUILDINFO Datei...
  -> Erstelle .MTREE-Datei...
  -> Komprimiere Paket...
==> Verlasse fakeroot Umgebung.
==> Beendete Erstellung: tbs-linux_media-git r1179.db82890_5.0.0_arch1_1_ARCH-1 (Do 14 Mär 2019 20:48:18 CET)

crasycat69 you have programmed crazyscan for windows. it gives cli tool for linux too. its better for
my own app to get dvb infos via verbose output and .....

misushot, 2019-03-14:
de-ms-lager-srv-vdr / # lsmod |grep dvb
dvb_usb_tbs5520se      20480  0
dvb_usb                36864  1 dvb_usb_tbs5520se
dvb_core              163840  2 dvb_usb,ddbridge
videobuf2_vmalloc      20480  1 dvb_core
videobuf2_common       61440  1 dvb_core
rc_core                57344  1 dvb_usb
media                  57344  4 videodev,si2157,dvb_core,videobuf2_common
perfect guys i'am happy. i have order tody a new card TBS-6909-X with s2x support. when kernel 5 support tbs i will use it.

TRPB, 2019-05-25:
I can't get this to build on Arch. Does it need updating for 5.1?

```
cc1: some warnings being treated as errors
make[3]: *** [scripts/Makefile.build:282: /home/media/.cache/yay/tbs-linux_media-git/src/media_build/v4l/ipu3-cio2.o] Error 1
make[3]: *** Waiting for unfinished jobs....
make[2]: *** [Makefile:1571: _module_/home/media/.cache/yay/tbs-linux_media-git/src/media_build/v4l] Error 2
make[2]: Leaving directory '/usr/lib/modules/5.1.4-arch1-1-ARCH/build'
make[1]: *** [Makefile:53: default] Error 2
make[1]: Leaving directory '/home/media/.cache/yay/tbs-linux_media-git/src/media_build/v4l'
make: *** [Makefile:26: all] Error 2
==> ERROR: A failure occurred in build().
    Aborting...
Error making: tbs-linux_media-git
```

crazycat69, 2019-05-26:
updated, please check

TRPB, 2019-05-27:
thank you, rebuilding from git now works on 5.1.

victorvhpg, 2019-10-28:
my TBS5520se didnt works on my raspberry pi 4   -  Raspbian Buster

erbasandrade, 2019-10-31:
+1 Awaiting correction of TBS5520SE drivers
To use LibreELEC.