tbsdtv/linux_media #157: Compilation issue for v4l2_fwnode on kernel 4.18

amenophiks, 2019-03-20:
Hi,

During compilation, I encountered issue for _v4l2-fwnode.c_ file compilation.

I saw that a patch is present to take into account the change on fwnode_reference_args argument (args one)
When I check patch to apply with `linux/patches_for_kernel.pl <version>`, the patch _v4.18_fwnode_args_args.patch_ is listed, but seems to not been applied as file remains with `u64 * args`

My configuration:
- 4.18.0-16-generic #17~18.04.1-Ubuntu SMP Tue Feb 12 13:35:51 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux

crazycat69, 2019-03-20:
strange, i compile on ubuntu 18.04 with 4.18 kernel without any issues.

amenophiks, 2019-04-03:
Hi,
After a new compilation from the beginning, no more issue (4.18.0-17).

closed 2019-04-03 by amenophiks