tbsdtv/linux_media #159: TBS 6909X - Performance Issues with TVHeadend

Krobar, 2019-05-04:
I'm using a 6909X in mode 1 with drivers built yesterday and the latest TVH 4.3 built from source. For now only 1 tuner on the 6909X is enabled, the performance problems get worse if I enable more. The tuning does work but the TVH interface is really laggy and tuning is slow, the VM remains responsive from command line. 

A mutex analysis in TVH shows this pattern repeating on every mux scan:
thread: mutex 0x55e36afe4560 at src/main.c:703 took 2120ms
thread: mutex 0x7faa0401f370 at src/input/mpegts/linuxdvb/linuxdvb_frontend.c:1010 took 1060ms
thread: mutex 0x55e36afe4560 at src/main.c:703 took 2120ms
thread: mutex 0x7faa0401f370 at src/input/mpegts/linuxdvb/linuxdvb_frontend.c:1010 took 1060ms

The exact same VM works fine if I just swap the 6909X for a 6985 so I think this must be either a 6909X driver issue or a compatibility issue with the 6909X and TVH. 

Are there any settings I can try or further information I can provide which might help resolve this?

crazycat69, 2019-05-04:
try increase Status period from default 1000 ms to 2000-5000Ms (in TVH advanced adapter settings).
and update driver with my latest fix
https://github.com/tbsdtv/linux_media/commit/294e2bfd4f5fd8742b457d3f3e6f8bb4297d9d9d

crazycat69, 2019-05-04:
as alternative disable signal statistic report (only lock status) with **nostat** option for _stid135_ module
https://github.com/tbsdtv/linux_media/commit/b2162770ce8b868b71a6ea4c2d9f2320de43ab11

Krobar, 2019-05-05:
Thank you.

Changing the 1 tuner for 5 seconds status delay help but just means the interface is locked up less of the time. Disabling status is better again and the interface is somewhat laggy but actually usable with 4 tuners enabled.  

The TVH status page still struggles even with the new option switch enabled. It looks like bandwidth and other parameters in the status page may be updating too quickly for the web interface to keep up but that is just a guess.

crazycat69, 2019-05-08:
https://github.com/tbsdtv/linux_media/commit/35e13036285523588b90e0153f9af820fb8b74a9

Krobar, 2019-05-11:
Big thank you for the above advice and patch. Its still not as quick as my 6985 but it is working well and the interface is fast enough.

closed 2019-05-11 by Krobar