tbsdtv/linux_media #161: TBS 5990 does not work on kernel 4.19 when both tuners enabled

JurajNyiri, 2019-05-10:
Hello, first let me thank you for awesome work here. 

Since I started using Unraid 6.7.0 RC6, RC7 and RC8 my TBS 5990 stopped working after switching 7-8 channels.
***This also only happens, if I am using both tuners on TBS 5990. If I only use 1 tuner, it works well.***
I also tried a combination of TBS QBOX-S2 enabled while 1 tuner enabled on TBS 5990 which also works fine.
It does not matter which tuner on TBS 5990 is enabled, as far as only 1 is, it works fine.

Unraid 6.7.0 started using kernel 4.19. Tuner works fine on Unraid 6.6.7 which uses kernel 4.18.

- Using latest tvheadend 4.3 via docker, also tried with docker stable 4.2
- Tuner is connected via USB 2
- Restart OS solves the issue: YES
- Restart TVHEADEND docker solves the issue: NO

***Tuners:***

TBS 5990 -> Stops working after a while when using both tuners
TBS QBOX S2 -> Works

***Tuners configuration:***

Tune before DiseqC: YES
DiseqC repeats:0
Full DiseqC: YES
Turn off LNB when idle: NO
Higher LNB voltage: Do not set
 
Enabled: YES
Name: name of the tuner
Over-the-air EPG: YES
Power save: NO
Satellite config: 4-Port switch (universal LNB)
Master tuner: This tuner
Priority: Different priorities per tuner
Streaming priority: 0
Initial scan: YES
Idle scan: YES
Free subscription weight: 10
Linked input: Not linked
Maximum PIDs: 32
Allow all PIDs: YES
tune repeats: 0
Skip initial bytes: 0
Input buffer (bytes): 188000
Status period (ms): 1000
Signal multiplier: 100
SNR multiplier: 100
Force old status: NO

***Sys specs:***

CPU: AMD Ryzen 5 2600 Six-Core @ 3400 MHz, 2 cores dedicated to docker
RAM: 24 GiB DDR4, 22GB free while testing
HDD: 4TB free

***How to replicate:*** 
Enable both tuners on TBS 5990 with the same settings (as stated above).
Switching around 8 channels randomly, waiting around 2-3 sec between a switch - it does not matter which in which order and what channels.

***uname -a:***
root@Tower:~# uname -a
Linux Tower 4.19.37-Unraid #1 SMP Tue May 7 20:44:39 BST 2019 x86_64 AMD Ryzen 5 2600 Six-Core Processor AuthenticAMD GNU/Linux

***Tried:***
- Setting Skip initial bytes: to 1880 and Status period (ms): to 5000
- Searching for the issue resulted in nothing found
- Turning off Allow all PIDs: 
- Setting timeshift to RAM only

SYS LOG 1:
[... 163 lines cut ...]
```
2019-05-11 00:06:02.280 linuxdvb: TurboSight TBS 5990 #2 : DVB-S #0 - failed to config dmx for pid 18 [e=Out of memory]
2019-05-11 00:06:02.280 mpegts: 12090V in 23.5E - tuning on TurboSight TBS 5990 #3 : DVB-S #0
2019-05-11 00:06:02.951 subscription: 0052: "epggrab" subscribing to mux "12090V", weight: 4, adapter: "TurboSight TBS 5990 #3 : DVB-S #0", network: "23.5E", service: "Raw PID Subscription"
2019-05-11 00:06:03.031 linuxdvb: TurboSight TBS 5990 #2 : DVB-S #0 - poll TIMEOUT
2019-05-11 00:06:03.969 linuxdvb: TurboSight TBS 5990 #2 : DVB-S #0 - retune nodata
2019-05-11 00:06:10.689 linuxdvb: TurboSight TBS 5990 #2 : DVB-S #0 - poll TIMEOUT
2019-05-11 00:06:11.050 linuxdvb: TurboSight TBS 5990 #2 : DVB-S #0 - retune nodata
2019-05-11 00:06:12.981 subscription: 0051: service instance is bad, reason: No input detected
2019-05-11 00:06:12.982 mpegts: 12054H in 0.8W - tuning on TurboSight TBS 5990 #2 : DVB-S #0
2019-05-11 00:06:13.502 subscription: 0051: "HTTP" subscribing on channel "Markiza", weight: 100, adapter: "TurboSight TBS 5990 #2 : DVB-S #0", network: "0.8W", mux: "12054H", provider: "Slovak Telekom", service: "Markiza HD", profile="pass", hostname="192.168.100.238", username="username", client="VLC/3.0.4 LibVLC/3.0.4"
2019-05-11 00:06:14.655 linuxdvb: TurboSight TBS 5990 #2 : DVB-S #0 - poll TIMEOUT
2019-05-11 00:06:15.556 linuxdvb: TurboSight TBS 5990 #2 : DVB-S #0 - retune nodata
2019-05-11 00:06:21.953 webui: Stop streaming /stream/channel/411aad88e3579050300041f8c8142192?ticket=74660ab27248b798ea4e288e992be3d9af29b945, timeout waiting for packets
2019-05-11 00:06:21.953 subscription: 0051: "HTTP" unsubscribing from "Markiza", hostname="192.168.100.238", username="username", client="VLC/3.0.4 LibVLC/3.0.4"
2019-05-11 00:06:21.970 http: 192.168.100.238: using ticket 74660ab27248b798ea4e288e992be3d9af29b945 for /stream/channel/411aad88e3579050300041f8c8142192
2019-05-11 00:06:21.970 mpegts: 11973V in 23.5E - tuning on TurboSight TBS 5990 #2 : DVB-S #0
2019-05-11 00:06:22.480 subscription: 0057: "HTTP" subscribing on channel "Markiza", weight: 100, adapter: "TurboSight TBS 5990 #2 : DVB-S #0", network: "23.5E", mux: "11973V", provider: "M7 Group", service: "Markiza HD", profile="pass", hostname="192.168.100.238", username="username", client="VLC/3.0.4 LibVLC/3.0.4"
2019-05-11 00:06:23.623 linuxdvb: TurboSight TBS 5990 #2 : DVB-S #0 - poll TIMEOUT
2019-05-11 00:06:24.498 linuxdvb: TurboSight TBS 5990 #2 : DVB-S #0 - retune nodata
 
...
```

Please let me know how can I help debug the issue further and thank you for any help.

crazycat69, 2019-05-11:
This some related to usb host driver or cx231xx core part.

JurajNyiri, 2019-05-11:
Thank you for quick response, if I understand correctly you are saying this is not related to drivers? Who should I then please report this to?

JurajNyiri, 2019-07-30:
Hello,

has there been any progress on this? Can I help to debug this further? How can I help?

Thank you very much

smallint, 2019-07-31:
I just want to jump in here as I am now also trying to get my TBS5990 up and running on Arch with a Raspberry Pi4. Lets say, it works to some extent. Let me give some historic background:

Some years ago I started with the TBS5900 connected to an iMX6 device and Linux support for ARM devices was zero at that time. I discovered the Github repo from @crazycat69 and started to use that code which worked OK but did not support the 2nd adapter at all and wasn't pretty stable. So we discussed that issues in a very long thread (which is lost unfortunately because the old repo was deleted) and figured out how to enable the second adapter. in my personal setup I got two adapter working reliably. They have been running for several year now with kernel 3.14 (I know).

As the current code does not run with the 2nd adapter (no chance at all) and produces burst sending timeout I ported my old working code to the new kernel (4.19.) just to see if it is related to some updates in this repo. And it did not work with respect to the 2nd adapter. That leads me to the conclusion that things must have changed on the kernel side (probably i2c or Pi4 usb subsystem).

The good thing is I know that it is possible to run both adapters on an ARM device. It is still working in my setup. I will test the current driver on an x86 just to know if it is possible at all to get the second adapter working.

Btw, bulk transfer is pretty unstable also with the first adapter. I enabled ISO and although I get some artifacts while switching channels, everything works stable and without tuning timeouts.

And to come back to your issue: it is hard to say what the underlying issue is without testing. And even testing is difficult as I don't know what to test actually. I am neither a kernel dev nor a hardware expert so I can't tell you where to continue digging. As I understood this driver development it is more of a reverse engineering effort than implementing a spec. Maybe switching to ISO transfer helps in your case. Bulk transfer has shown similar symptoms as yours.

P.S. Why did I write so much? Because I am going to keep working on the ARM issue (more trial and error) and maybe @crazycat69  has some idea what I could test or try to get the 2nd adapter working. It is an update of my current findings and the expression of the desire to solve the issue and to get some help ;)

smallint, 2019-07-31:
I did some further checks and maybe that helps in your case as well. Enabling bulk transfer results in those debug messages from time to time:
```
cx231xx #0 cx231xx_bulk_irq_callback :urb completion error - device is stalled.
```
After some time it recovers and works again for a while. You can enable the debug output as module option for cx231xx:
```
options cx231xx core_debug=1
```

JurajNyiri, 2019-08-01:
Thank you @smallint for your advice

I would try it but I am not sure how I would add this into Unraid since changes are not persistent between reboots. I would probably have to build it somehow on my own and I haven't yet looked into it.

smallint, 2019-08-30:
I have finally solved the issue for my installation with ArchLinux and a RPi4. Rather than compiling the whole V4L tree with this repo I extracted the affected source changes and created a standalone package with just builds the required modules for the host kernel to support the TBS 5990. Now both tuners are working nicely. All the code is in my repository at https://github.com/smallint/tbs-usb.

I tested a lot and at some point the drivers even did not work anymore on my x86 box. After some time I just restarted the TBS box and forced a cold start to reload the firmware. Then it worked again and also the RPi4 worked with both tuners. The nice thing about the stripped down repository is that is takes just a couple of seconds to build after a kernel update. And replacing the entire V4L tree prevented some RPi4 modules to load. This is also resolved. So far I am glad that it works finally. There is still an open issue with USB bulk transfer but that is independent of the TBS drivers. It seems to be a RPi4 kernel issue. I tested the modules with kernel 4.19 and 5.2.

@JurajNyiri maybe a cold boot of the TBS device might also help in your situation.

JurajNyiri, 2019-09-20:
A bit of update. I am trying to migrate to RPI3B+ with latest libreleec with tbs drivers.

1. Installed libreelec
2. Enabled TBS drivers
3. Enabled tvheadend
4. Tuner is found

When trying to use both tuners at the same time, the other one always drops with retune nodata/poll TIMEOUT and the Unknown tuner type configuring SIF.

This is **almost** resolved by putting
```
options cx231xx transfer_mode=1
options disable infrared remote support
```

JurajNyiri, 2019-09-20:
A bit of update. I am trying to migrate to RPI3B+ with latest libreleec with tbs drivers.

1. Installed libreelec
2. Enabled TBS drivers
3. Enabled tvheadend
4. Tuner is found

When trying to use both tuners at the same time, the other one always drops with retune nodata/poll TIMEOUT and the Unknown tuner type configuring SIF.

This is **almost** resolved by putting
```
options cx231xx transfer_mode=1
options disable infrared remote support
```
into the /storage/.config/modprobe.d/cx231xx.conf .

Now, when I stream from both tuners, both streams start artifacting with the ```Continuity counter error``` in tvheadend and no errors in dmesg (even with debug enabled), but they do not drop.