tbsdtv/linux_media #162: Issues with videobuf on latest branch

nrmmota, 2019-05-13:
[    3.095607] frame_vector: module license 'unspecified' taints kernel.
[    3.095614] Disabling lock debugging due to kernel taint
[    3.107636] media: Linux media interface: v0.10
[    3.114631] videodev: Linux video capture interface: v2.00
[    3.114638] WARNING: You are using an experimental version of the media stack.
[    3.114638] 	As the driver is backported to an older kernel, it doesn't offer
[    3.114638] 	enough quality for its usage in production.
[    3.114638] 	Use it with care.
[    3.114638] Latest git patches (needed if you report a bug to linux-media@vger.kernel.org):
[    3.114638] 	35e13036285523588b90e0153f9af820fb8b74a9 stid135: Improved signal status reading.
[    3.114638] 	1390bf188aa5a51a7e6f77095d1a689324f62cc5 stid135: Fix typo in option description.
[    3.114638] 	42a1373e41d9e54922b13d88b9add637c31527a3 change tbs6004 default lock rf to 114M
[    3.182306] videobuf2_core: disagrees about version of symbol v4l_vb2q_enable_media_source
[    3.182313] videobuf2_core: Unknown symbol v4l_vb2q_enable_media_source (err -22)

nonius@dvbip-gateway:~$ ls -l /lib/modules/3.13.0-117-generic/kernel/drivers/media/common/videobuf2
total 212
-rw-r--r-- 1 root root 62024 2019-05-13 11:20 videobuf2-common.ko
-rw-r--r-- 1 root root 49760 2019-05-13 11:20 videobuf2-core.ko
-rw-r--r-- 1 root root 16428 2019-05-13 11:20 videobuf2-dma-contig.ko
-rw-r--r-- 1 root root 14912 2019-05-13 11:20 videobuf2-dma-sg.ko
-rw-r--r-- 1 root root  9828 2019-05-13 11:20 videobuf2-dvb.ko
-rw-r--r-- 1 root root  5648 2019-05-13 11:20 videobuf2-memops.ko
-rw-r--r-- 1 root root 26644 2019-05-13 11:20 videobuf2-v4l2.ko
-rw-r--r-- 1 root root 11268 2019-05-13 11:20 videobuf2-vmalloc.ko
nonius@dvbip-gateway:~$ ls -l /lib/modules/3.13.0-117-generic/kernel/drivers/media/v4l2-core/videobuf*
-rw-r--r-- 1 root root  9828 2019-05-13 11:20 /lib/modules/3.13.0-117-generic/kernel/drivers/media/v4l2-core/videobuf2-dvb.ko
-rw-r--r-- 1 root root 22992 2019-05-13 11:20 /lib/modules/3.13.0-117-generic/kernel/drivers/media/v4l2-core/videobuf2-v4l2.ko
-rw-r--r-- 1 root root 25908 2019-05-13 11:20 /lib/modules/3.13.0-117-generic/kernel/drivers/media/v4l2-core/videobuf-core.ko
-rw-r--r-- 1 root root 12132 2019-05-13 11:20 /lib/modules/3.13.0-117-generic/kernel/drivers/media/v4l2-core/videobuf-dma-contig.ko
-rw-r--r-- 1 root root 16412 2019-05-13 11:20 /lib/modules/3.13.0-117-generic/kernel/drivers/media/v4l2-core/videobuf-dma-sg.ko
-rw-r--r-- 1 root root 12348 2019-05-13 11:20 /lib/modules/3.13.0-117-generic/kernel/drivers/media/v4l2-core/videobuf-dvb.ko
-rw-r--r-- 1 root root  9444 2019-05-13 11:20 /lib/modules/3.13.0-117-generic/kernel/drivers/media/v4l2-core/videobuf-vmalloc.ko

I have duplicate videobufs. They all seemed to have been touched at the same time.

crazycat69, 2019-05-14:
try remove staging media modules before insdtall (see isntall.sh script)
**sudo rm -r -f /lib/modules/$(uname -r)/kernel/drivers/staging/media**

nrmmota, 2019-05-14:
Thanks that helped.
I'm using a build system and then use checkinstall to build a debian package.
It would be helpful to integrate those commands in the make file itself otherwise people like me are skipping Important steps.