tbsdtv/linux_media #163: TBS 6909x Locking into multiple signals

JianSun1, 2019-05-20:
I tried to lock multiple signals with different demodulators using TBS 6909x in Hotbird satellite. After locking into a few transponders, the transport stream output cuts itself and the demodulator never locks to another frequencies unless restarting the computer. I think the bandwith is high and the limit is exceeding. Could you fix this which auto distribute the maximum throughput or limit so i will not have to restart or reload the modules myself?

crazycat69, 2019-05-20:
what dvb soft ? tvheadend ?
maybe need update driver ? some important fix need 35e13036285523588b90e0153f9af820fb8b74a9

Neolo, 2020-05-13:
Confirmed an issue with this card on TVH. After locking to 1 clear UHD HEVC and 1 HD scrambled channels on 2 different transponders (it's just 2 not 8), TVH gets slower, and if tuned a few more - totally unresponsive until reboot. I can't believe I wasted so much money on this card. And no, driver isn't fixed. Older 6909 does NOT have this problem.