tbsdtv/linux_media #164: TBS 5520SE  si2183 5-0067: read_status failed=-11

TRPB, 2019-05-27:
Firstly, I've had this issue before and it seems to be related to tvheadend.

```
[ 1414.013275] si2183 5-0067: Silicon Labs Si2183 successfully attached
[ 1414.018017] si2157 6-0061: Silicon Labs Si2147/2148/2157/2158 successfully attached
[ 1414.018107] i2c i2c-6: av201x: Airoha Technology AV201x successfully attached
[ 1414.018273] usb 1-2: DVB: registering adapter 0 frontend 0 (TurboSight TBS 5520SE DVB-T/T2/C/C2/ISDB-T)...
[ 1414.018539] usb 1-2: DVB: registering adapter 0 frontend 1 (TurboSight TBS 5520SE DVB-S/S2/S2X)...
[ 1414.018753] dvb-usb: TurboSight TBS 5520SE successfully initialized and connected.
[ 1414.018952] usb 1-2: USB disconnect, device number 9
[ 1414.020025] dvb-usb: TurboSight TBS 5520SE successfully deinitialized and disconnected.
[ 1415.671117] usb 1-2: new high-speed USB device number 10 using xhci_hcd
[ 1415.815116] usb 1-2: New USB device found, idVendor=734c, idProduct=5521, bcdDevice= 0.01
[ 1415.815126] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 1415.815131] usb 1-2: Product: TBS 5521
[ 1415.815135] usb 1-2: Manufacturer: TBS-Tech
[ 1415.816462] dvb-usb: found a 'TurboSight TBS 5520SE' in cold state, will try to load a firmware
[ 1415.816472] dvb-usb: downloading firmware from file 'dvb-usb-id5520se.fw'
[ 1415.816478] tbs5520se: start downloading TBS5520se firmware
[ 1415.927789] dvb-usb: found a 'TurboSight TBS 5520SE' in warm state.
[ 1415.927987] dvb-usb: will pass the complete MPEG2 transport stream to the software demuxer.
[ 1415.928549] dvbdev: DVB: registering new adapter (TurboSight TBS 5520SE)
[ 1416.015233] dvb-usb: MAC address: 00:22:ab:c0:4b:01
[ 1416.016800] i2c i2c-5: Added multiplexed i2c bus 6
[ 1416.016802] si2183 5-0067: Silicon Labs Si2183 successfully attached
[ 1416.019995] si2157 6-0061: Silicon Labs Si2147/2148/2157/2158 successfully attached
[ 1416.020016] i2c i2c-6: av201x: Airoha Technology AV201x successfully attached
[ 1416.068038] usb 1-2: DVB: registering adapter 0 frontend 0 (TurboSight TBS 5520SE DVB-T/T2/C/C2/ISDB-T)...
[ 1416.068353] usb 1-2: DVB: registering adapter 0 frontend 1 (TurboSight TBS 5520SE DVB-S/S2/S2X)...
[ 1416.068919] dvb-usb: TurboSight TBS 5520SE successfully initialized and connected.
[ 1438.146975] audit: type=1131 audit(1558959792.853:72): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=tvheadend comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=failed'
[ 1438.195749] audit: type=1130 audit(1558959792.903:73): pid=1 uid=0 auid=4294967295 ses=4294967295 msg='unit=tvheadend comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
[ 1438.253691] si2183 5-0067: found a 'Silicon Labs Si2183-B60'
[ 1438.253713] si2183 5-0067: downloading firmware from file 'dvb-demod-si2183-b60-01.fw'
[ 1438.483868] si2183 5-0067: firmware version: 6.0.2
[ 1438.497938] si2157 6-0061: found a 'Silicon Labs Si2157-A30'
[ 1438.548600] si2157 6-0061: firmware version: 3.0.5
[ 1438.548624] usb 1-2: DVB: adapter 0 frontend 0 frequency 0 out of range (174000000..862000000)
[ 1438.754610] usb 1-2: DVB: adapter 0 frontend 1 frequency 0 out of range (950000..2150000)
[ 1442.654638] si2183 5-0067: read_status failed=-11
[ 1442.858006] si2183 5-0067: read_status failed=-11
[ 1443.061333] si2183 5-0067: read_status failed=-11
[ 1443.264641] si2183 5-0067: read_status failed=-11
[ 1443.467901] si2183 5-0067: read_status failed=-11
[ 1443.671273] si2183 5-0067: read_status failed=-11
[ 1443.874646] si2183 5-0067: read_status failed=-11
[ 1444.077996] si2183 5-0067: read_status failed=-11
[ 1444.281289] si2183 5-0067: read_status failed=-11
[ 1444.484653] si2183 5-0067: read_status failed=-11
[ 1444.687988] si2183 5-0067: read_status failed=-11
[ 1444.894596] si2183 5-0067: read_status failed=-11
[ 1445.097984] si2183 5-0067: read_status failed=-11
[ 1445.301270] si2183 5-0067: read_status failed=-11
[ 1445.504617] si2183 5-0067: read_status failed=-11
[ 1445.707943] si2183 5-0067: read_status failed=-11
```

In previous kernel's, I'd get this same issue on boot and `sudo systemctl restart tvheadend` would fix it. I cannot get that workaround to work any more. Restarting tvheadend makes no difference, nor does plugging in the tuner after tvheadend is running.

No errors appear in dmesg until I visit the tvheadend web interface. The adapter shows but I can't use it to scan or watch channels.

crazycat69, 2019-05-27:
b8fd38aafc2fccc039496f67bff8967436b4d7c2

and update tuner/demod firmwares from https://github.com/tbsdtv/media_build/releases/download/latest/dvb-firmwares.tar.bz2

TRPB, 2019-05-27:
thanks again for the very speedy response and fix! Seems to have worked perfectly.

TRPB, 2019-06-06:
While this has made the problem less frequent, I am still getting it and need to restart tvheadend periodically. It seems to be tuning channels which occasionally breaks it and causes this error to appear. I have to manually restart tvheadend to fix it, then I can watch TV again.

crazycat69, 2019-06-08:
maybe "Power save" enabled for DVB adapter in TVH ?

TRPB, 2019-06-18:
No, it's not turned on. The issue seems to happen on boot most times but restarting tvheadend usually fixes it.

Sometimes it gets completely stuck and I have to unplug the tuner from the power and reconnect it before it's ok.

TRPB, 2019-07-12:
At the moment even restarting tvheadend doesn't fix it, I've tried unplugging and reconnecting the tuner but as soon as I start tvheadend I get `read_status failed=-11`. The only thing that will fix it is rebooting the PC.

lucky79, 2019-09-22:
I am having the same problem, but it randomly appears on tvheadend restart, sometimes another tvh restart fixes it, sometimes I need to reboot the machine to get it working

alivardar, 2019-12-17:
Any news about this problem, It is locking our systems on reboot.

Znuff, 2020-05-05:
We have also started seeing this.

We DO NOT use `tvheadend`:

```
May  5 07:13:32 iptv kernel: [36586.230499] TBSECP3 driver 0000:01:00.0: i2c xfer timeout
May  5 07:13:32 iptv kernel: [36586.230871] si2183 19-0067: read_status fe0 cmd_exec failed=-5
May  5 07:13:32 iptv kernel: [36586.231575] si2183 19-0067: read_status failed=-5

[...]

May  5 07:13:54 iptv kernel: [36607.894422] TBSECP3 driver 0000:01:00.0: i2c xfer timeout
May  5 07:13:55 iptv kernel: [36609.110543] TBSECP3 driver 0000:01:00.0: i2c xfer timeout
May  5 07:13:56 iptv kernel: [36610.134439] TBSECP3 driver 0000:01:00.0: i2c xfer timeout
May  5 07:13:57 iptv kernel: [36611.158426] TBSECP3 driver 0000:01:00.0: i2c xfer timeout
May  5 07:13:59 iptv kernel: [36612.396844] si2183 19-0064: set_params failed=-11
May  5 07:13:59 iptv kernel: [36612.598409] si2183 19-0064: read_status failed=-11
May  5 07:13:59 iptv kernel: [36612.802417] si2183 19-0064: read_status failed=-11
May  5 07:14:00 iptv kernel: [36613.401943] si2183 19-0064: read_status failed=-11
May  5 07:14:00 iptv kernel: [36613.506419] si2183 19-0064: read_status failed=-11
May  5 07:14:00 iptv kernel: [36613.914447] si2183 19-0064: read_status failed=-11
May  5 07:14:01 iptv kernel: [36614.730513] si2183 19-0064: read_status failed=-11
May  5 07:14:01 iptv kernel: [36614.934412] si2183 19-0064: read_status failed=-11
May  5 07:14:02 iptv kernel: [36615.405864] si2183 19-0064: read_status failed=-11
May  5 07:14:02 iptv kernel: [36615.550409] si2183 19-0064: read_status failed=-11
May  5 07:14:02 iptv kernel: [36615.958411] si2183 19-0064: read_status failed=-11
May  5 07:14:03 iptv kernel: [36616.407766] si2183 19-0064: read_status failed=-11
May  5 07:14:03 iptv kernel: [36616.774413] si2183 19-0064: read_status failed=-11
May  5 07:14:03 iptv kernel: [36617.182405] si2183 19-0064: read_status failed=-11
May  5 07:14:04 iptv kernel: [36617.386397] si2183 19-0064: read_status failed=-11
May  5 07:14:04 iptv kernel: [36617.409516] si2183 19-0064: read_status failed=-11
May  5 07:14:04 iptv kernel: [36617.613738] clocksource: timekeeping watchdog on CPU6: Marking clocksource 'tsc' as unstable because the skew is too large:
May  5 07:14:04 iptv kernel: [36617.613753] clocksource:                       'hpet' wd_now: 128a6913 wd_last: 12007a9f mask: ffffffff
May  5 07:14:04 iptv kernel: [36617.613755] clocksource:                       'tsc' cs_now: 6a77122386c0 cs_last: 6a76b6c36420 mask: ffffffffffffffff
May  5 07:14:04 iptv kernel: [36617.613759] tsc: Marking TSC unstable due to clocksource watchdog
``

This leads to a full system crash, only way to fix it is by rebooting the machine.

crazycat69, 2020-05-05:
what cards installed ?

Znuff, 2020-05-05:
We have 2 x **TBS6209** installed in the same system, under Ubuntu 18.04.

This worked fine for about ~2 weeks, we didn't change kernel and/or drivers (we have held the kernel for upgrades), and suddenly we got the above issue 2 times in less than 24 hours -- complete system crash, had to power-cycle.

We have currently decided to swap to a 5.x kernel (`5.3.0-51-generic`, part of `linux-generic-hwe-18.04`) and we have compiled the latest drivers as of today. 

I'll update ticket if this happens again.

jogrue, 2020-05-15:
I just had the same issue mentioned by @TRPB in his first post on the TBS 5520SE, running OpenMediaVault (Debian 10) with Tvheadend. I have it connected to satellite and cable, so maybe it is something with switching between modes. It all worked fine, until, I guess, I unplugged the TBS 5520SE device and plugged it in again (but maybe that was not the reason, could also be caused by a reboot, kernel update, or switching through channels too quickly).

In dmesg the error looks like the one above, in the end it's just along list of `si2183 4-0067: read_status failed=-11` messages with an occasional `si2183 4-0067: set_params failed=-11`:

```
[   74.642765] usb 1-5: new high-speed USB device number 9 using xhci_hcd
[   74.667568] usb 1-5: New USB device found, idVendor=734c, idProduct=5521, bcdDevice= 0.01
[   74.674192] usb 1-5: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[   74.680782] usb 1-5: Product: TBS 5521
[   74.683399] usb 1-5: Manufacturer: TBS-Tech
[   74.686538] dvb-usb: found a 'TurboSight TBS 5520SE' in cold state, will try to load a firmware
[   74.688871] dvb-usb: downloading firmware from file 'dvb-usb-id5520se.fw'
[   74.691250] tbs5520se: start downloading TBS5520se firmware
[   74.806772] dvb-usb: found a 'TurboSight TBS 5520SE' in warm state.
[   74.811243] dvb-usb: will pass the complete MPEG2 transport stream to the software demuxer.
[   74.815966] dvbdev: DVB: registering new adapter (TurboSight TBS 5520SE)
[   74.932162] dvb-usb: MAC address: 00:22:ab:c0:50:4a
[   74.941130] i2c i2c-4: Added multiplexed i2c bus 5
[   74.945550] si2183 4-0067: Silicon Labs Si2183 successfully attached
[   74.955024] si2157 5-0061: Silicon Labs Si2147/2148/2157/2158 successfully attached
[   74.957917] i2c i2c-5: av201x: Airoha Technology AV201x successfully attached
[   75.009014] usb 1-5: DVB: registering adapter 0 frontend 0 (TurboSight TBS 5520SE DVB-T/T2/C/C2/ISDB-T)...
[   75.012846] usb 1-5: DVB: registering adapter 0 frontend 1 (TurboSight TBS 5520SE DVB-S/S2/S2X)...
[   75.017072] dvb-usb: TurboSight TBS 5520SE successfully initialized and connected.
[  390.050128] si2183 4-0067: found a 'Silicon Labs Si2183-B60'
[  390.052630] si2183 4-0067: firmware: direct-loading firmware dvb-demod-si2183-b60-01.fw
[  390.055139] si2183 4-0067: downloading firmware from file 'dvb-demod-si2183-b60-01.fw'
[  390.760528] si2183 4-0067: firmware version: 6.0.13
[  390.781983] si2157 5-0061: found a 'Silicon Labs Si2157-A30'
[  390.837807] si2157 5-0061: firmware version: 3.0.5
[  390.843588] usb 1-5: DVB: adapter 0 frontend 0 frequency 0 out of range (174000000..862000000)
[  391.577328] usb 1-5: DVB: adapter 0 frontend 1 frequency 0 out of range (950000..2150000)
[  391.976397] si2183 4-0067: set_params failed=-11
[  392.010842] si2183 4-0067: read_status failed=-11
[  392.181268] si2183 4-0067: read_status failed=-11
[  392.389265] si2183 4-0067: read_status failed=-11
[  392.593244] si2183 4-0067: read_status failed=-11
[  392.805262] si2183 4-0067: read_status failed=-11
...
```

In TVHeadend the error looks like this:

```
> journalctl | grep "tvheadend"

May 15 00:18:20 nas.local tvheadend[2200]: mpegts: 386MHz in Magenta - tuning on TurboSight TBS 5520SE DVB-T/T2/C/C2/ISDB-T #0 : DVB-C #0
May 15 00:18:20 nas.local tvheadend[2200]: subscription: 0008: "scan" subscribing to mux "386MHz", weight: 5, adapter: "TurboSight TBS 5520SE DVB-T/T2/C/C2/ISDB-T #0 : DVB-C #0", network: "Magenta", service: "Raw PID Subscription"
May 15 00:18:20 nas.local tvheadend[2200]: linuxdvb: TurboSight TBS 5520SE DVB-T/T2/C/C2/ISDB-T #0 : DVB-C #0 - FE_READ_STATUS error Resource temporarily unavailable
May 15 00:18:21 nas.local tvheadend[2200]: linuxdvb: TurboSight TBS 5520SE DVB-T/T2/C/C2/ISDB-T #0 : DVB-C #0 - FE_READ_STATUS error Resource temporarily unavailable
May 15 00:18:22 nas.local tvheadend[2200]: linuxdvb: TurboSight TBS 5520SE DVB-T/T2/C/C2/ISDB-T #0 : DVB-C #0 - FE_READ_STATUS error Resource temporarily unavailable
May 15 00:18:23 nas.local tvheadend[2200]: linuxdvb: TurboSight TBS 5520SE DVB-T/T2/C/C2/ISDB-T #0 : DVB-C #0 - FE_READ_STATUS error Resource temporarily unavailable
May 15 00:18:24 nas.local tvheadend[2200]: linuxdvb: TurboSight TBS 5520SE DVB-T/T2/C/C2/ISDB-T #0 : DVB-C #0 - FE_READ_STATUS error Resource temporarily unavailable
May 15 00:18:25 nas.local tvheadend[2200]: mpegts: 386MHz in Magenta - scan no data, failed
May 15 00:18:25 nas.local tvheadend[2200]: subscription: 0008: "scan" unsubscribing
```

I tried kernels 5.5.0-0.bpo.2-amd64 and 4.19.0-8-amd64, removed the device, switched USB ports, paused tvheadend, rebuild drivers, installed firmware from GitHub (above link), and rebooted of course ... nothing could fix it.

I then disabled the DVB-C adapter in TVHeadend, here:
[image]

Afterward, everything worked fine and tvheadend scanned all muxes on the DVB-S adapter. I then re-enabled the DVB-C adapter, too. For now, it appears to work normally.

I will report back with some logs, if the issue appears again. These are some other places where I found references to this problem:

https://www.tbsdtv.com/forum/viewtopic.php?f=194&t=25007
https://www.tbsdtv.com/forum/viewtopic.php?f=194&t=25022
https://tvheadend.org/boards/4/topics/35755?r=39491

Logs:
[dmesg_error_reconnect_error.txt](https://github.com/tbsdtv/linux_media/files/4633083/dmesg_error_reconnect_error.txt)
[journalctl-tvheadend.txt](https://github.com/tbsdtv/linux_media/files/4633086/journalctl-tvheadend.txt)

jogrue, 2020-06-08:
It happened again now, after an update/reboot. Disabling DVB-C now led to these errors in TVHeadend:

```
2020-06-08 20:16:49.865 mpegts: 12480V in Astra - tuning on TurboSight TBS 5520SE DVB-S/S2/S2X #0 : DVB-S #1
2020-06-08 20:16:50.369 diseqc: failed to disable tone
2020-06-08 20:16:50.369 mpegts: 12551.5V in Astra - tuning on TurboSight TBS 5520SE DVB-S/S2/S2X #0 : DVB-S #1
2020-06-08 20:16:50.873 diseqc: failed to disable tone
```

dmesg again shows the read_status failed error
```
[  +0.204003] si2183 1-0067: read_status failed=-11
[  +0.204001] si2183 1-0067: read_status failed=-11
[  +0.204001] si2183 1-0067: read_status failed=-11
```

jogrue, 2020-06-22:
Okay, now my kernel was updated to 5.6.0. Which, by the way, meant rebuilding drivers, again. At first, this did not work, I then had to follow advice from a "closed" GitHub issue using sources from May 22. At least, this time, with kernel 5.6 this worked (after I added a few sudos in the right places).

In the meantime, I also installed an old USB DVB-T device, so the TBS 5520SE does not have to switch between USB-T and USB-C mode (which I thought was the reason for the error).

And then again, with only the USB-C device enabled: failed to disable tone and read_status failed=-11. This time it stopped after disabling the device in tvheadend, restarting tvheadend, and re-enabling the device again.

From my side, I am now done with TBS devices. Since this error was around for years, also with different devices and even without tvheadend, I would not hope it will get fixed. I am not willing to play around for hours after every reboot or kernel update or driver update or whatever.

lucky79, 2020-06-22:
I am on the same boat, TBS should refund us or at least replace the unusable device with a working one, even different model, but for the same price. This tuner is not cheap and having it laying in the locker doesn't make any sense spending so much money for it.

I have this problem since the beginning with this device and never had a real use for it because of this problem.

Developers, what would you say?

tbsiptv, 2020-06-23:
Hi All 

TBS always provide good service for all of our customers, if you think that the device can not make you satisfy , we can provide exchange ,return or refund . please direct contact our support@tbsdtv.com, we will not let you down.

Thanks

Kind Regards

mohsh86, 2020-09-21:
Problem still there Sep 2020, contacted support. when told them I am running Linux, no response.

mohsh86, 2020-09-23:
> Hi All
> 
> TBS always provide good service for all of our customers, if you think that the device can not make you satisfy , we can provide exchange ,return or refund . please direct contact our [support@tbsdtv.com](mailto:support@tbsdtv.com), we will not let you down.
> 
> Thanks
> 
> Kind Regards

I've contact your support that provides "good service for all customers" and they suggested that the device is not compatible with USB 3.0 ports. What a joke.

lucky79, 2020-09-23:
Hi,

 

Actually I did that and support only wanted to guide me how to use the device properly. I know how to use it. I am a linux user, so not sure why would they think I am noob who cannot connect the device properly via different USBs.

 

I can say its a joke too. Id like to get the device replaced with a PCIe card for the same value or I am willing to pay the difference. However I want the non stable device replaced with one I can really use. 

 

Right now this device means my $80 were just thrown away.

 

Thanks

Best regards

Lukas

 

 

 <http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=emailclient> 

Bez virů.  <http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=emailclient> www.avg.com

mohsh86, 2020-09-24:
If you try your luck with Sundtek, i initially had a faulty Sundtek SkyTV Ultimate 8 and got a replacement which works so far so good. The Sundtek is a hassle free driver setup, just run a .sh file and bang. Kernel independent, you don't have to re-compile your driver every time there is a kernel update.

http://www.sundtek.com/

Davin622, 2020-09-24:
Hi All
The main cause of the problem is the lack of power supply.
TBS5520SE is a portable usb tuner. so it is designed to have no external power source, and it relies on usb power. 
Firstly, it can not connect the large load, such as motor, large power LNB ,etc. because USB max power is 5W (usb 2.0 2.5W usb3.0 5W). especially in DVBS2 mode, the tuner and demod chip will need more power. and it need supply the LNB voltage(13V or 18V).  pleae use the original data cable.
Second, check your system power management setting, select the high -performance is better.or you can try setting your BIOS of the usb power(some MB support it).
Third,the black usb connector need connect the usb2.0, because tbs5520se use cypress's relatively old usb2.0 chip. so possibly Your MB's usb 3.0 port not good enough compatibility with it.
Last, if you still have problem,please contact to TBS(support@tbsdtv.com) .TBS will help you get refund or change to other products.
 
Best regards

closed 2020-09-24 by Davin622

lucky79, 2020-09-24:
I know how to properly connect it to enough power! And I am not using any motor and just a single LNB. I have tried different motherboard and different settings, nothing helps and the tuner is unstable at all times.

 

Connecting the original cable to two different USB ports. The pink one to USB3 as it can deliver up to 1Amp instead of USB2 which can only provide 500mA.

 

I am well ware about this and its still not stable even for DVB-T2 only!

 

Last, if you still have problem,please contact to TBS(support@tbsdtv.com <mailto:support@tbsdtv.com> ) .TBS will help you get refund or change to other products.

Best regards

 

 <http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=emailclient> 

Bez virů.  <http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=emailclient> www.avg.com

ahasbini, 2022-08-07:
For some reason I did not face this problem when I used the source binary provided by TBS, though I have to say one of the consequences when installing from the source binary was the gnome desktop would no longer start on my Ubuntu 20.04. Then at one point the kernel got updated so sort of made some kind of a reset and gnome desktop started working again. I did the other approach which is cloning from the git repositories, and it worked better because it didn't break gnome desktop. But the problem now is I'm facing the errors appearing in dmesg, which never faced with the first approach. I'll retry the first approach with source binary and share an update.

dotinsight, 2022-08-09:
Hi fellows, I'm facing exactly the same issue with tbs6209.
Can someone from tbs provide advice on the most suitable distro/distro version/kernel version for a production grade machine with tbs6209 installed, I do prefer to use the closed source binary drivers that you provide...

I read tons of posts on tbs forums, with messages like tested on kernel version x.y.z, try kernel < X, ... 

On windows, I've no issues, but it was only a POC.
As you know, your target is linux, not windows.

I'm a developer with 30 years of background, but it's completely non-sense to spend days just to make a commercial device to work as expected.... as a customer/user we need to focus on our own work, not on yours.

Davin622, 2022-08-10:
@dotinsight 
i use the ubuntn kernel 5.10.63. it can work fine with the 6209 in TVH and dvblast. 
what is the system you used? what is the servers hardware info.?
BTW. please let your system work in performance mode, not the powersave mode.

Best Regards

crazycat69, 2022-08-10:
> Hi fellows, I'm facing exactly the same issue with tbs6209. Can someone from tbs provide advice on the most suitable distro/distro version/kernel version for a production grade machine with tbs6209 [...]

Now [tbsdtv/media_build](https://github.com/tbsdtv/media_build) compatible with 4.4 - 5.18 kernels. Better use distribs with LTS kernels (see kernel.org list). I test media_build with ubuntu 16.04 with 4.4, 4.15 kernels and ubuntu 20.04 with 5.4 and 5.15.

Just insert 6209, no any similar issues.

dotinsight, 2022-08-10:
@crazycat69, I do thank you for the promptly reply.
I will try with that specific versions.

crazycat69, 2022-08-10:
Closed-source linux-tbs-drivers not supported since 2016 (kernels < 4.10) and 6209 supported only in open-source drivers (and many multi-standard si2183-based devices)

If you have some issues with card - provide kernel log and more info about used dvb soft.

signalquest-zz, 2023-01-08:
Hi, all.  I am having this described issue using a 5520SE using TVHeadend and UpdateDVB.  I have tried using UDL kernel.

Should I need the TBS driver if using the UDL kernel and v4L?  What is my best choice on what to use?  Thanks.

[ 1511.887698] si2183_cmd_execute_unlocked: Read: failed=-110
[ 1511.887708] si2183_cmd_execute_unlocked: W: 8a 00 00 00 00 00
[ 1511.887711] si2183_cmd_execute_unlocked: R: 00 08 00
[ 1511.887714] si2183_read_signal_strength: err getting RF strength
[ 1512.375581] kworker/dying (6707) used greatest stack depth: 11992 bytes left