tbsdtv/linux_media #165: TBS6814 "NO USEABLE TERRESTRIAL CARD FOUND"

jobrien2001, 2019-06-25:
Hello, 

I am trying to scan for frequencies for this adapter and get the following error in w_scan.

`
w_scan version 20170107 (compiled for DVB API 5.10)
using settings for BRAZIL
ISDB-T (SBTVD)
ISDB-T, BR
scan type TERRESTRIAL, channellist 11
output format vdr-2.0
WARNING: could not guess your codepage. Falling back to 'UTF-8'
output charset 'UTF-8', use -C <charset> to override
Info: using DVB adapter auto detection.
        /dev/dvb/adapter1/frontend0 -> "Turbosight TBS 6814 (Quad ISDB-T)" doesnt support TERRESTRIAL -> SEARCH NEXT ONE.
        /dev/dvb/adapter2/frontend0 -> "Turbosight TBS 6814 (Quad ISDB-T)" doesnt support TERRESTRIAL -> SEARCH NEXT ONE.
main:4007: FATAL: ***** NO USEABLE TERRESTRIAL CARD FOUND. *****
Please check wether dvb driver is loaded and
verify that no dvb application (i.e. vdr) is running.
`

cooliobr, 2019-07-23:
use a versão 170310

jobrien2001, 2019-07-23:
@cooliobr Do you own this card?  I am going to try that version you suggest.  I also experience some signal problems.

cooliobr, 2019-07-23:
try this link 
wget http://www.tbsdtv.com/download/document/linux/media_build-2018-05-25.tar.bz2

crazycat69, 2019-07-23:
look like w_scan have buggy ISDB-T support.
scan type detection for DVB API 5.5+ in fe_supports_scan() @ scan.c

crazycat69, 2019-07-24:
Try w_scan with my little fix for ISDB frontend detection
https://github.com/tbsdtv/w_scan

jobrien2001, 2019-07-25:
Thank you.

Using command  "w_scan -c BR" the tuners were recognized and after scanning for a couple of minutes ended with :
```
ERROR: Sorry - i couldn't get any working frequency/transponder
 Nothing to scan!!
```

Through TVH, I am also experiencing slow/fail tuning and continuity problems.  Once a channel is tuned it usually works fine but otherwise I have issues.  Not sure if its a faulty card.

Any ideas?

crazycat69, 2019-07-25:
What Strength, SNR on TVH ? look like frontend driver mtv23x provide correct extended signal statistic in dBm, dB.

jobrien2001, 2019-07-31:
@crazycat69 I have tuned a few times and copies some results here,  Its very inconsistent.  Tuning is sometimes successful at the beginning sometimes after a few tries sometimes not at all.  I have no problems with built in tuners in TV, only this card.

```
SNR 45 STRENGTH -95
2019-07-31 16:25:27.949 subscription: 0005: "HTTP" subscribing on channel "America HD", weight: 100, adapter: "Turbosight TBS 6814 (Quad ISDB-T) #1 : ISDB-T #0", network: "antena", mux: "533.142MHz", profile="pass", hostname="10.0.4.5", client="VLC/3.0.7.1 LibVLC/3.0.7.1"
2019-07-31 16:25:28.950 subscription: 0004: "HTTP" unsubscribing from "ATV+ HD", hostname="10.0.4.5", client="VLC/3.0.7.1 LibVLC/3.0.7.1"
2019-07-31 16:25:32.492 TS: antena/533.142MHz/{PMT:257} Transport error indicator (total 1)
2019-07-31 16:25:32.492 TS: antena/533.142MHz/{PMT:257}: PCR @ #272 Continuity counter error (total 1)
2019-07-31 16:25:32.492 TS: antena/533.142MHz/{PMT:257}: H264 @ #273 Continuity counter error (total 1)
2019-07-31 16:25:40.030 TS: antena/533.142MHz/{PMT:257}: AAC @ #274 Continuity counter error (total 1)
2019-07-31 16:25:42.013 tbl-pass: pass-pat: -: invalid checksum (len 28, errors 1)
2019-07-31 16:25:42.955 TS: antena/533.142MHz/{PMT:257} Transport error indicator (total 126)
2019-07-31 16:25:42.955 TS: antena/533.142MHz/{PMT:257}: H264 @ #273 Continuity counter error (total 81)
2019-07-31 16:25:42.955 TS: antena/533.142MHz/{PMT:257}: PCR @ #272 Continuity counter error (total 4)
2019-07-31 16:25:43.599 tbl-pass: pass-pmt: -: invalid checksum (len 58, errors 1)
```

```
SNR 30 SIGNAL -95
2019-07-31 16:26:34.048 subscription: 0006: "HTTP" subscribing on channel "Latina HD", weight: 100, adapter: "Turbosight TBS 6814 (Quad ISDB-T) #1 : ISDB-T #0", network: "antena", mux: "509.142MHz", provider: "Latina", service: "Latina HD", profile="pass", hostname="10.0.4.5", client="VLC/3.0.7.1 LibVLC/3.0.7.1"
2019-07-31 16:26:40.051 subscription: 0006: service instance is bad, reason: No input detected
2019-07-31 16:26:42.051 subscription: 0006: No input source available for subscription "HTTP" to channel "Latina HD"
2019-07-31 16:26:42.051 webui: Couldn't start streaming /stream/channelid/788999024?profile=pass, No input detected
2019-07-31 16:26:42.051 subscription: 0006: "HTTP" unsubscribing from "Latina HD", hostname="10.0.4.5", client="VLC/3.0.7.1 LibVLC/3.0.7.1"
2019-07-31 16:26:42.054 mpegts: 509.142MHz in antena - tuning on Turbosight TBS 6814 (Quad ISDB-T) #1 : ISDB-T #0
2019-07-31 16:26:42.054 subscription: 0007: "HTTP" subscribing on channel "Latina HD", weight: 100, adapter: "Turbosight TBS 6814 (Quad ISDB-T) #1 : ISDB-T #0", network: "antena", mux: "509.142MHz", provider: "Latina", service: "Latina HD", profile="pass", hostname="10.0.4.5", client="VLC/3.0.7.1 LibVLC/3.0.7.1"
2019-07-31 16:26:45.787 TS: antena/509.142MHz/Latina HD Transport error indicator (total 1)
2019-07-31 16:26:45.787 TS: antena/509.142MHz/Latina HD: H264 @ #4160 Continuity counter error (total 1)
2019-07-31 16:26:45.787 TS: antena/509.142MHz/Latina HD: AAC @ #4164 Continuity counter error (total 1)
2019-07-31 16:26:45.787 TS: antena/509.142MHz/Latina HD: AAC @ #4162 Continuity counter error (total 1)
2019-07-31 16:26:45.824 tbl-pass: pass-pat: -: invalid checksum (len 32, errors 1)
2019-07-31 16:26:46.516 tbl-pass: pass-pmt: -: invalid checksum (len 40, errors 1)
```

```
SNR 44 SIGNAL -105
2019-07-31 16:30:08.694 mpegts: 497.142MHz in antena - tuning on Turbosight TBS 6814 (Quad ISDB-T) #0 : ISDB-T #0
2019-07-31 16:30:08.694 subscription: 000A: "HTTP" subscribing on channel "ATV+ HD", weight: 100, adapter: "Turbosight TBS 6814 (Quad ISDB-T) #0 : ISDB-T #0", network: "antena", mux: "497.142MHz", profile="pass", hostname="10.0.4.5", client="VLC/3.0.7.1 LibVLC/3.0.7.1"
2019-07-31 16:30:08.720 subscription: 0009: "HTTP" unsubscribing from "ATV+ HD", hostname="10.0.4.5", client="VLC/3.0.7.1 LibVLC/3.0.7.1"
2019-07-31 16:30:14.720 subscription: 000A: service instance is bad, reason: No input detected
2019-07-31 16:30:14.720 mpegts: 497.142MHz in antena - tuning on Turbosight TBS 6814 (Quad ISDB-T) #1 : ISDB-T #0
2019-07-31 16:30:14.720 subscription: 000A: "HTTP" subscribing on channel "ATV+ HD", weight: 100, adapter: "Turbosight TBS 6814 (Quad ISDB-T) #1 : ISDB-T #0", network: "antena", mux: "497.142MHz", profile="pass", hostname="10.0.4.5", client="VLC/3.0.7.1 LibVLC/3.0.7.1"
2019-07-31 16:30:20.721 subscription: 000A: service instance is bad, reason: No input detected
2019-07-31 16:30:22.721 subscription: 000A: No input source available for subscription "HTTP" to channel "ATV+ HD"
2019-07-31 16:30:22.721 webui: Couldn't start streaming /stream/channelid/1396955143?ticket=f52c6150a4ebf9f4c2b5640b393d5214e5f63844&profile=pass, No input detected
2019-07-31 16:30:22.721 subscription: 000A: "HTTP" unsubscribing from "ATV+ HD", hostname="10.0.4.5", client="VLC/3.0.7.1 LibVLC/3.0.7.1"
2019-07-31 16:30:22.723 mpegts: 497.142MHz in antena - tuning on Turbosight TBS 6814 (Quad ISDB-T) #1 : ISDB-T #0
2019-07-31 16:30:22.723 subscription: 000B: "HTTP" subscribing on channel "ATV+ HD", weight: 100, adapter: "Turbosight TBS 6814 (Quad ISDB-T) #1 : ISDB-T #0", network: "antena", mux: "497.142MHz", profile="pass", hostname="10.0.4.5", client="VLC/3.0.7.1 LibVLC/3.0.7.1"
2019-07-31 16:30:26.153 TS: antena/497.142MHz/{PMT:258} Transport error indicator (total 1)
2019-07-31 16:30:26.153 TS: antena/497.142MHz/{PMT:258}: H264 @ #289 Continuity counter error (total 1)
[... cut ...] "VLC/3.0.7.1 LibVLC/3.0.7.1"
2019-07-31 16:34:37.005 subscription: 0014: service instance is bad, reason: No input detected
2019-07-31 16:34:39.005 subscription: 0014: No input source available for subscription "HTTP" to channel "Latina HD"
2019-07-31 16:34:39.005 webui: Couldn't start streaming /stream/channelid/788999024?profile=pass, No input detected
2019-07-31 16:34:39.005 subscription: 0014: "HTTP" unsubscribing from "Latina HD", hostname="10.0.4.5", client="VLC/3.0.7.1 LibVLC/3.0.7.1"
```

crazycat69, 2019-07-31:
So SNR ~30 dB ?

jobrien2001, 2019-08-01:
Yes 28-45 depending on the channel.