tbsdtv/linux_media #167: TBS 6285 & 6205

bpbastos, 2019-07-15:
I'm trying to put to work those two cards using the latest drives but can't get it to work .

The modules are loaded correctly acording to dmesg:

`[   14.129608] SAA716x Budget 0000:04:00.0: DVB: registering adapter 1 frontend 0 (TurboSight TBS 6285 DVB-T/T2/C)...
[   14.133780] TBSECP3 driver 0000:06:00.0: DVB: registering adapter 0 frontend 0 (TurboSight TBS 6205 DVB-T/T2/C )...
[   14.145153] SAA716x Budget 0000:04:00.0: DVB: registering adapter 2 frontend 0 (TurboSight TBS 6285 DVB-T/T2/C)...
[   14.160856] SAA716x Budget 0000:04:00.0: DVB: registering adapter 4 frontend 0 (TurboSight TBS 6285 DVB-T/T2/C)...
[   14.218497] SAA716x Budget 0000:04:00.0: DVB: registering adapter 5 frontend 0 (TurboSight TBS 6285 DVB-T/T2/C)...
[   14.564970] TBSECP3 driver 0000:06:00.0: DVB: registering adapter 3 frontend 0 (TurboSight TBS 6205 DVB-T/T2/C )...
[   14.664743] TBSECP3 driver 0000:06:00.0: DVB: registering adapter 6 frontend 0 (TurboSight TBS 6205 DVB-T/T2/C )...
[   14.805377] TBSECP3 driver 0000:06:00.0: DVB: registering adapter 7 frontend 0 (TurboSight TBS 6205 DVB-T/T2/C )...`

I'm using tvheadend as backend but it can`t lock any signal.

crazycat69, 2019-07-15:
need fix adapter enumeration per driver

**options tbsecp3 adapter_nr=0,1,2,3
options saa716x_core adapter_nr=4,5,6,7**

bpbastos, 2019-07-16:
> need fix adapter enumeration per driver
> 
> **options tbsecp3 adapter_nr=0,1,2,3 options saa716x_core adapter_nr=4,5,6,7**
> 
> this example second option for saa716x_core optional.

thank you @crazycat69 but unfortunately it didn't work. 

Trying to make it more simple, I have let only the tbs 6285 on the server and still can't get signal.

My setup:
Debian
Kernel 5.0.15
Tvheadend 4.3

closed 2019-07-17 by bpbastos