tbsdtv/linux_media #168: TBS 6285 not working

bpbastos, 2019-07-17:
I've closed the other issue to elaborate a little more about the problem.

Just to explain my setup, I'm running tvheadend inside a LXC container and passing the dvb cards to it, the driver and firmware are installed on the host machine - Proxmox (Debian), this setup worked as expected using ljalves drivers for a long team. Along this time I've tried to use the new open source drivers but I've never been able to put it to work, no mattering the kernel or distro, so I always ended up using ljalves drivers.

I was forced to upgrade my distro and therefore the kernel was also upgraded to 5.x. So I have no options besides give the tbs's open source drives an new try. 

I was able to build the latest driver and the modules are loaded correctly acording to dmesg:

`[   13.468276] SAA716x Budget 0000:04:00.0: DVB: registering adapter 0 frontend 0 (TurboSight TBS 6285 DVB-T/T2/C)...
[   13.481787] SAA716x Budget 0000:04:00.0: DVB: registering adapter 1 frontend 0 (TurboSight TBS 6285 DVB-T/T2/C)...
[   13.493918] SAA716x Budget 0000:04:00.0: DVB: registering adapter 2 frontend 0 (TurboSight TBS 6285 DVB-T/T2/C)...
[   13.506040] SAA716x Budget 0000:04:00.0: DVB: registering adapter 3 frontend 0 (TurboSight TBS 6285 DVB-T/T2/C)...
[   66.333380] SAA716x Budget 0000:04:00.0: DVB: adapter 3 frontend 0 frequency 0 out of range (48000000..870000000)
[   67.450111] SAA716x Budget 0000:04:00.0: DVB: adapter 2 frontend 0 frequency 0 out of range (48000000..870000000)
[   68.376688] SAA716x Budget 0000:04:00.0: DVB: adapter 1 frontend 0 frequency 0 out of range (48000000..870000000)
[   69.525845] SAA716x Budget 0000:04:00.0: DVB: adapter 0 frontend 0 frequency 0 out of range (48000000..870000000)`

The firmwares are also installed and loaded:

`[   65.644547] si2168 5-0066: downloading firmware from file 'dvb-demod-si2168-b40-01.fw'
[   66.054238] si2168 5-0066: firmware version: B 4.0.25
[   66.061693] si2157 10-0060: found a 'Silicon Labs Si2158-A20'
[   66.066695] si2157 10-0060: downloading firmware from file 'dvb-tuner-si2158-a20-01.fw'
[   66.333312] si2157 10-0060: firmware version: 2.1.9
[   66.333380] SAA716x Budget 0000:04:00.0: DVB: adapter 3 frontend 0 frequency 0 out of range (48000000..870000000)
[   66.346217] si2168 5-0066: downloading firmware from file 'dvb-demod-si2168-b40-01.fw'
[   66.350199] si2168 5-0064: downloading firmware from file 'dvb-demod-si2168-b40-01.fw'
[   67.168998] si2168 5-0064: firmware version: B 4.0.25
[   67.180148] si2157 9-0062: found a 'Silicon Labs Si2158-A20'
[   67.180261] si2157 9-0062: downloading firmware from file 'dvb-tuner-si2158-a20-01.fw'
[   67.194139] si2168 5-0066: firmware version: B 4.0.25
[   67.450075] si2157 9-0062: firmware version: 2.1.9
[   67.450111] SAA716x Budget 0000:04:00.0: DVB: adapter 2 frontend 0 frequency 0 out of range (48000000..870000000)
[   67.471101] si2168 5-0064: downloading firmware from file 'dvb-demod-si2168-b40-01.fw'
[   67.478004] si2168 6-0066: downloading firmware from file 'dvb-demod-si2168-b40-01.fw'
[   67.943410] si2168 5-0064: firmware version: B 4.0.25
[   68.099812] si2168 6-0066: firmware version: B 4.0.25
[   68.105529] si2157 8-0060: found a 'Silicon Labs Si2158-A20'
[   68.105619] si2157 8-0060: downloading firmware from file 'dvb-tuner-si2158-a20-01.fw'
[   68.376618] si2157 8-0060: firmware version: 2.1.9
[   68.376688] SAA716x Budget 0000:04:00.0: DVB: adapter 1 frontend 0 frequency 0 out of range (48000000..870000000)
[   68.412642] si2168 6-0066: downloading firmware from file 'dvb-demod-si2168-b40-01.fw'
[   68.413170] si2168 6-0064: downloading firmware from file 'dvb-demod-si2168-b40-01.fw'
[   69.244801] si2168 6-0066: firmware version: B 4.0.25
[   69.252074] si2168 6-0064: firmware version: B 4.0.25
[   69.256987] si2157 7-0062: found a 'Silicon Labs Si2158-A20'
[   69.257078] si2157 7-0062: downloading firmware from file 'dvb-tuner-si2158-a20-01.fw'
[   69.525798] si2157 7-0062: firmware version: 2.1.9
[   69.525845] SAA716x Budget 0000:04:00.0: DVB: adapter 0 frontend 0 frequency 0 out of range (48000000..870000000)
[   69.547854] si2168 6-0064: downloading firmware from file 'dvb-demod-si2168-b40-01.fw'
[   69.963368] si2168 6-0064: firmware version: B 4.0.25
[ 2044.639301] si2168 6-0066: downloading firmware from file 'dvb-demod-si2168-b40-01.fw'
[ 2045.047127] si2168 6-0066: firmware version: B 4.0.25
[ 2061.321879] si2168 6-0064: downloading firmware from file 'dvb-demod-si2168-b40-01.fw'
[ 2061.730451] si2168 6-0064: firmware version: B 4.0.25
[ 2236.569864] si2168 6-0064: downloading firmware from file 'dvb-demod-si2168-b40-01.fw'
[ 2236.978978] si2168 6-0064: firmware version: B 4.0.25
[ 2708.033831] si2168 6-0064: downloading firmware from file 'dvb-demod-si2168-b40-01.fw'
[ 2708.442194] si2168 6-0064: firmware version: B 4.0.25`

Unfortunately tvheadend can't scan mux or lock any signal.

Setup

Distro: Proxmox (Debian)
Kernel: 5.0.15
Backend: Tvheadend 4.3
Signal: DVB-C (Brazil)
Card: TBS 6285 / 6205
Dmesg: http://paste.debian.net/109192

Davin622, 2019-07-18:
HI
Because your  symbol rate less than 6000000,
you can see the si2168.c(drivers/media/dvb-frontends/).
the function of "si2168_set_frontend", when the symbol rate less than 6M, it will be set to DVBC_ANNEX_B mode,(line333~337)
		if (c->symbol_rate < 6000000) {
			delivery_system = 0x10;
			c->delivery_system = SYS_DVBC_ANNEX_B;
			c->bandwidth_hz = 6000000;
		}
so, you just disable them.
now it was fixed . you just update the drivers, then re-make and install.
thanks

bpbastos, 2019-07-18:
> HI
> Because your symbol rate less than 6000000,
> you can see the si2168.c(drivers/media/dvb-frontends/).
> the function of "si2168_set_frontend", when the symbol rate less than 6M, it will be set to DVBC_ANNEX_B mode,(line333~337)
> if (c->symbol_rate < 6000000) {
> delivery_system = 0x10;
> c->delivery_system = SYS_DVBC_ANNEX_B;
> c->bandwidth_hz = 6000000;
> }
> so, you just disable them.
> now it was fixed . you just need update the drivers, then re-make and install.
> thanks

Hi @Davin622 both TBS 6285 and 6205 are working working beautifully now.
Thank you!

closed 2019-07-19 by bpbastos