tbsdtv/linux_media #169: Unable to watch channels from some frequencies - DVB-C

bpbastos, 2019-07-20:
Guys, first of all I'd like to thank you for helping me with my other issue.

Now I'm having some problems with signal weakness trying to watch channels from frequencies above 705 Mhz,  thoses frequencies used to work with old driver (ljalves).

The problematic frequancies: 711Mhz, 717Mhz, 723Mhz e 729Mhz

My setup:
Backend: tvheadend
Tuner: TBS 6205/6285
Signal: DVB-C 
Symbol rate: 5217
Constellation: QAM/256
FEC: 3/4

crazycat69, 2019-07-21:
you have some "exotic" DVB-C provider :) before i have expirience with Uruguay DVB-C annex B 6MHz.
so you have DVB-C annex A/C 8MHz ?

bpbastos, 2019-07-21:
Welcome to Brazil! Everything in here have the need to not follow any standard. :)

I think it's DVB-C annex A 6Mhz, but I'm not sure. Is there a way to check?

crazycat69, 2019-07-21:
if this commit 17587ba554c68dff58d48f20669a7b37d7551547 fix your issue - so you havew DVB/C annexA/C 6MHz

bpbastos, 2019-07-21:
Frequencies from 123 to 705Mhz 
SNR: 33%
Signal Strength: 50%

Frequencies from 711 to 729Mhz
SNR: 27%
Signal Strength: 32%

crazycat69, 2019-07-22:
Need Strength, SNR in dB/dBm. Look like you "force old signal statistic" in tvh (dvb adapter settings)

bpbastos, 2019-07-23:
Frenquecies - OK

Freq: 231Mhz
SNR: 37 dB
Signal Strength: -47.0 dBm

Freq: 267Mhz
SNR: 37.5 dB
Signal Strength: -47.0 dBm

Freq: 381Mhz
SNR: 37 dB
Signal Strength: -50 dBm

Freq: 465Mhz
SNR: 34.5 dB
Signal Strength: -59.0 dBm

Freq: 453Mhz
SNR: 36 dB
Signal Strength: -56.0 dBm

Frequencies - NOT OK (Can't watch)

Freq: 705Mhz
SNR: 30.3 dB
Signal Strength: -67.0 dBm

Freq: 711Mhz
SNR: 30 dB
Signal Strength: -67.0 dBm

Freq: 723Mhz
SNR: 28.3 dB
Signal Strength: -70.0 dBm

crazycat69, 2019-07-23:
look like bad signal. try some ~20dBi amplifier

satdreamer, 2022-04-18:
@bpbastos Did the above issue occur on frequencies with HD or SD?