tbsdtv/linux_media #174: tbs6909x Tuning of channel fails depending on last channel tuned

tvhdeeptho, 2019-08-04:
Tuning to "The voice"  30.0E, 12271H Qpks symrate 30000 always fails if the 
previous channel tuned was  BNT3, 33.0E 12646H DVB-S2 PLS: Root+218997, stream 3

My setup: tbs6909x, all tuners except one disabled.  The tuner is connected to a 16-poisition switch which is connected to various satellite. Driver version: 17587ba554c68dff58d48f20669a7b37d7551547
Experiments done with tvheadend.

After I have tuned to BNT3 and then to The "Voice", there is no picture. 
Stopping "The voice" in tvheadend and trying to retune has no effect.
The problem can usually be fixed by tuning to some other channel, specifically "TF1", 33.0E, 12597V DVB-S2X 8PSKm symrate 10058.

The kernel reports that the tuner fails to lock (search algorithm failed)

So to summarize:
tune to the voice /39E: picture OK
tune to BNT3:/33E  picture OK
tune to the voice/39E: no picture
stop/restart "the voice": no picture
tune TF1/33E: picture OK
tune the voice/39E: picture OK
This is perfectly reproducible.

The same problem occurs if instead of the VOICE I use "folklor tv", 39.E, 12214V, QKSK 30000

Thinking that it might be QPSK related, I tried a few QPSK transponders on 282E to see if the have also the problem (first tune bnt3; then the channel on 28.2E; picture always ok) , but could not create the problem with those.

Thinking that it might be multistream related, I also tried the following:
tune to multistream on 5W; tune to the voice. In this case picture was always ok on the voide.

I do not think this is a diseqc problem, because I can reproduce the same problem with 
another switch connected to another tuner.

The LNB is fine. It is only a few weeks old and it works perfectly. I also verified that 
when the problem occurs, the LNB voltage = 18.2 Volt, which also shows that the hardware
is ok.

crazycat69, 2019-08-04:
check kernel log on tuning this DVB-S2/QPSK transponders, some like this

> 05.08.19 01:20	i2c i2c-0	delivery_system=6 modulation=9 frequency=1741000 symbol_rate=30000000 inversion=2 stream_id=296
> 05.08.19 01:20	i2c i2c-0	stid135_set_parameters: set pls_mode 0, pls_code 1 !
> 05.08.19 01:20	i2c i2c-0	stid135_set_parameters: locked !
> 05.08.19 01:20	i2c i2c-0	stid135_set_parameters: set ISI 40 !
>

tvhdeeptho, 2019-08-05:
Here are the logs (with some extra added debug output).
The logs show that simply tuning fails. 

The voice - OK
===================
Aug  4 23:31:43 streacom kernel: [100093.785627] i2c i2c-5: delivery_system=5 modulation=0 frequency=1671000 symbol_rate=30000000 inversion=2 stream_id=-1
Aug  4 23:31:43 streacom kernel: [100093.787807] i2c i2c-5: stid135_set_parameters: set pls_mode 0, pls_code 1 !
Aug  4 23:31:43 streacom kernel: [100093.914453] i2c i2c-5: stid135_set_parameters: RESULT locked=1 freq=1670360 srate=29998703 standard=0 pr=6 mod=0!
Aug  4 23:31:43 streacom kernel: [100093.914455] i2c i2c-5: stid135_set_parameters: locked !
Aug  4 23:31:43 streacom kernel: [100093.914456] i2c i2c-5: stid135_set_parameters: disable ISI filtering !

BNT3 OK
===========
Aug  4 23:32:05 streacom kernel: [100115.726979] i2c i2c-5: delivery_system=6 modulation=9 frequency=2046000 symbol_rate=13910000 inversion=2 stream_id=56063235
Aug  4 23:32:05 streacom kernel: [100115.728934] i2c i2c-5: stid135_set_parameters: set pls_mode 0, pls_code 218997 !
Aug  4 23:32:05 streacom kernel: [100115.920222] i2c i2c-5: stid135_set_parameters: RESULT locked=1 freq=2045441 srate=13916807 standard=1 pr=1 mod=0!
Aug  4 23:32:05 streacom kernel: [100115.920230] i2c i2c-5: stid135_set_parameters: locked !
Aug  4 23:32:05 streacom kernel: [100115.920234] i2c i2c-5: stid135_set_parameters: set ISI 3 !
Aug  4 23:32:06 streacom kernel: [100116.120644] tbsecp3_dma_tasklet: 57 callbacks suppressed

The voice - NOT OK
===================
Aug  4 23:32:12 streacom kernel: [100122.969945] i2c i2c-5: delivery_system=5 modulation=0 frequency=1671000 symbol_rate=30000000 inversion=2 stream_id=-1
Aug  4 23:32:12 streacom kernel: [100122.972046] i2c i2c-5: stid135_set_parameters: set pls_mode 0, pls_code 1 !
Aug  4 23:32:15 streacom kernel: [100125.483780] i2c i2c-5: stid135_set_parameters: RESULT locked=0 freq=0 srate=0 standard=642 pr=0 mod=0!
Aug  4 23:32:15 streacom kernel: [100125.486668] i2c i2c-5: stid135_set_parameters: not locked, band rf_power -37 dBm !
Aug  4 23:32:15 streacom kernel: [100125.486674] i2c i2c-5: stid135_set_parameters: disable ISI filtering !