tbsdtv/linux_media #175: please help with tbsci.c

slaweb, 2019-08-18:
Hello, is anyone have some custom tbsci.c because I open only 3ch from CI put in TBS 6991 SE and when I use NetUPS2 I open 6ch withouut any cusomizing driver?

Regards,

Davin622, 2019-08-20:
HI,
What is the app and CAM you used? they should be have the same function, Becasue they use the same CI protocol(en50221).
Regards

Davin622, 2019-08-20:
Could you give me the web link of the NetUps2. i want to see the info of it.

slaweb, 2019-08-20:
https://www.tbsdtv.com/forum/viewtopic.php?f=56&t=8979

I use astra cesbo and I have this card in the link, but I open only 3
ch...some gays open 9-10ch but they have some custom CI setup. Can you help
me?

Aug 20 23:03:07: ERROR: [Arena 4 #1] Bitrate:7108Kbit/s PES-Error Scrambled
Aug 20 23:03:07: ERROR: [Arena 4] Failed to switch to reserve
Aug 20 23:03:07: ERROR: [Arena 3 #1] Bitrate:5087Kbit/s PES-Error Scrambled
Aug 20 23:03:07: ERROR: [Arena 3] Failed to switch to reserve
Aug 20 23:03:07: ERROR: [NatGeo #1] Bitrate:1393Kbit/s PES-Error Scrambled
Aug 20 23:03:07: ERROR: [NatGeo] Failed to switch to reserve
Aug 20 23:03:07: ERROR: [RTL Kockica #1] Bitrate:1957Kbit/s PES-Error
Scrambled
Aug 20 23:03:07: ERROR: [RTL Kockica] Failed to switch to reserve
Aug 20 23:03:10: INFO: [dvb_ca 9:0] CA: Module MAXtv Sat CI+ CAM. 0x01
0x0000 0x0000
Aug 20 23:03:11: INFO: [dvb_ca 9:0] CA: Module CAID:0x1830 (session 0:4)
Aug 20 23:03:11: INFO: [dvb_ca 9:0] CA: Module CAID:0x1811 (session 0:4)
Aug 20 23:03:21: INFO: [RTL Kockica #1] Bitrate:1668Kbit/s
Aug 20 23:03:21: INFO: [RTL Kockica] Active input #1
Aug 20 23:03:25: INFO: [NatGeo #1] Bitrate:2056Kbit/s
Aug 20 23:03:25: INFO: [NatGeo] Active input #1
Aug 20 23:03:28: INFO: [Arena 3 #1] Bitrate:6327Kbit/s
Aug 20 23:03:28: INFO: [Arena 3] Active input #1

3 ch is max :(

slaweb, 2019-08-20:
I use TBS 6991 SE