tbsdtv/linux_media #176: Build process fails due to error during patching

Borim7, 2019-08-29:
When calling make the build process abort with an error

```
make -C /Daten/code/tvkarte/tbs-build/org-drivers/v4l 
make[1]: Verzeichnis „/Daten/code/tvkarte/tbs-build/org-drivers/v4l“ wird betreten
creating symbolic links...
make -C firmware prep
make[2]: Entering directory '/Daten/code/tvkarte/tbs-build/org-drivers/v4l/firmware'
make[2]: Nothing to be done for 'prep'.
make[2]: Leaving directory '/Daten/code/tvkarte/tbs-build/org-drivers/v4l/firmware'
make -C firmware
make[2]: Entering directory '/Daten/code/tvkarte/tbs-build/org-drivers/v4l/firmware'
make[2]: Nothing to be done for 'default'.
make[2]: Leaving directory '/Daten/code/tvkarte/tbs-build/org-drivers/v4l/firmware'
Kernel build directory is /lib/modules/4.15.0-55-generic/build
make -C ../linux apply_patches
make[2]: Entering directory '/Daten/code/tvkarte/tbs-build/org-drivers/linux'
make[3]: Entering directory '/Daten/code/tvkarte/tbs-build/org-drivers/linux'
Unapplying patches
make[3]: Leaving directory '/Daten/code/tvkarte/tbs-build/org-drivers/linux'
Applying patches for kernel 4.15.0-55-generic
patch -s -f -N -p1 -i ../backports/api_version.patch
The text leading up to this was:
--------------------------
|diff --git a/drivers/media/mc/mc-device.c b/drivers/media/mc/mc-device.c
|index 3bae24b15eaa..7f7cc64c637c 100644
|--- a/drivers/media/mc/mc-device.c
|+++ b/drivers/media/mc/mc-device.c
--------------------------
No file to patch.  Skipping patch.
1 out of 1 hunk ignored
Makefile:132: recipe for target 'apply_patches' failed
make[2]: *** [apply_patches] Error 1
make[2]: Leaving directory '/Daten/code/tvkarte/tbs-build/org-drivers/linux'
Makefile:51: recipe for target 'default' failed
make[1]: *** [default] Error 2
make[1]: Verzeichnis „/Daten/code/tvkarte/tbs-build/org-drivers/v4l“ wird verlassen
Makefile:26: recipe for target 'all' failed
make: *** [all] Error 2
```

The problem occurred on Ubuntu after updating the kernel to 4.15.0-58. The weird part is after switching back to the previous kernel (4.15.0-55) the problem persists.
Before the update the TBS driver could be compiled and work as expected.

So it looks like some other part cause the problem than the kernel itself. Any advise what part can cause the problem except the kernel?

crazycat69, 2019-08-31:
first need **make cleanall** for media_build, next **./build** and install
or more complex repo update as described in wiki [Upgrading sources and re-installing ](https://github.com/tbsdtv/linux_media/wiki#upgrading-sources-and-re-installing)

Borim7, 2019-09-02:
The build script does solve the problem, at least partly. A new linux media source was downloaded and with it the build process got further. But still I had to remove the drivers for zoran and ov9640 to complete the build process.

The reduced media tree works for me and my tv card is working again.

highfellow, 2020-03-06:
I had a similar error and it turned out to be due to not following the build instructions on the home page of this repository precisely. I have gone through it all again and it is now building okay.

Borim7, 2020-03-17:
You are correct, after resetup the complete build environment the build complete without any faults

closed 2020-03-17 by Borim7