tbsdtv/linux_media #177: TBS 5520SE on raspbian buster

hagaygo, 2019-09-29:
The process on the wiki page worked just fine (and faster) on pc with debian buster.

When doing same process for raspbian , except the stage of :

# uname -r
  4.9.0-8-amd64
  # apt-get install linux-headers-4.9.0-8-amd64

(did sudo apt-get install raspberrypi-kernel-headers)

After long compliation and no seen errors , raspbian boot with the following dmesg output :

> [    0.000000] Booting Linux on physical CPU 0xf00
[    0.000000] Linux version 4.19.66-v7+ (dom@buildbot) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611)) #1253 SMP Thu Aug 15 11:49:46 BST 2019
[    0.000000] CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=10c5387d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 2 Model B Rev 1.1
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] cma: Reserved 8 MiB at 0x3dc00000
[    0.000000] On node 0 totalpages: 255488
[    0.000000]   Normal zone: 2246 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 255488 pages, LIFO batch:63
[    0.000000] random: get_random_bytes called from start_kernel+0xac/0x4b4 with crng_init=0
[    0.000000] percpu: Embedded 17 pages/cpu s39488 r8192 d21952 u69632
[    0.000000] pcpu-alloc: s39488 r8192 d21952 u69632 alloc=17*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 253242
[    0.000000] Kernel command line: coherent_pool=1M bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708_fb.fbdepth=16 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3f000000 vc_mem.mem_size=0x3f600000  dwc_otg.lpm_enable=0 console=ttyAMA0,115200 console=tty1 root=PARTUUID=58a5f232-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 989824K/1021952K available (8192K kernel code, 629K rwdata, 2176K rodata, 1024K init, 821K bss, 23936K reserved, 8192K cma-reserved)
[    0.000000] Virtual kernel memory layout:
                   vector  : 0xffff0000 - 0xffff1000   (   4 kB)
                   fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
                   vmalloc : 0xbe800000 - 0xff800000   (1040 MB)
                   lowmem  : 0x80000000 - 0xbe600000   ( 998 MB)
                   modules : 0x7f000000 - 0x80000000   (  16 MB)
                     .text : 0x(ptrval) - 0x(ptrval)   (9184 kB)
                     .init : 0x(ptrval) - 0x(ptrval)   (1024 kB)
                     .data : 0x(ptrval) - 0x(ptrval)   ( 630 kB)
                      .bss : 0x(ptrval) - 0x(ptrval)   ( 822 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 26394 entries in 78 pages
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] arch_timer: cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000010] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000031] Switching to timer-based delay loop, resolution 52ns
[    0.000363] Console: colour dummy device 80x30
[    0.001085] console [tty1] enabled
[    0.001161] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[    0.001216] pid_max: default: 32768 minimum: 301
[    0.001665] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.001715] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.002990] CPU: Testing write buffer coherency: ok
[    0.003703] CPU0: update cpu_capacity 1024
[    0.003749] CPU0: thread -1, cpu 0, socket 15, mpidr 80000f00
[    0.004786] Setting up static identity map for 0x100000 - 0x10003c
[    0.005053] rcu: Hierarchical SRCU implementation.
[    0.006269] smp: Bringing up secondary CPUs ...
[    0.007601] CPU1: update cpu_capacity 1024
[    0.007615] CPU1: thread -1, cpu 1, socket 15, mpidr 80000f01
[    0.009055] CPU2: update cpu_capacity 1024
[    0.009069] CPU2: thread -1, cpu 2, socket 15, mpidr 80000f02
[    0.010505] CPU3: update cpu_capacity 1024
[    0.010516] CPU3: thread -1, cpu 3, socket 15, mpidr 80000f03
[    0.010688] smp: Brought up 1 node, 4 CPUs
[... 359 lines cut ...]
[    7.575951] dvb_usb_v2: Unknown symbol rc_unregister_device (err -22)
[    7.643130] sd 0:0:0:0: Attached scsi generic sg0 type 0
[    8.254460] FAT-fs (sda1): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.
[   10.033293] random: crng init done
[   10.033316] random: 7 urandom warning(s) missed due to ratelimiting
[   11.176089] 8021q: 802.1Q VLAN Support v1.8
[   11.337080] cfg80211: Loading compiled-in X.509 certificates for regulatory database
[   11.391953] Adding 102396k swap on /var/swap.  Priority:-2 extents:1 across:102396k SSFS
[   11.435841] cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
[   11.625187] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
[   13.173944] smsc95xx 1-1.1:1.0 eth0: link up, 100Mbps, full-duplex, lpa 0xC5E1
[   21.645904] FS-Cache: Netfs 'cifs' registered for caching
[   21.646881] Key type cifs.spnego registered
[   21.648178] Key type cifs.idmap registered

what catchs my eye as and error is :

> [    7.537140] dvb_usb_v2: disagrees about version of symbol rc_register_device
[    7.537163] dvb_usb_v2: Unknown symbol rc_register_device (err -22)

the RPI has 2 other dvb usb tuner (non TBS) that worked before this tbs driver install.

On debian on pc all tuners worked just fine.

Thanks in advnace for any help.

crazycat69, 2019-09-29:
remove kernel media modules before install, see install.sh script in [media_build](https://github.com/tbsdtv/media_build)

hagaygo, 2019-09-30:
Thank for your answer.

You mean i should install by executing install.sh?  (I installed by calling make install like the wiki page describes)

hagaygo, 2019-09-30:
just ran instal.sh instead of calling sudo make install , still erros , bit different ones :

> [    0.000000] Booting Linux on physical CPU 0xf00
[    0.000000] Linux version 4.19.66-v7+ (dom@buildbot) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611)) #1253 SMP                                                                                                          Thu Aug 15 11:49:46 BST 2019
[    0.000000] CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=10c5387d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 2 Model B Rev 1.1
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] cma: Reserved 8 MiB at 0x3dc00000
[    0.000000] On node 0 totalpages: 255488
[    0.000000]   Normal zone: 2246 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 255488 pages, LIFO batch:63
[    0.000000] random: get_random_bytes called from start_kernel+0xac/0x4b4 with crng_init=0
[    0.000000] percpu: Embedded 17 pages/cpu s39488 r8192 d21952 u69632
[    0.000000] pcpu-alloc: s39488 r8192 d21952 u69632 alloc=17*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 253242
[    0.000000] Kernel command line: coherent_pool=1M bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708_fb.fbdepth=16 bcm2708_fb                                                                                                         .fbswap=1 vc_mem.mem_base=0x3f000000 vc_mem.mem_size=0x3f600000  dwc_otg.lpm_enable=0 console=ttyAMA0,115200 console=tty1 root=PARTU                                                                                                         UID=58a5f232-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 989824K/1021952K available (8192K kernel code, 629K rwdata, 2176K rodata, 1024K init, 821K bss, 23936K reserv                                                                                                         ed, 8192K cma-reserved)
[    0.000000] Virtual kernel memory layout:
                   vector  : 0xffff0000 - 0xffff1000   (   4 kB)
                   fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
                   vmalloc : 0xbe800000 - 0xff800000   (1040 MB)
                   lowmem  : 0x80000000 - 0xbe600000   ( 998 MB)
                   modules : 0x7f000000 - 0x80000000   (  16 MB)
                     .text : 0x(ptrval) - 0x(ptrval)   (9184 kB)
                     .init : 0x(ptrval) - 0x(ptrval)   (1024 kB)
                     .data : 0x(ptrval) - 0x(ptrval)   ( 630 kB)
                      .bss : 0x(ptrval) - 0x(ptrval)   ( 822 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 26394 entries in 78 pages
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] arch_timer: cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000010] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000032] Switching to timer-based delay loop, resolution 52ns
[    0.000365] Console: colour dummy device 80x30
[    0.001087] console [tty1] enabled
[    0.001164] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[    0.001219] pid_max: default: 32768 minimum: 301
[    0.001672] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.001722] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.002989] CPU: Testing write buffer coherency: ok
[    0.003707] CPU0: update cpu_capacity 1024
[    0.003750] CPU0: thread -1, cpu 0, socket 15, mpidr 80000f00
[    0.004793] Setting up static identity map for 0x100000 - 0x10003c
[    0.005061] rcu: Hierarchical SRCU implementation.
[    0.006274] smp: Bringing up secondary CPUs ...
[    0.007604] CPU1: update cpu_capacity 1024
[    0.007617] CPU1: thread -1, cpu 1, socket 15, mpidr 80000f01
[    0.009064] CPU2: update cpu_capacity 1024
[    0.009075] CPU2: thread -1, cpu 2, socket 15, mpidr 80000f02
[    0.010506] CPU3: update cpu_capacity 1024
[    0.010518] CPU3: thread -1, cpu 3, socket 15, mpidr 80000f03
[    0.010696] smp: Brought up 1 node, 4 CPUs
[    0.010854] SMP: Total of 4 processors activated (153.60 BogoMIPS).
[    0.010882] CPU: All CPU(s) started in HYP mode.
[    0.010906] CPU: Virtualization extensions available.
[    0.012240] devtmpfs: initialized
[    0.027864] VFP support v0.3: implementor 41 architecture 2 part 30 variant 7 rev 5
[    0.028285] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.028349] futex hash table entries: 1024 (order: 4, 65536 bytes)
[    0.029234] pinctrl core: initialized pinctrl subsystem
[    0.030689] NET: Registered protocol family 16
[    0.034305] DMA: preallocated 1024 KiB pool for atomic coherent allocations
[... 320 lines cut ...]
[... cut ...]
[   11.818542] Adding 102396k swap on /var/swap.  Priority:-2 extents:1 across:102396k SSFS
[   11.975270] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
[   13.491899] smsc95xx 1-1.1:1.0 eth0: link up, 100Mbps, full-duplex, lpa 0xC5E1
[   21.756151] FS-Cache: Netfs 'cifs' registered for caching
[   21.757111] Key type cifs.spnego registered
[   21.757145] Key type cifs.idmap registered

Since debian on PC worked just fine , i guess wiki/guide for raspbian requires some more commands/settings.

hagaygo, 2019-10-03:
little update in case any one else is having a problem.

i followed the exact instruction from tbs forum post :

https://www.tbsdtv.com/forum/viewtopic.php?f=86&t=25151

it seems it work only with latest stretch release of raspibian  (i guess it means no RPI 4 support) :

https://downloads.raspberrypi.org/raspbian_lite/images/raspbian_lite-2019-04-09/
https://downloads.raspberrypi.org/raspbian/images/raspbian-2019-04-09/

The compile process seems to work and driver is loaded.

But as mentioned earlier i have other tuners attached and after kernel patch they stopped working :

> [ 1705.862275] usb 1-1.4.4: new high-speed USB device number 10 using dwc_otg
[ 1705.993580] usb 1-1.4.4: New USB device found, idVendor=0572, idProduct=c689
[ 1705.993600] usb 1-1.4.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1705.993611] usb 1-1.4.4: Product: EyeTV Stick
[ 1705.993621] usb 1-1.4.4: Manufacturer: Geniatech
[ 1705.993631] usb 1-1.4.4: SerialNumber: 160421
[ 1706.042815] dvb_usb_dvbsky: disagrees about version of symbol rc_keydown
[ 1706.042839] dvb_usb_dvbsky: Unknown symbol rc_keydown (err -22)
[ 1738.330513] usb 1-1.4.4: USB disconnect, device number 10
[ 1742.462579] usb 1-1.4.4: new high-speed USB device number 11 using dwc_otg
[ 1742.596740] usb 1-1.4.4: New USB device found, idVendor=048d, idProduct=9005
[ 1742.596760] usb 1-1.4.4: New USB device strings: Mfr=1, Product=0, SerialNumber=3
[ 1742.596771] usb 1-1.4.4: Manufacturer: ITE Technologies, Inc.
[ 1742.596782] usb 1-1.4.4: SerialNumber: AF0102020700001
[ 1742.659748] dvb_usb_af9035: disagrees about version of symbol rc_keydown
[ 1742.659769] dvb_usb_af9035: Unknown symbol rc_keydown (err -22)

Also , after trying raspbian package update,   the kernel got updated and the tbs driver stopped working and the other 2 tuners worked again.

So this kernel patching will have us need to keep kernel package from being updated.

any pointers , solutions ? 

Thanks in advance.

crazycat69, 2019-10-04:
Remove kernel media modules before media_build install. But look like some Broadcom SoC driver depend from v4l2 core and must be recompiled.

hagaygo, 2019-10-04:
If you mean run :
sudo rm -rf /lib/modules/`uname -r`/kernel/drivers/media/
Before compiling , i did that.
Tbs tuner is working, but other tuners stopped working.
Any way to keep them working after installing tbs drivers? 
I

victorvhpg, 2019-10-28:
my TBS5520se didnt works on my raspberry pi 4 - Raspbian Buster

victorvhpg, 2019-10-31:
@crazycat69 please

przemo1983, 2020-01-10:
In my case TBS5520se does not work with Raspbian Buster either. Any help?

elettrone1, 2020-05-01:
Any update on this issue please?

hagaygo, 2020-05-02:
Check

https://www.tbsdtv.com/forum/viewtopic.php?f=86&t=25151&start=10

melo77, 2020-05-12:
I confirm that the tbs5520se works    2019-04-08-raspbian-stretch.zip and media_build-rpi-2020-01-02.tar.bz2 but the other similar devices after build do not work not recognized example aver media hds2 and other work only tbs device

melo77, 2020-05-12:
dmesg | grep dvb
[    4.356887] dvb_usb_af9035: disagrees about version of symbol rc_keydown
[    4.356907] dvb_usb_af9035: Unknown symbol rc_keydown (err -22)
[  113.109379] dvb_usb_af9035: disagrees about version of symbol rc_keydown
[  113.109393] dvb_usb_af9035: Unknown symbol rc_keydown (err -22)
[  162.126584] dvb_usb_dvbsky: disagrees about version of symbol rc_keydown
[  162.126607] dvb_usb_dvbsky: Unknown symbol rc_keydown (err -22)
[  174.866869] dvb_usb_dvbsky: disagrees about version of symbol rc_keydown
[  174.866885] dvb_usb_dvbsky: Unknown symbol rc_keydown (err -22)
[  465.427322] dvb-usb: found a 'TurboSight TBS 5520SE' in cold state, will try to load a firmware
[  465.428867] dvb-usb: downloading firmware from file 'dvb-usb-id5520se.fw'
[  465.559729] dvb-usb: found a 'TurboSight TBS 5520SE' in warm state.
[  465.560340] dvb-usb: will pass the complete MPEG2 transport stream to the software demuxer.
[  465.561104] dvbdev: DVB: registering new adapter (TurboSight TBS 5520SE)

melo77, 2020-05-12:
dmesg | grep frontend
[  465.605307] usb 1-1.3: DVB: registering adapter 0 frontend 0 (TurboSight TBS 5520SE DVB-T/T2/C/C2/ISDB-T)...
[  465.605581] usb 1-1.3: DVB: registering adapter 0 frontend 1 (TurboSight TBS 5520SE DVB-S/S2/S2X)...
[  467.738631] usb 1-1.3: DVB: registering adapter 0 frontend 0 (TurboSight TBS 5520SE DVB-T/T2/C/C2/ISDB-T)...
[  467.738909] usb 1-1.3: DVB: registering adapter 0 frontend 1 (TurboSight TBS 5520SE DVB-S/S2/S2X)...
[  468.170353] usb 1-1.3: DVB: adapter 0 frontend 0 frequency 0 out of range (174000000..862000000)
[  468.409717] usb 1-1.3: DVB: adapter 0 frontend 1 frequency 0 out of range (950000..2150000)
[43772.668764] usb 1-1.1.2: DVB: registering adapter 0 frontend 0 (TurboSight TBS 5520SE DVB-T/T2/C/C2/ISDB-T)...
[43772.669911] usb 1-1.1.2: DVB: registering adapter 0 frontend 1 (TurboSight TBS 5520SE DVB-S/S2/S2X)...
[43774.680072] usb 1-1.1.2: DVB: registering adapter 0 frontend 0 (TurboSight TBS 5520SE DVB-T/T2/C/C2/ISDB-T)...
[43774.681111] usb 1-1.1.2: DVB: registering adapter 0 frontend 1 (TurboSight TBS 5520SE DVB-S/S2/S2X)...
[43775.106595] usb 1-1.1.2: DVB: adapter 0 frontend 0 frequency 0 out of range (174000000..862000000)
[43775.348142] usb 1-1.1.2: DVB: adapter 0 frontend 1 frequency 0 out of range (950000..2150000)

melo77, 2020-05-12:
dmesg | grep "registering adapter"
[  465.605307] usb 1-1.3: DVB: registering adapter 0 frontend 0 (TurboSight TBS 5520SE DVB-T/T2/C/C2/ISDB-T)...
[  465.605581] usb 1-1.3: DVB: registering adapter 0 frontend 1 (TurboSight TBS 5520SE DVB-S/S2/S2X)...
[  467.738631] usb 1-1.3: DVB: registering adapter 0 frontend 0 (TurboSight TBS 5520SE DVB-T/T2/C/C2/ISDB-T)...
[  467.738909] usb 1-1.3: DVB: registering adapter 0 frontend 1 (TurboSight TBS 5520SE DVB-S/S2/S2X)...
[43772.668764] usb 1-1.1.2: DVB: registering adapter 0 frontend 0 (TurboSight TBS 5520SE DVB-T/T2/C/C2/ISDB-T)...
[43772.669911] usb 1-1.1.2: DVB: registering adapter 0 frontend 1 (TurboSight TBS 5520SE DVB-S/S2/S2X)...
[43774.680072] usb 1-1.1.2: DVB: registering adapter 0 frontend 0 (TurboSight TBS 5520SE DVB-T/T2/C/C2/ISDB-T)...
[43774.681111] usb 1-1.1.2: DVB: registering adapter 0 frontend 1 (TurboSight TBS 5520SE DVB-S/S2/S2X)...

melo77, 2020-05-12:
root@raspberrypi:~# 
the problemi is Unknown symbol rc_keydown (err -22)

dmesg | grep dvb
[    4.356887] dvb_usb_af9035: disagrees about version of symbol rc_keydown
[    4.356907] dvb_usb_af9035: Unknown symbol rc_keydown (err -22)
[  113.109379] dvb_usb_af9035: disagrees about version of symbol rc_keydown
[  113.109393] dvb_usb_af9035: Unknown symbol rc_keydown (err -22)
[  162.126584] dvb_usb_dvbsky: disagrees about version of symbol rc_keydown
[  162.126607] dvb_usb_dvbsky: Unknown symbol rc_keydown (err -22)
[  174.866869] dvb_usb_dvbsky: disagrees about version of symbol rc_keydown
[  174.866885] dvb_usb_dvbsky: Unknown symbol rc_keydown (err -22)

crazycat69, 2020-05-12:
remove old media modules before install. See install.sh script

melo77, 2020-05-12:
/lib/modules  there are:
4.14.98+
4.14.98-v7+

root@raspberrypi:~# uname -r
4.14.98-v7+

sudo rm -rf /lib/modules/4.14.98-v7+/kernel/drivers/media
sudo rm -rf /lib/modules/4.14.98-v7+/kernel/drivers/staging/media
sudo rm -rf /lib/modules/4.14.98-v7+/kernel/drivers/linux
sudo rm -rf /lib/modules/4.14.98-v7+/extra

this is my action build:

sudo rm -rf /lib/modules/`uname -r`/kernel/drivers/media/

sudo rm -rf /lib/modules/$(uname -r)/kernel/drivers/media
sudo rm -rf /lib/modules/$(uname -r)/kernel/drivers/staging/media
sudo rm -rf /lib/modules/$(uname -r)/kernel/drivers/linux
sudo rm -rf /lib/modules/$(uname -r)/extra

cd /usr/src
wget http://www.tbsdtv.com/download/document/linux/media_build-rpi-2020-01-02.tar.bz2
tar jxvf media_build-rpi-2020-01-02.tar.bz2
cd /usr/src/media_build
./install.sh

after reboot i have this log

root@raspberrypi:~# dmesg | grep dvb
[    4.591657] dvb_usb_af9035: disagrees about version of symbol rc_keydown
[    4.591673] dvb_usb_af9035: Unknown symbol rc_keydown (err -22)
[    4.662506] dvb-usb: found a 'TurboSight TBS 5520SE' in cold state, will try to load a firmware
[    4.688023] dvb-usb: downloading firmware from file 'dvb-usb-id5520se.fw'
[    4.811718] dvb-usb: found a 'TurboSight TBS 5520SE' in warm state.
[    4.812299] dvb-usb: will pass the complete MPEG2 transport stream to the software demuxer.
[    4.820795] dvbdev: DVB: registering new adapter (TurboSight TBS 5520SE)
[    4.821133] dvb-usb: MAC address reading failed.
[    4.921385] dvb-usb: TurboSight TBS 5520SE successfully initialized and connected.
[    4.923937] dvb-usb: TurboSight TBS 5520SE successfully deinitialized and disconnected.
[    6.853114] dvb-usb: found a 'TurboSight TBS 5520SE' in cold state, will try to load a firmware
[    6.853132] dvb-usb: downloading firmware from file 'dvb-usb-id5520se.fw'
[    6.981640] dvb-usb: found a 'TurboSight TBS 5520SE' in warm state.
[    6.982232] dvb-usb: will pass the complete MPEG2 transport stream to the software demuxer.

melo77, 2020-05-12:
[image]
[image]

melo77, 2020-05-14:
[image]

melo77, 2020-05-15:
another problem occurs on reboot
does not remain active by tvheanded
[image]
[image]

/etc/modprobe.d
tbs.conf

##  TurboSight TBS 5520SE DVB-T/T2/C/C2/ISDB-T DVB-S/S2/S2X
options dvb-usb disable_rc_polling=1
options dvb_usb_tbs5520se adapter_nr=0

melo77, 2020-05-23:
@crazycat69  remove old media modules before install. See install.sh script
it is not only a problem of deleting previous drivers but to improve the package patch to install

melo77, 2021-01-27:
to solve the problem Unknown symbol rc_keydown (err -22) before build i use make menuselect 

this is my correct /media_build/4lv/.config

#
# Automatically generated file; DO NOT EDIT.
# V4L/DVB menu
#
CONFIG_MMU=y
CONFIG_SYSFS=y
CONFIG_OF=y
CONFIG_SND_SOC=m
CONFIG_MULTIPLEXER=m
CONFIG_HDMI=y
CONFIG_GENERIC_PHY=y
CONFIG_REGMAP_MMIO=y
CONFIG_FB_CFB_COPYAREA=y
CONFIG_PCI_MSI=y
CONFIG_MFD_WL1273_CORE=m
CONFIG_RATIONAL=y
CONFIG_HAVE_CLK=y
CONFIG_BITREVERSE=y
CONFIG_TTY=y
CONFIG_I2C=y
CONFIG_USB_ACM=m
CONFIG_SERIO_SERPORT=m
CONFIG_FW_LOADER=y
CONFIG_REGULATOR=y
CONFIG_FONT_SUPPORT=y
CONFIG_CRC32=y
CONFIG_SND_PCM=m
CONFIG_PWM=y
CONFIG_LEDS_CLASS=y
CONFIG_DMA_SHARED_BUFFER=y
CONFIG_HAS_IOMEM=y
CONFIG_REGMAP_I2C=y
CONFIG_REGMAP=y
CONFIG_FB=y
CONFIG_COMMON_CLK=y
CONFIG_FONT_8x16=y
CONFIG_PINCTRL=y
CONFIG_HAS_DMA=y
CONFIG_INPUT=y
CONFIG_USB_ARCH_HAS_HCD=y
CONFIG_NET=y
CONFIG_EXPERT=y
CONFIG_MFD_CORE=y
CONFIG_NEW_LEDS=y
CONFIG_RESET_CONTROLLER=y
CONFIG_PCI=y
CONFIG_SPI=y
CONFIG_GPIOLIB=y
CONFIG_TI_ST=m
CONFIG_DEBUG_FS=y
CONFIG_MTD=m
CONFIG_I2C_MUX=m
CONFIG_MFD_SYSCON=y
CONFIG_INET=y
CONFIG_MODULES=y
CONFIG_SERIO=m
CONFIG_ARCH_MULTIPLATFORM=y
CONFIG_SPI_MASTER=y
CONFIG_RFKILL=m
CONFIG_USB=y
CONFIG_I2C_ALGOBIT=m
CONFIG_SND=m
CONFIG_I2C_GPIO=m
CONFIG_PM=y
CONFIG_FB_CFB_FILLRECT=y
CONFIG_GENERIC_ALLOCATOR=y
[... 545 lines cut ...]
CONFIG_DVB_IX2505V=m
CONFIG_DVB_HORUS3A=m
CONFIG_DVB_ASCOT2E=m
CONFIG_DVB_HELENE=m

#
# Common Interface (EN50221) controller drivers
#
CONFIG_DVB_CXD2099=m
CONFIG_DVB_SP2=m

#
# Tools to develop new frontends
#
CONFIG_DVB_DUMMY_FE=m
# CONFIG_DVB_TAS2971 is not set
CONFIG_AUDIO_SUPPORT=y

#
# ALSA sound
#
# CONFIG_SND_BT87X is not set
CONFIG_FRAME_VECTOR=m
# CONFIG_MISC_DEVICES is not set
# CONFIG_STAGING is not set