tbsdtv/linux_media #186: TBS 5520SE does not work with Raspberry 4B

przemo1983, 2020-01-11:
Hi.

I downloaded and built drivers/firmware according to https://github.com/tbsdtv/linux_media/wiki/Installating both on **Raspbian Buster** and **Ubuntu 19.10.1** on **Raspberry Pi 4B**.  
I both cases compilation was OK, everything processed correctly.   
Of course before building I made  
`sudo rm -rf /lib/modules/``uname -r``/kernel/drivers/media/*`

After installing drivers and reboot there are errors in dmesg on both OSes (even I removed files as stated above - this is not a workaround for Raspberry PI 4B):
```
[   10.632179] dvb_usb: disagrees about version of symbol rc_register_device    
[   10.632188] dvb_usb: Unknown symbol rc_register_device (err -22)    
[   10.632212] dvb_usb: disagrees about version of symbol rc_free_device    
[   10.632218] dvb_usb: Unknown symbol rc_free_device (err -22)  
[   10.632238] dvb_usb: disagrees about version of symbol rc_allocate_device  
[   10.632244] dvb_usb: Unknown symbol rc_allocate_device (err -22)
[   10.632278] dvb_usb: disagrees about version of symbol rc_unregister_device
[   10.632284] dvb_usb: Unknown symbol rc_unregister_device (err -22)
```

It seems that TBS device does not work at all with new Raspberry 4B using different OSes.

Please help with that. Moving back to Raspberry 3 is not an option at all.

philippjess, 2020-02-09:
Its the same on RPi3 with Ubuntu 19.10 5.3.0-1017-raspi2 arm64

orrpan, 2020-02-13:
Hade the same problem, fixed it with `make menuconfig` and deselected a lot of packages.
I have tbs5220 working fine. Not sure which package is destroying the build, but here is my `media_build/v4l/.config` compiling file. 
[.config file](https://gist.githubusercontent.com/orrpan/65ec68592a79feea79fa2f2c5d2459c6/raw/7c9ee29e7520f18d20bc25a23c4bd6c390b83629/tbs5220)

Solfire, 2020-03-03:
> Hade the same problem, fixed it with `make menuconfig` and deselected a lot of packages.
> I have tbs5220 working fine. Not sure which package is destroying the build, but here is my [...]

@orrpan

Which linux version you are running?
Raspbian Buster (Lite)
Ubuntu 18.04.4 LTS
Ubuntu 19.10
on **Raspberry Pi 4B**.

orrpan, 2020-03-04:
> Which linux version you are running?
> Raspbian Buster (Lite)
> Ubuntu 18.04.4 LTS
> Ubuntu 19.10
> on **Raspberry Pi 4B**.

Buster 4.19.97-v7+, RPi 3B+

Solfire, 2020-03-04:
@orrpan thanks

I'll give Buster 4.19.97-v7+ on **RPi 4B** (Rev 1.2) a try

kalashnikov2, 2020-03-29:
> 
> 
> @orrpan thanks
> 
> I'll give Buster 4.19.97-v7+ on **RPi 4B** (Rev 1.2) a try

do not work.

full dmesg  [https://pastebin.com/raw/DxwEvR19](url)

```
.....
[    3.491105] brcmfmac: F1 signature read @0x18000000=0x15264345
[    3.504478] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    3.504841] usbcore: registered new interface driver brcmfmac
[    3.738568] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    3.751850] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Mar  2 2020 23:30:41 version 7.45.202 (r724630 CY) FWID 01-72f6ece2

.....
[   83.556796] usb 1-1.3: new high-speed USB device number 3 using xhci_hcd
[   83.687098] usb 1-1.3: New USB device found, idVendor=734c, idProduct=5521, bcdDevice= 0.01
[   83.687114] usb 1-1.3: New USB device strings: Mfr=0, Product=0, SerialNumber=0
```