tbsdtv/linux_media #188: Bug: tbs6508 - closing frontend on odd-numbered adapter sets voltage to zero on other tuner

tvhdeeptho, 2020-01-26:
The tbs6508 card has 4 rf inputs but 8 tuners and demods.
An even/odd numbered tuner pair always must tune to a transponder of the same band and  polarity. This works only partially: when the odd numbered tuner is closed, the voltage on the lnb is removed and the even numbered tuner stops functioning until it is reopened.

The bug is in tbsecp3_set_voltage.

tvhdeeptho, 2020-01-26:
The following patch provides a workaround for this specific card.
[patch1.txt](https://github.com/tbsdtv/linux_media/files/4114622/patch1.txt)

crazycat69, 2020-02-21:
This not bug :) This normal behavior, only need proper tvh setup. Each odd as master, even slave to odd.

tvhdeeptho, 2020-05-25:
I think it is a bug because of the following problem:
1) tune channel 1 on master
2) tune channel 2 on slave
3) stop channel 1, but keep channel 2 running.

Expected behaviour is that channel 2 keeps working, but it does not, because when 
you stop viewing channel 1, lnb power is removed. This is not what user wants....

crazycat69, 2020-05-25:
disable power save for adapter in tvh