tbsdtv/linux_media #190: No message returned when running dmesg | grep frontend

andyprata, 2020-02-07:
I've followed the instructions here to install the driver for the TBS 5927 DVB-S/S2: https://www.youtube.com/watch?v=x6hx4o5TPHA

Everything seems to go well except when I run `dmesg | grep frontend` in the terminal no messages return. In otherwords, grep doesn't find anything. I also tried `dmesg | grep TBS` and nothing returns.

Is there anything else I can try to see if the driver is installed correctly? I'm on Ubuntu:
Distributor ID:	Ubuntu
Description:	Ubuntu 18.04.4 LTS
Release:	18.04
Codename:	bionic

crazycat69, 2020-02-10:
1. remove kernel media modules before install media_build (see install.sh script)
2. disable Secure boot in BIOS setup

claus007, 2021-11-17:
> disable Secure boot in BIOS setup