tbsdtv/linux_media #191: Unable to compile on kernel 5.5

TRPB, 2020-02-07:
When trying to compile on 5.5 I get the following errors: 

```
Syncing with dir ../../media
Patches for 5.5.2-arch1-1 already applied.
make[2]: Leaving directory '/home/media/.cache/yay/tbs-linux_media-git/src/media_build/linux'
make -C /lib/modules/5.5.2-arch1-1/build M=/home/media/.cache/yay/tbs-linux_media-git/src/media_build/v4l  modules
make[2]: Entering directory '/usr/lib/modules/5.5.2-arch1-1/build'
  CC [M]  /home/media/.cache/yay/tbs-linux_media-git/src/media_build/v4l/msp3400-driver.o
  CC [M]  /home/media/.cache/yay/tbs-linux_media-git/src/media_build/v4l/msp3400-kthreads.o
  CC [M]  /home/media/.cache/yay/tbs-linux_media-git/src/media_build/v4l/smiapp-core.o
  CC [M]  /home/media/.cache/yay/tbs-linux_media-git/src/media_build/v4l/smiapp-regs.o
  CC [M]  /home/media/.cache/yay/tbs-linux_media-git/src/media_build/v4l/smiapp-quirk.o
  CC [M]  /home/media/.cache/yay/tbs-linux_media-git/src/media_build/v4l/smiapp-limits.o
  CC [M]  /home/media/.cache/yay/tbs-linux_media-git/src/media_build/v4l/et8ek8_mode.o
  CC [M]  /home/media/.cache/yay/tbs-linux_media-git/src/media_build/v4l/et8ek8_driver.o
In file included from <command-line>:
/home/media/.cache/yay/tbs-linux_media-git/src/media_build/v4l/compat.h: In function 'i2c_new_secondary_device':
In file included from <command-line>:
/home/media/.cache/yay/tbs-linux_media-git/src/media_build/v4l/compat.h: In function 'i2c_new_secondary_device':
In file included from <command-line>:
/home/media/.cache/yay/tbs-linux_media-git/src/media_build/v4l/compat.h: In function 'i2c_new_secondary_device':
In file included from <command-line>:
/home/media/.cache/yay/tbs-linux_media-git/src/media_build/v4l/compat.h: In function 'i2c_new_secondary_device':
In file included from <command-line>:
/home/media/.cache/yay/tbs-linux_media-git/src/media_build/v4l/compat.h: In function 'i2c_new_secondary_device':
In file included from <command-line>:
/home/media/.cache/yay/tbs-linux_media-git/src/media_build/v4l/compat.h: In function 'i2c_new_secondary_device':
In file included from <command-line>:
/home/media/.cache/yay/tbs-linux_media-git/src/media_build/v4l/compat.h: In function 'i2c_new_secondary_device':
/home/media/.cache/yay/tbs-linux_media-git/src/media_build/v4l/compat.h:2437:9: error: implicit declaration of function 'i2c_new_dummy' [-Werror=implicit-function-declaration]
 2437 |  return i2c_new_dummy(client->adapter, addr);
      |         ^~~~~~~~~~~~~
In file included from <command-line>:
/home/media/.cache/yay/tbs-linux_media-git/src/media_build/v4l/compat.h: In function 'i2c_new_secondary_device':
/home/media/.cache/yay/tbs-linux_media-git/src/media_build/v4l/compat.h:2437:9: error: implicit declaration of function 'i2c_new_dummy' [-Werror=implicit-function-declaration]
 2437 |  return i2c_new_dummy(client->adapter, addr);
      |         ^~~~~~~~~~~~~
/home/media/.cache/yay/tbs-linux_media-git/src/media_build/v4l/compat.h:2437:9: error: implicit declaration of function 'i2c_new_dummy' [-Werror=implicit-function-declaration]
 2437 |  return i2c_new_dummy(client->adapter, addr);
      |         ^~~~~~~~~~~~~
/home/media/.cache/yay/tbs-linux_media-git/src/media_build/v4l/compat.h:2437:9: error: implicit declaration of function 'i2c_new_dummy' [-Werror=implicit-function-declaration]
 2437 |  return i2c_new_dummy(client->adapter, addr);
      |         ^~~~~~~~~~~~~
/home/media/.cache/yay/tbs-linux_media-git/src/media_build/v4l/compat.h:2437:9: warning: returning 'int' from a function with return type 'struct i2c_client *' makes pointer from integer without a cast [-Wint-conversion]
 2437 |  return i2c_new_dummy(client->adapter, addr);
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/media/.cache/yay/tbs-linux_media-git/src/media_build/v4l/compat.h:2437:9: error: implicit declaration of function 'i2c_new_dummy' [-Werror=implicit-function-declaration]
 2437 |  return i2c_new_dummy(client->adapter, addr);
      |         ^~~~~~~~~~~~~
/home/media/.cache/yay/tbs-linux_media-git/src/media_build/v4l/compat.h:2437:9: error: implicit declaration of function 'i2c_new_dummy' [-Werror=implicit-function-declaration]
 2437 |  return i2c_new_dummy(client->adapter, addr);
      |         ^~~~~~~~~~~~~
/home/media/.cache/yay/tbs-linux_media-git/src/media_build/v4l/compat.h:2437:9: warning: returning 'int' from a function with return type 'struct i2c_client *' makes pointer from integer without a cast [-Wint-conversion]
 2437 |  return i2c_new_dummy(client->adapter, addr);
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/media/.cache/yay/tbs-linux_media-git/src/media_build/v4l/compat.h:2437:9: warning: returning 'int' from a function with return type 'struct i2c_client *' makes pointer from integer without a cast [-Wint-conversion]
 2437 |  return i2c_new_dummy(client->adapter, addr);
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/media/.cache/yay/tbs-linux_media-git/src/media_build/v4l/compat.h:2437:9: warning: returning 'int' from a function with return type 'struct i2c_client *' makes pointer from integer without a cast [-Wint-conversion]
 2437 |  return i2c_new_dummy(client->adapter, addr);
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/media/.cache/yay/tbs-linux_media-git/src/media_build/v4l/compat.h:2437:9: warning: returning 'int' from a function with return type 'struct i2c_client *' makes pointer from integer without a cast [-Wint-conversion]
 2437 |  return i2c_new_dummy(client->adapter, addr);
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/media/.cache/yay/tbs-linux_media-git/src/media_build/v4l/compat.h:2437:9: warning: returning 'int' from a function with return type 'struct i2c_client *' makes pointer from integer without a cast [-Wint-conversion]
 2437 |  return i2c_new_dummy(client->adapter, addr);
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/media/.cache/yay/tbs-linux_media-git/src/media_build/v4l/compat.h:2437:9: error: implicit declaration of function 'i2c_new_dummy' [-Werror=implicit-function-declaration]
 2437 |  return i2c_new_dummy(client->adapter, addr);
      |         ^~~~~~~~~~~~~
/home/media/.cache/yay/tbs-linux_media-git/src/media_build/v4l/compat.h:2437:9: error: implicit declaration of function 'i2c_new_dummy' [-Werror=implicit-function-declaration]
 2437 |  return i2c_new_dummy(client->adapter, addr);
      |         ^~~~~~~~~~~~~
/home/media/.cache/yay/tbs-linux_media-git/src/media_build/v4l/compat.h:2437:9: warning: returning 'int' from a function with return type 'struct i2c_client *' makes pointer from integer without a cast [-Wint-conversion]
 2437 |  return i2c_new_dummy(client->adapter, addr);
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/media/.cache/yay/tbs-linux_media-git/src/media_build/v4l/compat.h:2437:9: warning: returning 'int' from a function with return type 'struct i2c_client *' makes pointer from integer without a cast [-Wint-conversion]
 2437 |  return i2c_new_dummy(client->adapter, addr);
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
make[3]: *** [scripts/Makefile.build:266: /home/media/.cache/yay/tbs-linux_media-git/src/media_build/v4l/et8ek8_mode.o] Error 1
make[3]: *** Waiting for unfinished jobs....
cc1: some warnings being treated as errors
make[3]: *** [scripts/Makefile.build:266: /home/media/.cache/yay/tbs-linux_media-git/src/media_build/v4l/smiapp-limits.o] Error 1
cc1: some warnings being treated as errors
make[3]: *** [scripts/Makefile.build:266: /home/media/.cache/yay/tbs-linux_media-git/src/media_build/v4l/smiapp-quirk.o] Error 1
cc1: some warnings being treated as errors
make[3]: *** [scripts/Makefile.build:266: /home/media/.cache/yay/tbs-linux_media-git/src/media_build/v4l/smiapp-regs.o] Error 1
cc1: some warnings being treated as errors
make[3]: *** [scripts/Makefile.build:266: /home/media/.cache/yay/tbs-linux_media-git/src/media_build/v4l/msp3400-driver.o] Error 1
cc1: some warnings being treated as errors
make[3]: *** [scripts/Makefile.build:266: /home/media/.cache/yay/tbs-linux_media-git/src/media_build/v4l/et8ek8_driver.o] Error 1
cc1: some warnings being treated as errors
make[3]: *** [scripts/Makefile.build:266: /home/media/.cache/yay/tbs-linux_media-git/src/media_build/v4l/msp3400-kthreads.o] Error 1
cc1: some warnings being treated as errors
make[3]: *** [scripts/Makefile.build:266: /home/media/.cache/yay/tbs-linux_media-git/src/media_build/v4l/smiapp-core.o] Error 1
make[2]: *** [Makefile:1693: /home/media/.cache/yay/tbs-linux_media-git/src/media_build/v4l] Error 2
make[2]: Leaving directory '/usr/lib/modules/5.5.2-arch1-1/build'
make[1]: *** [Makefile:53: default] Error 2
make[1]: Leaving directory '/home/media/.cache/yay/tbs-linux_media-git/src/media_build/v4l'
make: *** [Makefile:26: all] Error 2
==> ERROR: A failure occurred in build().
    Aborting...
Error making: tbs-linux_media-git
```

crazycat69, 2020-02-21:
upd

shaifeyna56, 2020-05-03:
May I know latest kernel support for this driver? now im using linuxmint 19.03 , tried with stock kernel, not functioning, but when compile latest kernel on kernel.org 5.6.10.. I can scan muxed and got services list, but when try to play that TV channel, it show no input source on tvheadend and no signal at kodi and also usb tuner turn blue light which give sign that the tuner locked..

crazycat69, 2020-05-04:
See tvh log and Status/stream. Maybe TVH EPG grabber work