tbsdtv/linux_media #192: TBS 6290SE: Cannot tune

Triggaaaaa, 2020-03-10:
The issue I am facing is very similar to #79 , except for the different model.

My hardware worked once with older drivers and kernel 4.4.x. But after several upgrades I could not get it working anymore. In my last attempt I compiled the latest version of the drivers successfully against kernel 5.5.8 on Arch Linux. The driver seems to load fine. 

```
[    3.142970] TBSECP3 driver 0000:01:00.0: TurboSight TBS 6290SE DVB-T/T2/C + 2xCI 
[    3.143211] dvbdev: DVB: registering new adapter (TBSECP3 DVB Adapter)
[    3.357785] si2168 1-0064: Silicon Labs Si2168-B40 successfully identified
[    3.357787] si2168 1-0064: firmware version: B 4.0.2
[    3.394037] TBSECP3 driver 0000:01:00.0: DVB: registering adapter 0 frontend 0 (TurboSight TBS 6290SE DVB-T/T2/C + 2xCI )...
[    3.902209] si2168 2-0064: downloading firmware from file 'dvb-demod-si2168-b40-01.fw'
[    5.881612] TBSECP3 driver 0000:01:00.0: DVB: adapter 1 frontend 0 frequency 0 out of range (48000000..870000000)
```

But `w_scan` then tells me e.g.
```
514000: (time: 01:52.172)         signal ok:    QAM_AUTO f = 514000 kHz I999B8C999D999T999G999Y999P0 (0:0:0)
        Info: no data from PAT after 6 seconds
        deleting (QAM_AUTO f = 514000 kHz I999B8C999D999T999G999Y999P0 (0:0:0)
```

I am out of idea. Is there something I can check? Many thanks in advance!

crazycat69, 2020-05-20:
maybe buggy w_scan ? 6280 and 6290SE is 100% different hw.

Triggaaaaa, 2020-05-20:
Thank you for your reply and explanation! 

I do not think it is w_scan. I tried other tools as well and, in the end, Tvheadend does not find any channels.

Is there something I can provide or check in order to approach this issue?

crazycat69, 2020-05-21:
you can give me remote access to your PC (ssh or some else)