tbsdtv/linux_media #193: media_build compilation breaks snd_usb_audio module loading

Tnix76, 2020-03-20:
Hi all,
I usually work very well with TBS5520 DVB device. I run the next steps to compile, build and install the media drivers into my kernel:

```
sudo rm -rf /lib/modules/`uname -r`/kernel/drivers/media
git clone https://bitbucket.org/CrazyCat/media_build.git
cd media_build
./build
sudo make install
```

Reboot and DVB works fine.

But I noted that snd_usb_audio refuses to load and I get the next errors:

```
[    4.773446] snd_usb_audio: disagrees about version of symbol media_devnode_remove
[    4.773449] snd_usb_audio: Unknown symbol media_devnode_remove (err -22)
[    4.773694] snd_usb_audio: disagrees about version of symbol __media_device_register
[    4.773695] snd_usb_audio: Unknown symbol __media_device_register (err -22)
[    4.773748] snd_usb_audio: disagrees about version of symbol media_create_intf_link
[    4.773749] snd_usb_audio: Unknown symbol media_create_intf_link (err -22)
[    4.773756] snd_usb_audio: disagrees about version of symbol media_create_pad_link
[    4.773757] snd_usb_audio: Unknown symbol media_create_pad_link (err -22)
[    4.773864] snd_usb_audio: disagrees about version of symbol media_device_register_entity
[    4.773865] snd_usb_audio: Unknown symbol media_device_register_entity (err -22)
[    4.774030] snd_usb_audio: disagrees about version of symbol media_entity_pads_init
[    4.774031] snd_usb_audio: Unknown symbol media_entity_pads_init (err -22)
[    4.774074] snd_usb_audio: disagrees about version of symbol media_remove_intf_link
[    4.774075] snd_usb_audio: Unknown symbol media_remove_intf_link (err -22)
[    4.774247] snd_usb_audio: disagrees about version of symbol media_devnode_create
[    4.774248] snd_usb_audio: Unknown symbol media_devnode_create (err -22)
[    4.774261] snd_usb_audio: disagrees about version of symbol media_device_unregister_entity
[    4.774262] snd_usb_audio: Unknown symbol media_device_unregister_entity (err -22)

```
My system is: `Linux mymachine 5.4.0-4-amd64 #1 SMP Debian 5.4.19-1 (2020-02-13) x86_64 GNU/Linux`

How can I resolve this?

Thanks in advance.

as86mr, 2020-03-22:
I noticed the same problem a few month ago starting with kernel 5.2
I retested a few kernels now:
4.19.x are all o.k  (from 4.19.0 up to actual 4.19.112 (4.19 is my std kernel, I've updated often)  
5.1.16 is o.k.

All 5.2.x, 5.3.x, 5.4.x, 5.5.x have the same snd_usb_audio problem after installing tbs driver. (module cannot be loaded). Before installation of the tbs driver loading of the module is o.k. and sound works.
So something seems to have changed in kernel 5.2 that is affected

crazycat69, 2020-03-22:
Look like after this commit
https://github.com/tbsdtv/linux_media/commit/66354f18fe5fbb65f7b10a519654013d6df09f80

Tnix76, 2020-03-23:
So if I understand it fine the solution is update ALSA drivers?

LankyGit1, 2021-04-20:
I have just had the same issue where I didn't previously. 

Just updated to kernel 5.4.0-72-generic on Ubuntu so updated to the latest TBS drivers they work fine but then came across this same issue when I plugged in my usb camera/microphone. Camera works but microphone doesn't. Any help appreciated

Apr 17 19:42:22 homeserver mtp-probe: bus: 1, device: 4 was not an MTP device
Apr 17 19:42:22 homeserver kernel: [ 1034.822947] snd_usb_audio: disagrees about version of symbol media_devnode_remove
Apr 17 19:42:22 homeserver kernel: [ 1034.822948] snd_usb_audio: Unknown symbol media_devnode_remove (err -22)
Apr 17 19:42:22 homeserver kernel: [ 1034.823039] snd_usb_audio: disagrees about version of symbol __media_device_register
Apr 17 19:42:22 homeserver kernel: [ 1034.823039] snd_usb_audio: Unknown symbol __media_device_register (err -22)
Apr 17 19:42:22 homeserver kernel: [ 1034.823081] snd_usb_audio: disagrees about version of symbol media_create_intf_link
Apr 17 19:42:22 homeserver kernel: [ 1034.823082] snd_usb_audio: Unknown symbol media_create_intf_link (err -22)
Apr 17 19:42:22 homeserver kernel: [ 1034.823090] snd_usb_audio: disagrees about version of symbol media_create_pad_link
Apr 17 19:42:22 homeserver kernel: [ 1034.823090] snd_usb_audio: Unknown symbol media_create_pad_link (err -22)
Apr 17 19:42:22 homeserver kernel: [ 1034.823138] snd_usb_audio: disagrees about version of symbol media_device_register_entity

as86mr, 2021-04-21:
Hi all
I don't have a mitigation for this. I just stay at the last long-time-support kernel 4.19 that is not having that problem.
I have additionally an up to date kernel (now 5.11.15) : All kernels 5.2 up to 5.11 have this problem.

@crazycat69 
I tried also one year ago without success after comment of crazycat69 to revert the commit 66354f1 of the tbs media-stack. i don't remeber anymore which kernel maybe 5.4.x or 5.5.x

Updating alsa doesn't help as well.

So for me it looks like the tbs media stack that is substituting completely the linux-kernel media stack is somehow incompatible with the rest of the kernel concerning snd_usb_audio.

So at moment tbs card cannot be recommended in multimedia PC with USB soundcard or you need some workaround with older kernel. 

Best would be to report this error more prominent at www.tbsdtv.com to give that problem more attention and link to here
How to report a problem with TBS Linux drivers: https://www.tbsdtv.com/forum/viewtopic.php?f=87&t=2194

as86mr, 2021-04-21:
The title is also a little misleading (but hopefully obvious): 
The problem occurs AFTER INSTALLATION of the tbs driver and therefore after substituting the original kernel-media stack ( and NOT by compiling )

LankyGit1, 2021-04-21:
Thanks for your reply @as86mr 

But something is not quite right. USB microphone sound was working prior to the my latest update.
Before the update I was on kernel 5.4.0.65 and then went to ..72 so only a minor update. I then updated the TBS driver source (which could only be a month newer, sorry don't know exactly). It was only after this kernel and source update that the usb sound issue occurred. So I don't think it could be the commit you say as that was way back in 2019 which would have already been in my previous source.

Sorry if I am getting something confused.

crazycat69, 2021-04-21:
Some drivers beside media tree depend from core media divers (media-controller, cec and etc)

marki5678, 2021-09-22:
I also have this problem. Is there any resolution?

Taymirec, 2022-01-20:
I found some than may be help for solve this problem: https://unix.stackexchange.com/questions/125624/unable-to-load-module-disagrees-about-version-of-symbol-module-layout

Looks like problem in CONFIG_MODVERSIONS.

Taymirec, 2022-01-21:
Yes! I now hear sound from my USB Audio interface with TBS DVB module, this happened after I recompile kernel from source with disabled CONFIG_MODVERSIONS…

LankyGit1, 2022-02-12:
This needs sorting in a different way than having to recompile kernel from source other USB drivers should NOT be affected by having a TBS driver. I refer to my previous comment that the USB audio worked on an earlier version of the TBS driver.

as86mr, 2022-03-06:
Only solution seems to be now recompiling the whole kernel with disabled CONFIG_MODVERSIONS ...
My workaround until 2 months ago was staying at the last LTS kernel 4.19 (which was offered to download for my OpenSuse )
Now I switched to another tv card ... (digital devices Cine S2 ) where driver in included in kernel.

nshiell, 2022-04-10:
Will this be fixed for Ubuntu 22.04, which will be released on 21st April 2022 and will use Kernel 5.15 by default?

nshiell, 2022-04-25:
FYI the latest version of Ubuntu came out and this issue still exists!
I have contacted tbsdtv about this

alexjfisher, 2022-11-11:
@nshiell Did you get any response?

stulluk, 2024-07-20:
@crazycat69  is there any solution to this issue ?

crazycat69, 2024-07-20:
need recompile usb-audio modules with media_build

stulluk, 2024-07-20:
> need recompile usb-audio modules with media_build

I couldn't understand. it already recompiles and overrides usb-audio modules used by existing kernel in the host. this is the reason of the problem already.

How can we avoid this? can you show an example?

kaswa42, 2025-01-04:
+1
my USB audio devices don't work after compiling the drivers

thecosmicslug, 2025-02-13:
> need recompile usb-audio modules with media_build

I'm still struggling with this issue, can we have a little clarification about how to recompile usb-audio modules with media_build ??

skyweb2005, 2025-03-01:
I use a workaround and force load module
_modprobe -f snd_usb_audio_
and add it at startup
not perfect but it's ok for me

andimik, 2025-05-19:
Same here. The github driver version works perfectly under latest Ubuntu, 25.04 

BUT whenever I install the offline drivers from the TBS homepage, the video and audio drivers are gone and the screen resolution is low.

My proposal is that TBS removes these offline drivers.

https://www.tbsiptv.com/index.php?route=product/download/drivers&id=119