tbsdtv/linux_media #195: Can not Load mn88436 driver.

zmdk0913, 2020-04-10:
I installed tbsdtv driver follows menual, So I complete build and install on my linux ubuntu 19.04 kernel 5.4.0-rc1.
And I typed this command 'modprobe mn88436, modprobe mxl603, modprobe dvb-usb-cxusb'
And I reboot server and command typed. dmesg
USB Tuner A681 is can saw, but about DVB enabled don't see. Why I have problem? Please help me.

crazycat69, 2020-04-10:
Yoy must add support for your usb stick to cxusb driver :)

zmdk0913, 2020-04-11:
how can I add cxusb deiver?I installed dvb-usb-cxusb, Shoud I need to install another?
I use Mygica a681(mohu)

crazycat69, 2020-04-11:
add device usb id and tuner/frontend attach

zmdk0913, 2020-04-11:
I don't know anything. Can you help me with command?

crazycat69, 2020-04-11:
better try support from Chris Lee - [v4l-updatelee](https://gitlab.com/updatelee/v4l-updatelee)
some descriptions from US - https://rickcaylor.websitetoolbox.com/post/v4lupdatelee-6433163?pid=1309850794&highlight=a681

this based on Geniatech V4L DVB sources for mxl603/mn88436
https://gitlab.com/updatelee/v4l-updatelee/-/blob/master/drivers/media/tuners/mxl603.c
https://gitlab.com/updatelee/v4l-updatelee/-/blob/master/drivers/media/dvb-frontends/mn88436.c
https://gitlab.com/updatelee/v4l-updatelee/-/blob/master/drivers/media/usb/dvb-usb/cxusb.c#L1406

closed 2020-04-11 by crazycat69