tbsdtv/linux_media #199: 6909x degraded performance with a single UHD HEVC channel

Neolo, 2020-05-16:
6909x quickly becomes incredible slow with a single UHD HEVC channel tuned of 15Mbps bitrate, which is also being descrambled. Works behind tvheadend. Please fix this in the driver.

Davin622, 2020-05-16:
Could you give me more info of this frequency? and please send the email to support@tbsdtv.com.

Neolo, 2020-05-16:
Motherboard X99, Intel CPU Xeon 12 Cores.

```
May 16 03:06:00 kernel: [ 3801.090526] pcieport 0000:00:01.0: AER: Corrected error received: 0000:00:01.0
May 16 03:06:00  kernel: [ 3801.090530] pcieport 0000:00:01.0: AER: PCIe Bus Error: severity=Corrected, type=Data Link Layer, (Transmitter ID)
May 16 03:06:00  kernel: [ 3801.090533] pcieport 0000:00:01.0: AER:   device [8086:2f02] error status/mask=00001000/00002000
May 16 03:06:00 i kernel: [ 3801.090535] pcieport 0000:00:01.0: AER:    [12] Timeout
May 16 03:06:00 kernel: [ 3801.267325] pcieport 0000:00:01.0: AER: Corrected error received: 0000:00:01.0
May 16 03:06:00  kernel: [ 3801.267329] pcieport 0000:00:01.0: AER: PCIe Bus Error: severity=Corrected, type=Data Link Layer, (Transmitter ID)
May 16 03:06:00  kernel: [ 3801.267332] pcieport 0000:00:01.0: AER:   device [8086:2f02] error status/mask=00001000/00002000
May 16 03:06:00 kernel: [ 3801.267333] pcieport 0000:00:01.0: AER:    [12] Timeout
```
10994  |   UHD 1 | DVB-S2HEVC/UHD 4K | 22000-5/68PSK

Kernel flags pci=nommconf pcie_aspm=off - don’t help. No problems with 6909 (not X)

crazycat69, 2020-05-21:
What SNR on thsi transponder ?
Or you question about PCIe bus errors ?

Neolo, 2020-05-21:
> What SNR on this transponder ?
> Or you question about PCIe bus errors ?

65%

I don’t know what to ask, this card either it’s drivers or combination of both are unusable. TBS is sending v.2 hardware build of it.

crazycat69, 2020-05-21:
I can't check this transponder in eastern europe. But now receive 6 transponder from 4.8E with 6909x PCB 1.0 and tvh 4.2. No any CC or PCIe errors, no tvh gui hangs (with latest performance improvement in stid135 driver)

Neolo, 2020-05-21:
Astra 19 well received in Easter Europe.
And no, tune 1-2 UHD HEVC channels with descrambler and then tell me how it goes.

crazycat69, 2020-05-21:
Ukraine in far eastern europe :)

Neolo, 2020-05-21:
Then check with something else UHD.

crazycat69, 2020-05-21:
Ok, **NASA UHD @ 13E** 10727,H,30000Ks,8PSK,3/4.
Total transponder bitrate ~66Mbps, channel bitrate ~15-20mbit.
+ another 7 DVB-S2,8PSK,~30000Ks transponders for STi135 stress-test (LDPC decoder performance limited to 720 LLR, so possible receive 8 transponder 8PSK,30000 Ks). + 9 VLC 3.x instances for CPU loading.

Conclusion: after 3 hours no CC on 7 transponders, just one on high frequency have some little SNR and some CC. TVH GUI responsible, signal status update default 1000 ms.
[image]
[image]
[image]

P.S. 6 year old PC with Core i5-3470, B75 chipset, 8Gb RAM (DDR3-1600) and 6909x PCB 1.0. Kubuntu 18.04 LTS with HWE 5.3 kernel, TVH 4.2 latest.

Neolo, 2020-05-22:
What CC? I’m not talking about CC, but it’s performance, causing card to be unresponsive, throwing PCIe data link errors and crashing TVH. And this is confirmed by TBS.
If you will tune those channels and watch them still - this is not a stress test. Go ahead tune your channels and try to scan 30-64 transponders simultaneously using free tuners, you will be surprised how garbage that card is, comparing to older model 6909 or any others.
It’s just easier for you to verify if it’s actually hardware or driver, but please don’t make things look like I’m making it up here, I wasted over $600 on these junk cards.

crazycat69, 2020-05-22:
CC is Continuity errors, column before SNR in TVH stream statistic.

I think you have some hw problems, maybe related to PCB 1.x or FPGA firmware.

Neolo, 2020-05-22:
I know what’s CC.
I have? On 2 cards at the same time in different hosts? Yeah right… I will get a different model of the tuner and will ask for proprietary driver instead. Seem these talks lead to nowhere, at least because you aren’t even able to tune Astra 19.

Neolo, 2020-05-22:
Wow, after few days of runtime, we have something new. Not just TBS card has crashed, but also a Hauppauge quadHD, which is installed aside, stopped responding.

May 22 12:50:27-tvheadend[16256]: mpegts: 10803H in Astra 19.2E [HL VL] - tuning on TurboSight TBS 6909x (Octa DVB-S/S2/S2X) #11 : DVB-S #0 (A CABLE-2)
May 22 12:50:28-kernel: [524376.555777] TBSECP3 driver 0000:01:00.0: i2c xfer timeout
May 22 12:50:29-kernel: [524377.579777] TBSECP3 driver 0000:01:00.0: i2c xfer timeout
May 22 12:50:30-kernel: [524378.603769] TBSECP3 driver 0000:01:00.0: i2c xfer timeout
May 22 12:50:31-kernel: [524379.627754] TBSECP3 driver 0000:01:00.0: i2c xfer timeout
May 22 12:50:32-kernel: [524380.651729] TBSECP3 driver 0000:01:00.0: i2c xfer timeout
May 22 12:50:33-kernel: [524381.675711] TBSECP3 driver 0000:01:00.0: i2c xfer timeout
May 22 12:50:34-kernel: [524382.699688] TBSECP3 driver 0000:01:00.0: i2c xfer timeout
May 22 12:50:35-kernel: [524383.723706] TBSECP3 driver 0000:01:00.0: i2c xfer timeout
May 22 12:50:36-kernel: [524384.747679] TBSECP3 driver 0000:01:00.0: i2c xfer timeout
May 22 12:50:37-kernel: [524385.771662] TBSECP3 driver 0000:01:00.0: i2c xfer timeout
May 22 12:50:38-tvheadend[16256]: diseqc: failed to send diseqc cmd (e=Operation not permitted)
May 22 12:50:38-tvheadend[16256]: en50494: error send tune command

Neolo, 2020-05-22:
May 22 12:54:02  kernel: [524591.245241] i2c i2c-0: stid135_init: enable tuner 0 + demod 4 error 29 !
May 22 12:54:02  kernel: [524591.245304] TBSECP3 driver 0000:01:00.0: DVB: adapter 8 frontend 0 frequency 0 out of range (950000..2150000)
May 22 12:54:03  kernel: [524592.265242] TBSECP3 driver 0000:01:00.0: i2c xfer timeout
May 22 12:54:04  kernel: [524593.289195] TBSECP3 driver 0000:01:00.0: i2c xfer timeout
May 22 12:54:05  kernel: [524594.313185] TBSECP3 driver 0000:01:00.0: i2c xfer timeout
May 22 12:54:06  kernel: [524595.401170] TBSECP3 driver 0000:01:00.0: i2c xfer timeout
May 22 12:54:07  kernel: [524595.529197] i2c i2c-0: stid135_init: enable tuner 0 + demod 4 error 29 !
May 22 12:54:08  kernel: [524596.553182] TBSECP3 driver 0000:01:00.0: i2c xfer timeout
May 22 12:54:09  kernel: [524597.577122] TBSECP3 driver 0000:01:00.0: i2c xfer timeout
May 22 12:54:10  kernel: [524598.601156] TBSECP3 driver 0000:01:00.0: i2c xfer timeout
May 22 12:54:11  kernel: [524599.689142] TBSECP3 driver 0000:01:00.0: i2c xfer timeout
May 22 12:54:11  tvheadend[11579]: linuxdvb: adapter added /dev/dvb/adapter7
May 22 12:54:11  kernel: [524599.817103] i2c i2c-0: stid135_init: enable tuner 0 + demod 3 error 29 !
May 22 12:54:12  kernel: [524600.841126] TBSECP3 driver 0000:01:00.0: i2c xfer timeout
May 22 12:54:13  kernel: [524601.865123] TBSECP3 driver 0000:01:00.0: i2c xfer timeout
May 22 12:54:14  kernel: [524602.889067] TBSECP3 driver 0000:01:00.0: i2c xfer timeout
May 22 12:54:15  kernel: [524603.913063] TBSECP3 driver 0000:01:00.0: i2c xfer timeout
May 22 12:54:16  kernel: [524604.937061] TBSECP3 driver 0000:01:00.0: i2c xfer timeout
May 22 12:54:17  kernel: [524605.961040] TBSECP3 driver 0000:01:00.0: i2c xfer timeout
May 22 12:54:18  kernel: [524606.985056] TBSECP3 driver 0000:01:00.0: i2c xfer timeout
May 22 12:54:19  kernel: [524608.009045] TBSECP3 driver 0000:01:00.0: i2c xfer timeout
May 22 12:54:20  kernel: [524609.032988] TBSECP3 driver 0000:01:00.0: i2c xfer timeout
May 22 12:54:21  kernel: [524610.056998] TBSECP3 driver 0000:01:00.0: i2c xfer timeout
May 22 12:54:22  kernel: [524611.081000] TBSECP3 driver 0000:01:00.0: i2c xfer timeout
May 22 12:54:23  kernel: [524612.104988] TBSECP3 driver 0000:01:00.0: i2c xfer timeout
May 22 12:54:24  kernel: [524613.128938] TBSECP3 driver 0000:01:00.0: i2c xfer timeout
May 22 12:54:24  kernel: [524613.128952] i2c i2c-0: fe_stid135_get_band_power_demod_not_locked error
May 22 12:54:24  kernel: [524613.128969] i2c i2c-0: stid135_set_parameters: fe_stid135_set_pls error 13 !
May 22 12:54:24  kernel: [524613.128976] i2c i2c-0: stid135_set_parameters: fe_stid135_reset_modcodes_filter error 13 !
May 22 12:54:24  kernel: [524613.128985] i2c i2c-0: stid135_set_parameters: fe_stid135_search error 13 !
May 22 12:54:25  kernel: [524614.152955] TBSECP3 driver 0000:01:00.0: i2c xfer timeout
May 22 12:54:26  kernel: [524615.176937] TBSECP3 driver 0000:01:00.0: i2c xfer timeout
May 22 12:54:27  kernel: [524616.200929] TBSECP3 driver 0000:01:00.0: i2c xfer timeout
May 22 12:54:28  kernel: [524617.288912] TBSECP3 driver 0000:01:00.0: i2c xfer timeout
May 22 12:54:28  tvheadend[11579]: linuxdvb: adapter added /dev/dvb/adapter6
May 22 12:54:28  kernel: [524617.416916] i2c i2c-0: stid135_init: enable tuner 0 + demod 2 error 29 !
May 22 12:54:28  kernel: [524617.416938] TBSECP3 driver 0000:01:00.0: DVB: adapter 6 frontend 0 frequency 0 out of range (950000..2150000)
May 22 12:54:29  kernel: [524618.440897] TBSECP3 driver 0000:01:00.0: i2c xfer timeout
May 22 12:54:31  kernel: [524619.464883] TBSECP3 driver 0000:01:00.0: i2c xfer timeout
May 22 12:54:32  kernel: [524620.488860] TBSECP3 driver 0000:01:00.0: i2c xfer timeout

crazycat69, 2020-05-22:
Some wrong with your CPU & MB setup ? Need more details about hw and linux distro.

Neolo, 2020-05-22:
No, nothing wrong with that, because whenever I put older 6909, it runs smoothly. And cpu is 12 cores.

crazycat69, 2020-05-22:
But you report before about same problems with Hauppauge QuadHD card (cx23885-based?).

6909 based on mxl584 demod, which have on-chip MCU (big firmware loaded on driver init). So btw mxl584 and driver low I2C traffic (so low I2C interrupts).

Try "lock" card interrupt handler to one CPU core + disable irqbalance service. You can put optimization script to /etc/rc.local
`#!/bin/bash

#### interrupts-tuning

service irqbalance stop
killall irqbalance

ncpus=`grep -ciw ^processor /proc/cpuinfo`
test "$ncpus" -gt 1 || exit 1
n=0

for irq in `cat /proc/interrupts | grep 'cx23885\|saa716x_core\|tbsecp3\|eth\|ddbridge\|eno\|ens' | awk '{print $1}' | sed s/\://g` ; do
    f="/proc/irq/$irq/smp_affinity"
    test -r "$f" || continue
    cpu=$[$ncpus - ($n % $ncpus) - 1]
    if [ $cpu -ge 0 ] ; then
        mask=`printf %x $[2 ** $cpu]`
        echo "Assign SMP affinity: dev$n, irq $irq, cpu $cpu, mask 0x$mask"
        echo "$mask" > "$f"
        let n+=1
    fi
done

#### cpu-cloc

cpucount=$(grep -c 'model name' /proc/cpuinfo)
sysdir=/sys/devices/system/cpu
for cpu in $(eval echo cpu{0..$((cpucount-1))}); do
        cat $sysdir/$cpu/cpufreq/scaling_max_freq > $sysdir/$cpu/cpufreq/scaling_min_freq
done
`

Neolo, 2020-05-23:
> Almost the same, because it works out of the same linuxdvb, which gets stuck. Problem goes away when 6909x is off

Script you have provided doesn’t seem to be a solution, makes no difference. And I don’t want to lock it to just one core, that’s dull...

Neolo, 2020-05-24:
Appears it doesn’t even execute.

/interrupts_tuning.sh: line 12: syntax error near unexpected token `|'
/interrupts_tuning.sh: line 12: `for irq in cat /proc/interrupts | grep 'cx23885\|saa716x_core\|tbsecp3\|eth\|ddbridge\|eno\|ens' | awk '{print $1}' | sed s/\://g ; do’

crazycat69, 2020-05-24:
latest source tune system script
http://cesbo.com/download/astra/scripts/tune.sh

Neolo, 2020-05-24:
I fixed the script, you didn’t wrap commands in $() on certain lines.

After applying:

Assign SMP affinity: dev0, irq 47, cpu 11, mask 0x800
Assign SMP affinity: dev1, irq 50, cpu 10, mask 0x400
Assign SMP affinity: dev2, irq 51, cpu 9, mask 0x200

And now all the CC errors are flooding on any VH bands of any satellite. Picture unwatchable. Note, I keep Astra 19 running aside on purpose, it does slow down the TVH when tuned in.

crazycat69, 2020-05-24:
So some wrong with your MB & CPU setup.

Neolo, 2020-05-24:
Thanks, executed:

model name      : Intel(R) Xeon(R) CPU E5-2620 v3 @ 2.40GHz
power scheme
performance
changed to
performance
sysctl: cannot stat /proc/sys/kernel/exec-shield: No such file or directory
sysctl: cannot stat /proc/sys/kernel/maps_protect: No such file or directory
sysctl: setting key "net.ipv4.ipfrag_high_thresh": Invalid argument
sysctl: cannot stat /proc/sys/net/ipv4/netfilter/ip_conntrack_buckets: No such file or directory
sysctl: cannot stat /proc/sys/net/ipv4/netfilter/ip_conntrack_generic_timeout: No such file or directory
sysctl: cannot stat /proc/sys/net/ipv4/netfilter/ip_conntrack_icmp_timeout: No such file or directory
sysctl: cannot stat /proc/sys/net/ipv4/netfilter/ip_conntrack_tcp_timeout_close: No such file or directory
sysctl: cannot stat /proc/sys/net/ipv4/netfilter/ip_conntrack_tcp_timeout_close_wait: No such file or directory
sysctl: cannot stat /proc/sys/net/ipv4/netfilter/ip_conntrack_tcp_timeout_established: No such file or directory
sysctl: cannot stat /proc/sys/net/ipv4/netfilter/ip_conntrack_tcp_timeout_fin_wait: No such file or directory
sysctl: cannot stat /proc/sys/net/ipv4/netfilter/ip_conntrack_tcp_timeout_last_ack: No such file or directory
sysctl: cannot stat /proc/sys/net/ipv4/netfilter/ip_conntrack_tcp_timeout_syn_recv: No such file or directory
sysctl: cannot stat /proc/sys/net/ipv4/netfilter/ip_conntrack_tcp_timeout_syn_sent: No such file or directory
sysctl: cannot stat /proc/sys/net/ipv4/netfilter/ip_conntrack_tcp_timeout_syn_sent2: No such file or directory
sysctl: cannot stat /proc/sys/net/ipv4/netfilter/ip_conntrack_tcp_timeout_time_wait: No such file or directory
sysctl: cannot stat /proc/sys/net/ipv4/netfilter/ip_conntrack_udp_timeout: No such file or directory
sysctl: cannot stat /proc/sys/net/ipv4/netfilter/ip_conntrack_udp_timeout_stream: No such file or directory
sysctl: cannot stat /proc/sys/net/ipv4/tcp_tw_recycle: No such file or directory
Assign SMP affinity: dev0, irq 47, cpu 11, mask 0x800

VH bands are still in CC flood as described in previous reply. There is nothing wrong with cabling or LNB, that was checked already.

Neolo, 2020-05-24:
Would be wrong, then none of TBS cards would work, but problem is exclusively with 6909X model. Not 6909, not 6590.

Neolo, 2020-05-24:
Today on completely different host with different mobs and CPU, with card 6909 (not X) I’m facing the same problem after I tried to scan Hotbird 13E transponders, while 2 channels where tuned aside.
Do you still think this is me?

GRUB_CMDLINE_LINUX_DEFAULT="pcie_aspm=performance"

Neolo, 2020-05-24:
Damn, this is terrible…. Is there anything else to try but not TBSECP3 driver?? This thing is f*ed up.

tvhdeeptho, 2020-05-29:
Is the UHD1 channel the one on  10993.75 H 

I can receive it without problems.
[image]

Neolo, 2020-05-29:
I did not say I can’t receive it, I said that the card is getting slow and crashes when this channels is tuned AND I try to scan something else ALONGSIDE or tune other channels or browse TVH menus, which take forever to load.

tvhdeeptho, 2020-05-29:
@Neolo 
I understand what you are saying. The point is that I do not experience these problems. I know that the screenshot does not prove that, but there are no i2c errors in the logs. tvheadend works as normal. There are no crashes and no slowness. it just works as expected.

It is possible there is something wrong with your card, but one of your posts states you also have problems with the tbs6909:
> `Today on completely different host with different mobs and CPU, with card 6909 (not X) I’m facing the same problem after I tried to scan Hotbird 13E transponders, while 2 channels where tuned aside.
Do you still think this is me?`
So this sounds like the common factor may be something else.

When you have the problem with tbs6909, was the tbs6909x also inserted at the same time?

When you receive the v.2 card: if this does not solve your problem, try to reproduce the problem with
only  1 card inserted in the pc. It is easier to find the cause then.

I have been testing recently the v.2 card, including epg scans. With 6 or 7 tuners enabled, including 1 live channel it worked fine. 8 is too much unless all the transponder have a symbolrate <= 30Ms/s.
That is a hardware limitation. The number of transponders in my database is huge....

I also ran a number of other tests: high symbolrates (45 Ms/s), low symbolrates (below 1000 with some driver patch), multistreams, weak channels. No problems found. When I finally found a transponder with errors, it turned out to be a problem with the stream (verified by running another completely different type of card in parallel on the same mux. The errors occurred exactly at the same times on both cards.)

There may still be some small driver problems. All software has some bugs which pop up at unexpectedly. 

So don't despair.

Neolo, 2020-05-29:
No, I have 6909 and 6909x in different PCs. You just don’t read what I write, don’t you? "Today on completely different host with different mobs and CPU, with card 6909”
Neither you read what I wrote in previous post. Did I ever mention EPG scan? You making up your own conditions and saying there is nothing wrong. This is quite frustrating, because I start feel like I’m an idiot here.
Please read this one more time: 6909x card is getting slow and crashes when channel UHD1 4K HEVC and/or Hotbird 4K and/or Nasa UHD HEVC are tuned AND I try to scan something else ASIDE (entire satellite like 16E or 13E through the list of pre-saved transponders in TransEdit tool which is a part of DVBViewer) and when I browse TVH menus, it takes forever to load those menus.

6909 and 6909x are work out of the same driver, apparently.

6909 got fixed by disabling MSI in
/etc/modprobe.d/tbsecp3.conf
options tbsecp3 enable_msi=0

Thanks to TBS support over email. However, this has no effect on 6909x and I have 2x 6909x cards acting same way, as well as 3 PCs build with different components. So please don’t tell me it’s so individual.

tvhdeeptho, 2020-05-30:
@Neolo 
-Where did I write you are wrong?
-an epg scan causes tuning to a transponder and has the same or worse effects as tuning for viewing; I also wrote of other tests. This included tests with 7 live channels.
-if the problem is with two different cards, then it is unlikely (not impossible) that the card is the problem
-I was trying to work out from your emails if you also have problems on tbs 909 (not X). Still not sure.
-getting angry at people who try to help (two so far) is not productive; they will just stop helping. This is also going to happen now.

Neolo, 2020-05-30:
Ok. In past 5 emails all you do is trying to prove it’s my solo problem, reading my messages between lines. That wasn’t helpful.